hbldh/bleak #1352: OSError: [WinError -2140864509] The attribute cannot be written in start_notify

Kaz1200r, 2023-06-29:
* bleak version:0.20.1
* Python version:3.11.2
* Operating System:Windows 11 22H2

### Description

I'm trying to connect to a notify characteristic that I know it works on Android app. Now I'm creating a Python app because I need to create a Windows version, but the notify characteristic always says the same error: 
OSError: [WinError -2140864509] The attribute cannot be written.

### What I Did
I am checking the operation of start_notify for custom characteristic
```
# 1. Enable Notify
await client.write_gatt_char("00010203-0405-0607-0809-0a0b0c0d1911", bytes([0x01]))

# 2. Send Command
await client.write_gatt_char("00010203-0405-0607-0809-0a0b0c0d1912", send_command(["on"]))

# 3. Start Notify
await client.start_notify("00010203-0405-0607-0809-0a0b0c0d1911", notification_handler)
```

The output of this fragment of code is:
1. Enable Notify
Write 0x01 to the custom UUID for the connected device to send a notify.
2. Send Command
Send a command to send a notify from the connected device.
3. Start Notify
Starting notify.

### Logs

All error messages
```
Traceback (most recent call last):
  File "C:\Users\kaz\PycharmProjects\MyProject\ble_device.py", line 58, in <module>
    loop.run_until_complete(main(mac_address))
  File "C:\Users\kaz\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\kaz\PycharmProjects\MyProject\ble_device.py", line 46, in main
    await client.start_notify("00010203-0405-0607-0809-0a0b0c0d1911", notification_handler)
  File "C:\Users\kaz\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\__init__.py", line 717, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "C:\Users\kaz\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 951, in start_notify
    await winrt_char.write_client_characteristic_configuration_descriptor_async(
OSError: [WinError -2140864509] The attribute cannot be written.
```

Result of service_explorer.py
```
2023-06-28 22:07:57,207 __main__ INFO: [Service] 00010203-0405-0607-0809-0a0b0c0d1910 (Handle: 16): Unknown
2023-06-28 22:07:57,236 __main__ INFO:   [Characteristic] 00010203-0405-0607-0809-0a0b0c0d1911 (Handle: 17): Status (read,write,notify), Value: bytearray(b'\x01')
2023-06-28 22:07:57,266 __main__ INFO:     [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 19): Characteristic User Description, Value: bytearray(b'Status')
2023-06-28 22:07:57,298 __main__ INFO:   [Characteristic] 00010203-0405-0607-0809-0a0b0c0d1912 (Handle: 20): Command (read,write-without-response,write), Value: bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-06-28 22:07:57,328 __main__ INFO:     [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 22): Characteristic User Description, Value: bytearray(b'Command')
```

Result of Wireshark
```
236	8.321131	localhost ()	a5:c1:01:00:04:45 (12345678_1)	ATT	13	Sent Write Command, Handle: 0x0012 (Unknown: Unknown)
237	8.322888	localhost ()	a5:c1:01:00:04:45 (12345678_1)	ATT	32	Sent Write Command, Handle: 0x0015 (Unknown: Unknown)
238	8.324333	localhost ()	a5:c1:01:00:04:45 (12345678_1)	ATT	32	Sent Write Command, Handle: 0x0015 (Unknown: Unknown)
239	8.331755	host	controller	HCI_CMD	6	Sent LE Set Scan Enable
240	8.334049	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Scan Enable)
241	8.334090	host	controller	HCI_CMD	11	Sent LE Remove Device From White List
242	8.335044	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
243	8.336046	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
244	8.337041	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
245	8.338039	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
246	8.339039	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Remove Device From White List)
247	8.339051	host	controller	HCI_CMD	6	Sent LE Set Scan Enable
248	8.341081	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Scan Enable)
249	8.362642	a5:c1:01:00:04:45 (12345678_1)	localhost ()	ATT	32	Rcvd Handle Value Notification, Handle: 0x0012 (Unknown: Unknown)
250	8.377663	a5:c1:01:00:04:45 (12345678_1)	localhost ()	ATT	32	Rcvd Handle Value Notification, Handle: 0x0012 (Unknown: Unknown)
251	8.392659	a5:c1:01:00:04:45 (12345678_1)	localhost ()	ATT	32	Rcvd Handle Value Notification, Handle: 0x0012 (Unknown: Unknown)
252	8.822114	host	controller	HCI_CMD	18	Sent LE Connection Update
253	8.824082	controller	host	HCI_EVT	7	Rcvd Command Status (LE Connection Update)
254	8.949095	controller	host	HCI_EVT	13	Rcvd LE Meta (LE Connection Update Complete)
255	9.292507	a5:c1:01:00:04:45 (12345678_1)	localhost ()	L2CAP	21	Rcvd Connection Parameter Update Request
256	9.292525	localhost ()	a5:c1:01:00:04:45 (12345678_1)	L2CAP	15	Sent Connection Parameter Update Response (Accepted)
257	9.331045	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
258	11.341532	host	controller	HCI_CMD	7	Sent Disconnect
```
I guess that the reason for the error is that cccd (0x2902) of Handle 18 cannot be obtained from the device.
According to the Wireshark log, Notification is being sent from the device with Handle 18 (0x0012), is there a way to get the Notification?

If someone knows what is the mistake I'm doing, I will appreciate some help.

Thanks a lot.

dlech, 2023-06-29:
There have been numerous reports of devices like this that don't properly implement the Client Characteristic Configuration Descriptor. We might be able to separate enabling notifications from subscribing to notifications, but it needs some investigation to make sure it will actually work on all OSes.

Kaz1200r, 2023-06-29:
Thank you for your reply.
I would like to see a feature added to separate enabling notifications from subscribing to notifications.
thank you very much. This matter will be closed.

closed 2023-06-29 by Kaz1200r

rockluoge, 2023-11-21:
I have the same issue. @dlech how to separate enabling notifications from subscribing to notifications, I can have a try firstly.