hbldh/bleak #1358: RPi - BlueZ: client.services returning empty

Rxb2300, 2023-07-04:
* bleak version: 0.20.2
* Python version: 3.9.2
* Operating System: Linux (Raspbian running on Raspberry Pi 4)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description
I am trying to run a GATT server on a raspberry pi 4 and have been using the following example to get me started: https://github.com/PunchThrough/espresso-ble. The only modification I made to this code was to comment out the content of the AurthorizeService method in the Agent class to automatically return without requesting a "yes" from the user when polled by another connected device.

This server seems to run alright, I can identify it using my Android device running nRF connect, I can connect with it, and read / write to the service / characteristics. 

On a second raspberry pi I am trying to run a simple BLE central to scan, detect, connect, and read/write to the previously mentioned peripheral. I am utilizing bleak for this portion of the project. 

On this Pi I am running the service_explorer.py example from this repository to connect with the other device running the Vivaldi service and request a list of services / characteristics. 

While it does connect to the peripheral pi, and the peripheral pi does print out that it automatically granted service request to the central pi, I see nothing printed out to the terminal. 

[image]

It seems as if though client.services is returning empty. 

I tried the same but using the outdated BluePy module running this script:

```
from bluepy.btle import Peripheral, BTLEException

def main():
    
    device_mac_address = "D8:3A:DD:15:6A:87"

    
    try:
        
        print(f"Attemptingt to connect to device: {device_mac_address}")
        p = Peripheral(device_mac_address)

        
        for service in p.getServices():
            print(f"Service: {service}")
            
            for char in service.getCharacteristics():
                print(f"\tCharacteristic: {char}")
            
    except BTLEException as e:
        print(f"Unable to connect to device {e}")
        
    finally:
        
        print(f"Disconnecting form device: {device_mac_address}")
        p.disconnect()
        
        
if __name__ == "__main__":
    
    main()

```

And I got the following when it was able to successfully connect. 

[image]

The service / characteristics at the bottom are what I was looking for. 

### Logs

Attached below is a portion of the BTMON log running on the central device (The pi running the bleak example). 

[central_btmon_log.txt](https://github.com/hbldh/bleak/files/11944377/central_btmon_log.txt)

The central address is : D8:3A:DD:0C:AF:9F
The peripheral address is : D8:3A:DD:15:6A:87
The service uuid of interest is : "12634d89-d598-4874-8e86-7d042ee07ba7". 

If you search for uuid in the log of the central, you can find the service when it ran the bleak example.

StefJar, 2023-07-17:
the bluez implementation is connecting to dbus objects. The dbus objects are created via bluez. Bluez is only resolving known services & characteristics.
So I guess your service is not resolved by the bluetoothd. Maybe there are some bugs at the GATT database or so. You can use "sudo btmon" to check for errors.

Older python modules like bluepy that are using hcitool etc directly. So you can resolve you GATT service directly there. To my knowledge is bleak not supporting the registration of user specific GATT profiles

dlech, 2023-07-19:
I cannot get https://github.com/PunchThrough/espresso-ble/blob/master/ble.py to run on Raspbian Bullseye. Can you please provide a working test case?