hbldh/bleak #1373: macOS: BluetoothServiceInfoBleak fails with AdvertisementData from internal device

oyvindwe, 2023-07-24:
* bleak version: 0.20.2
* Python version: 3.11.4
* Operating System: macOS 12.6.7
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

I tried to run a clean Home Assistant from the dev branch on macOS. This logs the below exception several times per second.

It is possible to work around the issue in Home Assistant, but the error occurs when the value from `AdvertisementData.rssi` is passed to `BluetoothServiceInfoBleak()`, so this pretty much feels like a Bleak issue.

I _think_ it's only a matter of setting `rssi=int(r)` and `platform_data=(p, a, int(r))` here: https://github.com/hbldh/bleak/blob/b60f793cf14861686c5beae94822d7c0686b91ba/bleak/backends/corebluetooth/scanner.py#L130-L131

### What I Did

Ran Home Assistant on directly on macOS (not in a Docker container), per https://developers.home-assistant.io/docs/development_environment#manual-environment

```
brew install python3 autoconf ffmpeg cmake
git clone git@github.com:home-assistant/core.git home-assistant # preferably use your own fork
cd home-assistant
./script/setup
source venv/bin/activate
hass -c config
```

### Logs

The following exception is logged several times per second:
```
2023-07-24 22:52:42.683 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback CentralManagerDelegate.did_discover_peripheral(<CBCentralMan...x60000247c980>, <CBPeripheral... disconnected>, {
    kCBAdvD...rLevel = 12;
}, -96)
Traceback (most recent call last):
  File "/Users/omw/.pyenv/versions/3.11.4/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/omw/proj/home-assistant/venv/lib/python3.11/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 265, in did_discover_peripheral
    callback(peripheral, advertisementData, RSSI)
  File "/Users/omw/proj/home-assistant/venv/lib/python3.11/site-packages/bleak/backends/corebluetooth/scanner.py", line 152, in callback
    self._callback(device, advertisement_data)
  File "/Users/omw/proj/home-assistant/homeassistant/components/bluetooth/scanner.py", line 194, in _async_detection_callback
    BluetoothServiceInfoBleak(
TypeError: Argument 'rssi' has incorrect type (expected int, got OC_PythonLong)
```

oyvindwe, 2023-07-24:
I realize that `BluetoothServiceInfoBleak` is HA and not Bleak, but `AdvertisementData.rssi` is in any case typed as `int`, so I still think this should be fixed in Bleak.

dlech, 2023-07-24:
I'm curious what is actually doing the type check. I looked in the HA code and couldn't find anything. `OC_PythonLong` inherits from `int` so it seems like it shouldn't cause any problem (i.e. not really a Bleak problem). But adding the hacks you suggested is probably fine.

bdraco, 2023-07-25:
Cython 3 is more strict on the types. We can likely relax it where the cython check is happening

oyvindwe, 2023-07-25:
Thanks @bdraco !

closed 2023-07-25 by oyvindwe