hbldh/bleak #1380: Pairing Issues w/ the BlueZ Backend

hosse005, 2023-07-27:
* bleak version: 0.20.0, 0.21.0a1 (latest dev)
* Python version: 3.10.6
* Operating System: Ubuntu 22.04 (kernel v5.19)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description

Unable to pair Bose LE Sport Earbuds w/ the bleak client. Have no issues connecting or reading characteristics w/ the bleak client, just not able to pair. Also have tried pairing some peripheral samples from Nordic on an nrf5340 dev kit with no luck. However, I am able to pair these devices using bluetoothctl w/o any issues.

### What I Did

Below is my test script for attempting to pair a device:
```python
import asyncio
from bleak import BleakClient

bose_buds_addr = _DEVICE_MAC_ADDR_

async def main(address):
    async with BleakClient(address) as client:
        print("Connected: {0}".format(client.is_connected))
        await asyncio.sleep(5)
        await client.pair()
        print(f"Successfully paired with {address}") 

asyncio.run(main(bose_buds_addr))
```

Corresponding script output:
```
Connected: True
Traceback (most recent call last):
  File "/home/hosse005/public-repos/bleak/examples/pairing_test.py", line 40, in <module>
    asyncio.run(main(bose_buds_addr))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/hosse005/public-repos/bleak/examples/pairing_test.py", line 23, in main
    await client.pair()
  File "/home/hosse005/public-repos/bleak/bleak/__init__.py", line 587, in pair
    return await self._backend.pair(*args, **kwargs)
  File "/home/hosse005/public-repos/bleak/bleak/backends/bluezdbus/client.py", line 470, in pair
    assert_reply(reply)
  File "/home/hosse005/public-repos/bleak/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed
```

### Logs

Below is a relevant dbus log captured running the following command after client.is_connected returns True.
`$ sudo busctl monitor org.bluez`

```
‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=3  Timestamp="Thu 2023-07-27 13:54:15.705375 UTC"
  Sender=:1.568  Destination=org.bluez  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.freedesktop.DBus.Properties  Member=Get
  UniqueName=:1.568
  MESSAGE "ss" {
          STRING "org.bluez.Device1";
          STRING "Paired";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1 Cookie=19644  ReplyCookie=3  Timestamp="Thu 2023-07-27 13:54:15.705942 UTC"
  Sender=:1.15  Destination=:1.568
  UniqueName=:1.15
  MESSAGE "v" {
          VARIANT "b" {
                  BOOLEAN false;
          };
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=4  Timestamp="Thu 2023-07-27 13:54:15.706756 UTC"
  Sender=:1.568  Destination=org.bluez  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.freedesktop.DBus.Properties  Member=Set
  UniqueName=:1.568
  MESSAGE "ssv" {
          STRING "org.bluez.Device1";
          STRING "Trusted";
          VARIANT "b" {
                  BOOLEAN true;
          };
  };

‣ Type=signal  Endian=l  Flags=1  Version=1 Cookie=19645  Timestamp="Thu 2023-07-27 13:54:15.707766 UTC"
  Sender=:1.15  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.freedesktop.DBus.Properties  Member=PropertiesChanged
  UniqueName=:1.15
  MESSAGE "sa{sv}as" {
          STRING "org.bluez.Device1";
          ARRAY "{sv}" {
                  DICT_ENTRY "sv" {
                          STRING "Trusted";
                          VARIANT "b" {
                                  BOOLEAN true;
                          };
                  };
          };
          ARRAY "s" {
          };
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1 Cookie=19646  ReplyCookie=4  Timestamp="Thu 2023-07-27 13:54:15.707942 UTC"
  Sender=:1.15  Destination=:1.568
  UniqueName=:1.15
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=5  Timestamp="Thu 2023-07-27 13:54:15.708792 UTC"
  Sender=:1.568  Destination=org.bluez  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.bluez.Device1  Member=Pair
  UniqueName=:1.568
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=19647  Timestamp="Thu 2023-07-27 13:54:15.715021 UTC"
  Sender=:1.15  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=AddMatch
  UniqueName=:1.15
  MESSAGE "s" {
          STRING "type='signal',sender='org.freedesktop.DBus',path='/org/freedesktop/DBus',interface='org.freedesktop.DBus',member='NameOwnerChanged',arg0=':1.568'";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1 Cookie=395  ReplyCookie=19647  Timestamp="Thu 2023-07-27 13:54:15.715100 UTC"
  Sender=org.freedesktop.DBus  Destination=:1.15
  MESSAGE "" {
  };

‣ Type=error  Endian=l  Flags=1  Version=1 Cookie=19648  ReplyCookie=5  Timestamp="Thu 2023-07-27 13:54:15.857615 UTC"
  Sender=:1.15  Destination=:1.568
  ErrorName=org.bluez.Error.AuthenticationFailed  ErrorMessage="Authentication Failed"
  UniqueName=:1.15
  MESSAGE "s" {
          STRING "Authentication Failed";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=19649  Timestamp="Thu 2023-07-27 13:54:15.857729 UTC"
  Sender=:1.15  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=RemoveMatch
  UniqueName=:1.15
  MESSAGE "s" {
          STRING "type='signal',sender='org.freedesktop.DBus',path='/org/freedesktop/DBus',interface='org.freedesktop.DBus',member='NameOwnerChanged',arg0=':1.568'";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1 Cookie=396  ReplyCookie=19649  Timestamp="Thu 2023-07-27 13:54:15.857810 UTC"
  Sender=org.freedesktop.DBus  Destination=:1.15
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=6  Timestamp="Thu 2023-07-27 13:54:15.858677 UTC"
  Sender=:1.568  Destination=org.bluez  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.bluez.Device1  Member=Disconnect
  UniqueName=:1.568
  MESSAGE "" {
  };

‣ Type=signal  Endian=l  Flags=1  Version=1 Cookie=19650  Timestamp="Thu 2023-07-27 13:54:18.617551 UTC"
  Sender=:1.15  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.freedesktop.DBus.Properties  Member=PropertiesChanged
  UniqueName=:1.15
  MESSAGE "sa{sv}as" {
          STRING "org.bluez.Device1";
          ARRAY "{sv}" {
                  DICT_ENTRY "sv" {
                          STRING "ServicesResolved";
                          VARIANT "b" {
                                  BOOLEAN false;
                          };
                  };
          };
          ARRAY "s" {
          };
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1 Cookie=19651  ReplyCookie=6  Timestamp="Thu 2023-07-27 13:54:18.617727 UTC"
  Sender=:1.15  Destination=:1.568
  UniqueName=:1.15
  MESSAGE "" {
  };

‣ Type=signal  Endian=l  Flags=1  Version=1 Cookie=19652  Timestamp="Thu 2023-07-27 13:54:18.618224 UTC"
  Sender=:1.15  Path=/org/bluez/hci0/dev_MAC_ADDR  Interface=org.freedesktop.DBus.Properties  Member=PropertiesChanged
  UniqueName=:1.15
  MESSAGE "sa{sv}as" {
          STRING "org.bluez.Device1";
          ARRAY "{sv}" {
                  DICT_ENTRY "sv" {
                          STRING "Connected";
                          VARIANT "b" {
                                  BOOLEAN false;
                          };
                  };
          };
          ARRAY "s" {
          };
  };
```

dlech, 2023-07-28:
Pairing is still a work in progress in #1100.

hosse005, 2023-10-16:
I was able to workaround my particular issue by first registering the "NoInputNoOutput" authentication agent as outlined in the following doc: https://technotes.kynetics.com/2018/pairing_agents_bluez/.