hbldh/bleak #1384: write_gatt_char causes future read_gatt_char requests to fail (Code=7, The offset is invalid.)

blindman2k, 2023-08-06:
* bleak version: v0.20.2
* Python version: Python 3.11.4
* Operating System: MacOS Ventura 13.3.1 (22E261)
* BlueZ version (`bluetoothctl -v`) in case of Linux: ?

### Description

I am working on a custom BLE device (built using the Electric Imp 006 module). I have created a GATT profile that I can read from without problem. I have been working on a read-then-write pattern and it worked for a while ... then suddenly stopped. I am not sure what I did to break it but I am hoping for some help. I am getting the following error:

```
bleak.exc.BleakError: Failed to read characteristic 41: Error Domain=CBATTErrorDomain Code=7 "The offset is invalid." UserInfo={NSLocalizedDescription=The offset is invalid.}
```

### What I Did

Attached is the code and the logs.
[debug.py](https://github.com/hbldh/bleak/files/12269680/debug.txt)
[debug.log](https://github.com/hbldh/bleak/files/12269678/debug.log)

```

### Logs

My service and characteristics as bleak sees them are:
```
INFO:root:MTU Size: 185
INFO:root:Service: 0000180a-0000-1000-8000-00805f9b34fb (Handle: 1): Device Information
INFO:root: Characteristic: 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read)
INFO:root: Characteristic: 00002a05-0000-1000-8000-00805f9b34fb (Handle: 4): Service Changed (read)
INFO:root: Characteristic: 00002a23-0000-1000-8000-00805f9b34fb (Handle: 6): System ID (read)
INFO:root: Characteristic: 00002a24-0000-1000-8000-00805f9b34fb (Handle: 8): Model Number String (read)
INFO:root: Characteristic: 00002a25-0000-1000-8000-00805f9b34fb (Handle: 10): Serial Number String (read)
INFO:root: Characteristic: 00002a26-0000-1000-8000-00805f9b34fb (Handle: 12): Firmware Revision String (read)
INFO:root: Characteristic: 00002a28-0000-1000-8000-00805f9b34fb (Handle: 14): Software Revision String (read)
INFO:root: Characteristic: 00002a29-0000-1000-8000-00805f9b34fb (Handle: 16): Manufacturer Name String (read)
INFO:root:Service: 0000180f-0000-1000-8000-00805f9b34fb (Handle: 18): Battery Service
INFO:root: Characteristic: 00002a19-0000-1000-8000-00805f9b34fb (Handle: 19): Battery Level (read)
INFO:root:Service: fada47be-c455-48c9-a5f2-af7cf368d719 (Handle: 21): Unknown
INFO:root: Characteristic: 00001956-32d3-47c6-81a6-a7e59f18dac0 (Handle: 22): Unknown (read)
INFO:root: Characteristic: 01001956-32d3-47c6-81a6-a7e59f18dac0 (Handle: 24): Unknown (read,write)
INFO:root: Characteristic: 02004ab9-4756-4528-aa3a-799a4fd11117 (Handle: 26): Unknown (write)
INFO:root: Characteristic: 0300c342-8a8a-4544-ac42-08c841737b1b (Handle: 28): Unknown (write)
INFO:root: Characteristic: 04001956-32d3-47c6-81a6-a7e59f18dac0 (Handle: 30): Unknown (read)
INFO:root:Service: fa2b6a7f-fc3f-4628-9bd8-84fec29181df (Handle: 32): Unknown
INFO:root: Characteristic: 00001956-32d3-47c6-81a6-a7e59f18dac1 (Handle: 33): Unknown (read)
INFO:root: Characteristic: f3829f8a-5854-45fa-988c-52e7244dfe94 (Handle: 35): Unknown (write)
INFO:root: Characteristic: 02001956-32d3-47c6-81a6-a7e59f18dac0 (Handle: 37): Unknown (write)
INFO:root: Characteristic: 03001956-32d3-47c6-81a6-a7e59f18dac1 (Handle: 39): Unknown (read)
INFO:root: Characteristic: 04001956-32d3-47c6-81a6-a7e59f18dac1 (Handle: 41): Unknown (read)
INFO:root: Characteristic: 05001956-32d3-47c6-81a6-a7e59f18dac1 (Handle: 43): Unknown (write)
```

This is a rough step by step of what's happening in the code, leaving out some detail that can be seen in the code itself. When I execute a read_gatt_char() ...
```
response_char = "04001956-32d3-47c6-81a6-a7e59f18dac1"
await self.client.read_gatt_char(response_char)
```
I get the correct value (a blank response).
```
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Read characteristic value
DEBUG:bleak.backends.corebluetooth.client:Read Characteristic 04001956-32d3-47c6-81a6-a7e59f18dac1 : bytearray(b'')
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didWriteValueForCharacteristic_error_
```
Then I write to a different characteristic:
```
command_char = "02001956-32d3-47c6-81a6-a7e59f18dac0"
command = bytes('{"command": "status"}', "ascii")
await self.client.write_gatt_char(command_char, command, response=True);
```
This arrives on the device, all good.
```
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Write Characteristic Value
DEBUG:bleak.backends.corebluetooth.client:Write Characteristic 02001956-32d3-47c6-81a6-a7e59f18dac0 : b'{"command": "status"}'
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didUpdateValueForCharacteristic_error_
```
Finally, I read the original characteristic to see the result (it's actually in a loop).
```
response_char = "04001956-32d3-47c6-81a6-a7e59f18dac1"
await self.client.read_gatt_char(response_char)
```
This results in an error.
```
DeviceID: 600a0c2a69e83f12
DeviceOS: 69dbb48 - release-46.5 - Mon Jan  9 11:39:22 2023 - production
Traceback (most recent call last):
  File "/Users/aronsteg/Documents/Hark Medical/BLE Explorer/./debug.py", line 94, in <module>
    explorer = BLE_Explorer()
               ^^^^^^^^^^^^^^
  File "/Users/aronsteg/Documents/Hark Medical/BLE Explorer/./debug.py", line 21, in __init__
    asyncio.run(self.run_ble_scanner())
  File "/usr/local/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/aronsteg/Documents/Hark Medical/BLE Explorer/./debug.py", line 57, in run_ble_scanner
    print("Status: " + await self.chr_write_read(command_async, command_response, '{ "command": "status" }'))
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aronsteg/Documents/Hark Medical/BLE Explorer/./debug.py", line 80, in chr_write_read
    response = (await self.client.read_gatt_char(response_char)).decode()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/__init__.py", line 637, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/corebluetooth/client.py", line 272, in read_gatt_char
    output = await self._delegate.read_characteristic(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 155, in read_characteristic
    return await future
           ^^^^^^^^^^^^
bleak.exc.BleakError: Failed to read characteristic 41: Error Domain=CBATTErrorDomain Code=7 "The offset is invalid." UserInfo={NSLocalizedDescription=The offset is invalid.}
```

blindman2k, 2023-08-06:
It's 1am and I have worked out the cause. If I send packets greater than about 200 bytes, this happens. Under 200 bytes and it doesn't. I don't know why it seems to corrupt the bleak data but it does.

dlech, 2023-08-07:
If you log Bluetooth packets, you will see the MTU exchange. Both the central and the peripheral independently send their max allowable MTU and (should) use the smaller of the two values.

When you write an attribute value larger than the MTU size, the central will split up the data using multiple long writes that contain an offset to be able to tell how to reassemble the data. It sounds like your peripheral is not configured to handled this properly currently.

blindman2k, 2023-08-08:
That's good info. I can't fix the bluetooth stack on the Imp but I can report a bug. I have it working reliably (and slowly) at 185 bytes. It will have to do for now. I appreciate the feedback!

blindman2k, 2023-08-08:
I'll close this as it sounds like an Imp problem not a Bleak problem.

closed 2023-08-08 by blindman2k