hbldh/bleak #1385: Blank UUIDs list while using BlueZ in a Raspberry Pi 4

AVAM16, 2023-08-07:
* bleak version: 0.20.2
* Python version: 3.11.4 32-bit
* Operating System: Linux (Debian)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

The list of UUIDs in the device.details is blank even tough the UUID for a service exists. A few days ago it was working.
I am using a raspberry pi4. It is reading data from an ESP32.

### What I Did

devices = await bleak.BleakScanner.discover(timeout=5)
for d in devices:
    print(d.details)

dlech, 2023-08-07:
Tip: `details` is an internal OS-specific implementation and may change without notice. Use `BleakScanner.discover(..., return_adv=True)` instead.

Without logs, we can't see if the device is just not sending the UUID in the advertising data or if this is actually a problem with Bleak.

AVAM16, 2023-08-08:
[image]
I used return_adv = True and no service_uuids appeared. Just the local_name and the rssi.

dlech, 2023-08-08:
Again, without logs, we cannot know what is actually happening.

AVAM16, 2023-08-09:
{'path': '/org/bluez/hci0/dev_78_21_84_9A_BA_F2', 'props': {'Address': '78:21:84:9A:BA:F2', 'AddressType': 'public', 'Name': 'NimBLE-Server', 'Alias': 'NimBLE-Server', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -45, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}}

closed 2025-09-28 by github-actions[bot]