hbldh/bleak #1386: Overwhelming amounts of Notifications - any way to handle that?

shrewdlogarithm, 2023-08-10:
* bleak version: 0.20.2
* Python version: 3.11.0
* Operating System: Windows 10,11 - but also seen on Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I have a start-notify listening to a Bluetooth multimeter something like this

  async with BleakClient(d.address) as client:
    while client.is_connected:
      await client.start_notify(MODEL_NBR_UUID,handle)                                
    print("BLE Disconnected")      
    
It works well for the most part - until it runs on slightly less powerful hardware (say an Intel 6th gen laptop) where it seems to 'fall behind' the actual data being sent.

I've removed ALL processing - "handle" just returns without doing anything - still seems to fall behind (the results you see are from seconds and then even MINUTES earlier)

If you disconnect the device, data stops being received instantly and Python then hangs - all threads stop responding e.g. the code never reaches "BLE Disconnected" (there's a Flask thread which also stops responding)

My guess is the meter is sending WAY too much data but I can't control that - the only thought I have is to disconnect and reconnect constantly but that's a slow process/not really ideal - thoughts?

Is there a way to tell the notification to throw-away pending messages or clear it's queue or???

dlech, 2023-08-10:
`client.start_notify()` should only be called once, not in an infinite loop!

shrewdlogarithm, 2023-08-10:
I made a hash of hacking-out the code - I've updated the question with a full example (which obviously requires the multimeter to work!)

The issue still exists of course - it seems the meter can send upto 10 messages a second which seems rather excessive but I can't control that.

Unfortunately - and maybe it's the bluetooth hardware or the "power" of this older laptop - but it takes more than 1/10th of a second to process each notification (and I'm doing NOTHING with it) and so it all goes a bit wrong...

shrewdlogarithm, 2023-08-10:
After quite a bit of fiddling, it really seems this is down to how the Bluetooth chip inside this laptop works (It's a Lenovo Yoga 900)

I tried this on a really low-spec Acer laptop and it doesn't have QUITE the same issue - it's definately slower than on my desktop/other laptop but it doesn't fall WAY behind and hadn't crashed after 30 mins which I think is long enough...

Adding a super-cheap Bluetooth dongle to the Yoga also made it work better - I'll close this and hopefully if someone else has a similar issue they'll find this here...

Thanks for the timely help!!

closed 2023-08-10 by shrewdlogarithm