hbldh/bleak #1387: I started working on BLE now, I wrote a simple code to try something. I got this error and couldn't find any solutions for it.

sstprk, 2023-08-11:
That's my code

```python
async def main():
    address = "D4:22:CD:00:03:9B"
    async with BleakClient(address) as client:
        svcs = await client.get_services()
        print("Services:")
        company = svcs.get_characteristic("15170x0886-4947-11E9-8646-D663BD873D93")
        print(company)
if __name__ == "__main__":
    asyncio.run(main())
```
And the error;
```
yield self  # This tells Task to wait for completion.
pythonapi.PyErr_SetFromWindowsErr(error_code)
OSError: [WinError -2147023673] The operation was canceled by the user
```

_Originally posted by @sstprk in https://github.com/hbldh/bleak/discussions/1243#discussioncomment-6702134_

dlech, 2023-08-11:
Which version of Bleak are you using? Is it still a problem if you use the `develop` branch? Can you share any logs?

sstprk, 2023-08-11:
I'm using bleak 0.20.2
I can't share any logs for now, but I will as soon as possible

sstprk, 2023-08-21:
Python version : 3.11.4
bleak version : 0.20.2

The code;
```
import asyncio
from bleak import BleakClient, BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData
import logging

logging.basicConfig(filename = 'app.log', level = logging.INFO)
logger = logging.getLogger(__name__)

async def main():
    try:
        logger.info("Scanning..")
        devices = await BleakScanner.discover()
        logger.info("%d devices found", len(devices))
        logger.info("Printing devices..")
        for dev in devices:
            print(dev)
        try:
            address = "D4:22:CD:00:03:9B"
            logging.info("Starting process")
            async with BleakClient(address) as client:
                logger.info("Getting client services..")
                svcs = await client.read_gatt_char("15170x0886-4947-11E9-8646-D663BD873D93")
                logger.info("Got %d services", len(svcs))
                logger.info("Printing services..")
            print("Services:")
            for service in svcs:
                print(service)
        except:
            logger.exception("Can't start this section")
    except:
        logger.exception("Func has a problem")

asyncio.run(main())
```

Logs, if I did it right;
```
INFO:__main__:Scanning..
INFO:__main__:13 devices found
INFO:__main__:Printing devices..
INFO:root:Starting process
ERROR:__main__:Can't get services
Traceback (most recent call last):
  File "c:\Users\Salih\Desktop\Scripts\movella-dot-main\scanner.py", line 21, in main
    async with BleakClient(address) as client:
  File "C:\Users\Salih\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\__init__.py", line 491, in __aenter__
    await self.connect()
  File "C:\Users\Salih\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Salih\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 440, in connect
    self.services = await self.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Salih\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 724, in get_services
    await FutureLike(characteristic.get_descriptors_async(*args)),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Salih\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 1102, in __await__
    yield self  # This tells Task to wait for completion.
    ^^^^^^^^^^
  File "C:\Users\Salih\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 1049, in result
    pythonapi.PyErr_SetFromWindowsErr(error_code)
OSError: [WinError -2147023673] The operation was canceled by the user
```

I tried to solve it by trying everything I could find, including `develop` branch and couldn't solve it. If I didn't miss anything, there is not any solution I can find.
Thanks in advance.

pbackx, 2023-09-19:
I have a similar error when trying to 'start_notify' when connected and paired to a GoPro Hero 9. However, it doesn't occur all the time. It usually works the first time, next fails a few times  and eventually works again.

If there is any debugging I can do, please let me know what and how.

pbackx, 2023-09-19:
I think I found a possible workaround. I discovered that there are specific `winrt` options that you can add to the client. The `use_cached_services` option seems to fix or at least greatly improve the situation:

    client = BleakClient(device, winrt={"use_cached_services": False})

When this is added, the connect method will sometimes fail with `Could not get GATT services: Unreachable`

However, this is recoverable. And if the connection is retried it will work eventually (I have seen it fail a few times in a row, but usually it succeeds on the second try).

dlech, 2025-12-28:
A lot has changed in the winrt backend since this was reported. If this is still reproducible, we can reopen or start a new issue.