hbldh/bleak #1388: Need help - TIMETOUT ERROR

NeverWe1Come, 2023-08-14:
* bleak version:  0.20.2
* Python version: 3.11.4
* Operating System: Windows 11
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I am trying to obtain all services, characteristics and descriptors, for all ble devices i found near.

### What I Did

```
import argparse
import asyncio
import logging
from bleak import BleakClient, BleakScanner, BleakError

logger = logging.getLogger(__name__)

async def process_device(device, args):
    try:
        logger.info("connecting to device %s...", device)

        async with BleakClient(
            device,
            services=args.services,
        ) as client:
            try:
                logger.info("connected to device %s", device)

                for service in client.services:
                    logger.info("[Service] %s", service)

                    for char in service.characteristics:
                        if "read" in char.properties:
                            try:
                                value = await client.read_gatt_char(char.uuid)
                                logger.info(
                                    "  [Characteristic] %s (%s), Value: %r",
                                    char,
                                    ",".join(char.properties),
                                    value,
                                )
                            except Exception as e:
                                logger.error(
                                    "  [Characteristic] %s (%s), Error: %s",
                                    char,
                                    ",".join(char.properties),
                                    e,
                                )
                        else:
                            logger.info(
                                "  [Characteristic] %s (%s)", char, ",".join(char.properties)
                            )

                        for descriptor in char.descriptors:
                            try:
                                value = await client.read_gatt_descriptor(descriptor.handle)
                                logger.info("    [Descriptor] %s, Value: %r", descriptor, value)
                            except Exception as e:
                                logger.error("    [Descriptor] %s, Error: %s", descriptor, e)
                logger.info("disconnecting from device %s", device)
            except (asyncio.CancelledError, TimeoutError):
                logger.warning("Connection or timeout error for device %s", device)
            except Exception as e:
                logger.error("Error during device processing: %s", e)
    except BleakError as e:
        logger.error("Error while processing device %s: %s", device, e)

async def main(args: argparse.Namespace):
[... 244 lines cut ...]
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\src\BLEScanDetect\pruabas_vuln.py", line 67, in main
    await process_device(device, args)
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\src\BLEScanDetect\pruabas_vuln.py", line 13, in process_device
    async with BleakClient(
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\__init__.py", line 491, in __aenter__
    await self.connect()
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\backends\winrt\client.py", line 396, in connect
    async with async_timeout(timeout):
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

I don't have wireshark packet

NeverWe1Come, 2023-08-14:
Hi, thank you very much for the quick response, uninstall the version of bleak I have in use and install the version of the develop branch, but the failure continues.

Logs:
2023-08-14 20:28:44,771 __main__ INFO: starting scan...
2023-08-14 20:28:49,813 __main__ INFO: discovered 11 devices
2023-08-14 20:28:49,813 __main__ INFO: connecting to device D0:03:DF:E9:57:A9: None...
Traceback (most recent call last):
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\backends\winrt\client.py", line 435, in connect
    self.services = await self.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\backends\winrt\client.py", line 663, in get_services
    await FutureLike(self._requester.get_gatt_services_async(*srv_args)),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\backends\winrt\client.py", line 1072, in __await__
    yield self  # This tells Task to wait for completion.
    ^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\backends\winrt\client.py", line 1012, in result
    raise asyncio.CancelledError
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\src\BLEScanDetect\pruabas_vuln.py", line 94, in <module>
    asyncio.run(main(args))
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete  
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\src\BLEScanDetect\pruabas_vuln.py", line 67, in main
    await process_device(device, args)
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\src\BLEScanDetect\pruabas_vuln.py", line 13, in process_device
    async with BleakClient(
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\__init__.py", line 523, in __aenter__
    await self.connect()
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\__init__.py", line 563, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vgarc\Desktop\TFM\ProgramaEscritorio\BLEScanDetect-Desktop\venv\Lib\site-packages\bleak\bleak\backends\winrt\client.py", line 391, in connect
    async with async_timeout(timeout):
  File "C:\Users\vgarc\AppData\Local\Programs\Python\Python311\Lib\asyncio\timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

dlech, 2023-08-14:
We would need wireshark logs for further insight. For example, it could be the same issue as #1359.

NeverWe1Come, 2023-08-14:
Yeah i have read that issue too, but the solve didnt work for me. You know any way to captura that problem or exception, to avoid that the rest of the execution finishes?

dlech, 2023-08-14:
I would just make the timeout shorter or add an advertisement filter to avoid trying to connect to the problematic device in the first place. You could try to put each BleakClient in a separate task so they can run in parallel, but you would probably overload the Bluetooth adapter by trying to do too many things at the same time and cause other errors.

NeverWe1Come, 2023-08-14:
I have tried to get a wireshark capture of bluettoth interface, but it doesn't capture anything, so i can't give more info than my code and the problem. Talking about your avises, first advise, i think it doesn't work, but i will try, and the filter, it's not possible because i don't think its only one device (I don't know exactly but i don't think so). For the second advise, thats not good for my project, the code i hace post here is a small example, my project have another concurrent scanner so its complicated trying not to overload the Buetooth interface. THank you very much for your advises

NeverWe1Come, 2023-08-14:
Also i have another question. i don't understand why i scan the enviroment looking for   BLE Devices, but when the scanner finished and try to connect to some devices, the responses are things like this:

2023-08-14 22:47:36,535 __main__ INFO: connecting to device D4:9D:C0:3F:9C:CF: None...
2023-08-14 22:47:36,546 __main__ ERROR: Error while processing device D4:9D:C0:3F:9C:CF: None: Device with address D4:9D:C0:3F:9C:CF was not found.

How its possible?

dlech, 2023-08-14:
It was too long between the scan and the connect and the device went to sleep or something like that.

dlech, 2024-05-04:
Is this still a problem with Bleak v0.22.0?

closed 2025-05-16 by dlech