hbldh/bleak #1389: BleakScanner.discover repeating error/stack-trace

vincintz, 2023-08-15:
* bleak version: 0.20.2
* Python version: 3.8.10
* Operating System: Ubuntu 20.04.4 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description

Calling `BleakScanner.discover`, succeeds on almost all environment I run it. But fails on one particular machine.
The logs point to some weird infinite exception look happening in a call to dbus-fast.

### What I Did

Here's the code to reproduct the problem.
```
import asyncio
from bleak import BleakScanner

async def main():
    print("scanning for 5 seconds, please wait...")
    devices = await BleakScanner.discover(timeout=5.0, return_adv=True)
    print(devices)

if __name__ == "__main__":
    asyncio.run(main())
```

As mentioned, so far I've seen it happen only on one specific machine. But I'd like to figure out what's different about that, and how to fix it in case it fails in prod.

Rolling back to bleak v0.16.0 fixes the problem.
Earliest version I can reproduce this is v0.17.0

### Logs

This output is printed over and over, until I hit ctr-c.
```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src/dbus_fast/aio/message_reader.py", line 24, in dbus_fast.aio.message_reader.build_message_reader._message_reader
  File "src/dbus_fast/_private/unmarshaller.py", line 662, in dbus_fast._private.unmarshaller.Unmarshaller._unmarshall
  File "src/dbus_fast/_private/unmarshaller.py", line 536, in dbus_fast._private.unmarshaller.Unmarshaller._read_header
  File "src/dbus_fast/_private/unmarshaller.py", line 284, in dbus_fast._private.unmarshaller.Unmarshaller._read_to_pos
  File "src/dbus_fast/_private/unmarshaller.py", line 253, in dbus_fast._private.unmarshaller.Unmarshaller._read_sock
AttributeError: 'cython_function_or_method' object has no attribute 'recvmsg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "src/dbus_fast/aio/message_reader.py", line 35, in dbus_fast.aio.message_reader.build_message_reader._message_reader
TypeError: 'bool' object is not callable
Exception in callback build_message_reader.<locals>._message_reader()
handle: <Handle build_message_reader.<locals>._message_reader()>
Traceback (most recent call last):
  File "src/dbus_fast/aio/message_reader.py", line 24, in dbus_fast.aio.message_reader.build_message_reader._message_reader
  File "src/dbus_fast/_private/unmarshaller.py", line 662, in dbus_fast._private.unmarshaller.Unmarshaller._unmarshall
  File "src/dbus_fast/_private/unmarshaller.py", line 536, in dbus_fast._private.unmarshaller.Unmarshaller._read_header
  File "src/dbus_fast/_private/unmarshaller.py", line 284, in dbus_fast._private.unmarshaller.Unmarshaller._read_to_pos
  File "src/dbus_fast/_private/unmarshaller.py", line 253, in dbus_fast._private.unmarshaller.Unmarshaller._read_sock
AttributeError: 'cython_function_or_method' object has no attribute 'recvmsg'

During handling of the above exception, another exception occurred:
```

dlech, 2023-08-15:
This looks like an error in `dbus-fast`, so I would suggest reporting the error there. https://github.com/Bluetooth-Devices/dbus-fast

vincintz, 2023-08-16:
Thanks for the quick response. I raised this [issue](https://github.com/Bluetooth-Devices/dbus-fast/issues/224) in dbus-fast

bdraco, 2023-08-17:
Issue can be closed. Was a local install issue

closed 2023-08-17 by dlech