hbldh/bleak #1401: ESP32 BLE Characteristic Write Issue: 'Attribute Not Found (0x0A)' Error Despite Valid Handle

dejimarquis, 2023-08-26:
* bleak version: 0.20.2
* Python version: 3.10.6
* Operating System: Mac OS Ventura 13.5
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I have an Arduino ESP32 device that I am trying to write to trigger a vibration. When I try to write to the characteristic I get no action. On looking at the packet logs using PacketLogger, I get an error `Error Response - Attribute Handle: 0x0005 - Error Code: Attribute Not Found (0x0A)`. However, I see the characteristic exist with the write attribute, see below code. I would appreciate any pointers to address this.

_Side note: When I use a different BLE library in C#, the functionality works as expected_

### What I Did

Code I am executing
```
....
    async with BleakClient(device) as client:
        if not client.is_connected:
            print("Failed to connect")
            return

        services = await client.get_services()
        for service in services:
            print(service.uuid)
            for char in service.characteristics:
                print(f"  - {char.uuid} (Handle: {char.handle}) | Properties: {char.properties}")

        pattern = "1" + VibrationPattern.STRONG_BUZZ_100
        await client.write_gatt_char("d9a9f9c6-f8db-4f3b-b6b3-6eae8f3c4d7b", pattern.encode())

        await client.disconnect()
```

Output

```
a9e90000-194c-4523-a473-5fdf36aa4d20
  - bcb6f4e4-4fe4-4f06-890e-2b2d7f9a8bca (Handle: 41) | Properties: ['read', 'write', 'notify']
  - e0f8f717-b67a-4e63-a1e7-8b1a8d7a9ac2 (Handle: 43) | Properties: ['read', 'write', 'notify']
  **- d9a9f9c6-f8db-4f3b-b6b3-6eae8f3c4d7b (Handle: 45) | Properties: ['read', 'write', 'notify']**
```

### Logs

[image]

orcim, 2023-09-03:
pattern = "1" + VibrationPattern.STRONG_BUZZ_100
        await client.write_gatt_char("d9a9f9c6-f8db-4f3b-b6b3-6eae8f3c4d7b", pattern.encode())

try this:

		# BlueZ doesn't have a proper way to get the MTU, so we have this hack.
		# If this doesn't work for you, you can set the client._mtu_size attribute
		# to override the value instead.
		if client._backend.__class__.__name__ == "BleakClientBlueZDBus":
			await client._backend._acquire_mtu()

		print("MTU:", client.mtu_size)

		# Write without response is limited to MTU - 3 bytes

		# real data
		pattern = "1" + VibrationPattern.STRONG_BUZZ_100
		
		chunk_size = client.mtu_size - 3
		for chunk in (
			pattern[i : i + chunk_size] for i in range(0, len(pattern), chunk_size)
		):
			await client.write_gatt_char("d9a9f9c6-f8db-4f3b-b6b3-6eae8f3c4d7b", chunk.encode('ascii'), response=False)

dejimarquis, 2023-09-14:
> Also on Mac?
> 
> >

Yes, I am using this unity asset - http://www.shatalmic.com/home/bluetooth-le-unity-asset. For the asset, here are the logs. I see no errors.

[image]

Whereas when I use Bleak, I get the errors. One observation is that the handle number is different with the bleak client. **Maybe this could be the issue?**

[image]

> ```
>     pattern = "1" + VibrationPattern.STRONG_BUZZ_100
>     await client.write_gatt_char("d9a9f9c6-f8db-4f3b-b6b3-6eae8f3c4d7b", pattern.encode())
> ```
>
> try this:
>
> ```
> 	# BlueZ doesn't have a proper [...]

Unfortunately, the issue still persists when I try the above

dejimarquis, 2023-09-28:
Solved the problem by setting `response` to `True` --  ` await client.write_gatt_char("d9a9f9c6-f8db-4f3b-b6b3-6eae8f3c4d7b", pattern.encode(), response=True)`

Thanks all

closed 2023-09-28 by dejimarquis