hbldh/bleak #1402: Unreliable Reconnection to BLE Heart Rate Monitor

nullstalgia, 2023-08-28:
* bleak version: v0.20.0+ (develop branch commit `87ec6bd`)
* Python version: 3.11.1
* Operating System: Windows 10 22H2

### Description

I am writing a script that connects to generic BLE Heart Rate Monitors (HRM) and sends the data to VRChat over OSC.

The script functions normally most of the time (minus all of the random exceptions I have to handle), but issues can occur when I walk out of the room while wearing the HRM, and later re-enter.

A good portion of the time, it will re-connect successfully, but there is a chance that the script will get stuck in a loop of attempting to connect and failing. And during this loop, it will give the same ***CancelledError*** regardless of if the HRM is even _on_ anymore.

Restarting the script will often break this loop and allow the script to function again, but I really don't want to ask users to do that.

### What I Did

[Here is the full script in question](https://github.com/nullstalgia/PyHRPresence/blob/main/PyHRPresence.py), but below is my attempt at a Minimum Reproducible Example

```
import asyncio
import logging
from bleak import BleakScanner, BleakClient
from bleak.exc import BleakDeviceNotFoundError, BleakError
from bleak.backends.device import BLEDevice
from bleak.backends.characteristic import BleakGATTCharacteristic
from bleak.backends.scanner import AdvertisementData
from io import BytesIO
import time
from pprint import pformat, pprint
import struct
import sys
import traceback
from async_timeout import timeout

ble_address = "C8:45:A8:16:66:CC"

HEART_RATE_SERVICE_UUID = "0000180d-0000-1000-8000-00805f9b34fb"
HEART_RATE_MEASUREMENT_CHARACTERISTIC_UUID = "00002a37-0000-1000-8000-00805f9b34fb"

def read_hr_buffer(data):
    length = len(data)
    if length == 0:
        return None

    ms = BytesIO(data)
    flags = ms.read(1)[0]
    isshort = flags & 1
    contactSensor = (flags >> 1) & 3
    hasEnergyExpended = flags & 8
    hasRRInterval = flags & 16
    minLength = 3 if isshort else 2

    if len(data) < minLength:
        return None

    reading = {}
    reading['Flags'] = flags
    reading['Status'] = contactSensor
    reading['BeatsPerMinute'] = struct.unpack('<H', ms.read(2))[0] if isshort else ms.read(1)[0]

    if hasEnergyExpended:
        reading['EnergyExpended'] = struct.unpack('<H', ms.read(2))[0]

    if hasRRInterval:
        rrvalueCount = int((len(data) - ms.tell()) / 2)
        rrvalues = [struct.unpack('<H', ms.read(2))[0] for _ in range(rrvalueCount)]
        reading['RRIntervals'] = rrvalues

    return reading

[... 132 lines cut ...]
 '  File "asyncio\\locks.py", line 213, in wait\n'
 'asyncio.exceptions.CancelledError\n')
```

```
2023-08-27 23:14:10,962 - DEBUG - Exception: <class 'TimeoutError'> - 
2023-08-27 23:14:10,963 - DEBUG - <traceback object at 0x000001C7B779B380>
2023-08-27 23:14:10,964 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "bleak\\backends\\winrt\\client.py", line 435, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 687, in get_services\n'
 '  File "bleak\\backends\\winrt\\client.py", line 1072, in __await__\n'
 '  File "bleak\\backends\\winrt\\client.py", line 1007, in result\n'
 'asyncio.exceptions.CancelledError\n'
 '\n'
 'During handling of the above exception, another exception occurred:\n'
 '\n'
 'Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 563, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 391, in connect\n'
 '  File "asyncio\\timeouts.py", line 98, in __aexit__\n'
 'TimeoutError\n')
```

[polar-wireshark-logs.zip](https://github.com/hbldh/bleak/files/12450944/polar-wireshark-logs.zip)

dlech, 2023-08-28:
This sounds suspiciously like https://github.com/hbldh/bleak/issues/1359#issuecomment-1694902870. Does adding the delays suggested in the linked comment fix the issue for you?

nullstalgia, 2023-09-03:
Apologies for delay, wanted to run it through for a few days to see how it affected it.

I *think* it's able to reconnect a bit more reliably, I was able to exit the vicinity a few times and not notice any hiccups.

But a few times, either when re-entering the space *or sometimes, **after** I had returned*, the issue would rear it's head again.

```
2023-09-02 15:41:09,988 - DEBUG - Exception: <class 'asyncio.exceptions.CancelledError'> - 
2023-09-02 15:41:09,988 - DEBUG - <traceback object at 0x000001D221E5C780>
2023-09-02 15:41:09,989 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 563, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 443, in connect\n'
 '  File "asyncio\\locks.py", line 213, in wait\n'
 'asyncio.exceptions.CancelledError\n')
```

```
2023-08-30 00:03:18,546 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 563, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 237, in connect\n'
 '  File "bleak\\__init__.py", line 346, in find_device_by_address\n'
 '  File "bleak\\__init__.py", line 402, in find_device_by_filter\n'
 '  File "bleak\\__init__.py", line 239, in advertisement_data\n'
 '  File "asyncio\\queues.py", line 158, in get\n'
 'asyncio.exceptions.CancelledError\n')
```

But I have not tested with v.0.21.0 yet. That's next on my to-do.

nullstalgia, 2023-09-13:
@dlech So 0.21.0 seems more reliable? Hard to tell since its such a transient issue.

BPM returns 0 for a bit, eventually raises a BleakError, then starts raising CancelledErrors.

A sampling some of the log errors:

```
2023-09-10 00:55:59,036 - DEBUG - Exception: <class 'TimeoutError'> - 
2023-09-10 00:55:59,037 - DEBUG - <traceback object at 0x000002086098BF40>
2023-09-10 00:55:59,038 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "bleak\\backends\\winrt\\client.py", line 480, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 756, in get_services\n'
 '  File "bleak\\backends\\winrt\\client.py", line 1122, in __await__\n'
 '  File "bleak\\backends\\winrt\\client.py", line 1065, in result\n'
 'asyncio.exceptions.CancelledError\n'
 '\n'
 'During handling of the above exception, another exception occurred:\n'
 '\n'
 'Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 605, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 436, in connect\n'
 '  File "asyncio\\timeouts.py", line 98, in __aexit__\n'
 'TimeoutError\n')

2023-09-10 03:21:34,547 - DEBUG - Exception: <class 'asyncio.exceptions.CancelledError'> - 
2023-09-10 03:21:34,548 - DEBUG - <traceback object at 0x0000020860487540>
2023-09-10 03:21:34,548 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 605, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 282, in connect\n'
 '  File "bleak\\__init__.py", line 385, in find_device_by_address\n'
 '  File "bleak\\__init__.py", line 444, in find_device_by_filter\n'
 '  File "bleak\\__init__.py", line 244, in advertisement_data\n'
 '  File "asyncio\\queues.py", line 158, in get\n'
 'asyncio.exceptions.CancelledError\n')

2023-09-12 23:29:15,554 - DEBUG - Exception: <class 'asyncio.exceptions.CancelledError'> - 
2023-09-12 23:29:15,554 - DEBUG - <traceback object at 0x0000019C3707A1C0>
2023-09-12 23:29:15,555 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 605, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 488, in connect\n'
 '  File "asyncio\\locks.py", line 213, in wait\n'
 'asyncio.exceptions.CancelledError\n')
```

dlech, 2023-09-13:
The `asyncio.CancelledError`s are caused by the `async with timeout()`. Since the try/except is inside of the `async with` block, it is catching the `CancelledError` which would usually be masked by the `asyncio.TimeoutError`. I don't think there should be any `try/except` there.

But that still won't fix the Windows Bluetooth stack. We would need more detailed logs to see if there is anything we are leaving hanging (e.g. some closable object that we are not closing) that would be preventing things from working unless we restart the app.

nullstalgia, 2023-09-16:
@dlech Aight, had it happen with debug logging on:

```2023-09-16 00:20:39,760 - DEBUG - Exception: <class 'asyncio.exceptions.CancelledError'> - 
2023-09-16 00:20:39,760 - DEBUG - <traceback object at 0x000001B1F1858540>
2023-09-16 00:20:39,761 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 605, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 488, in connect\n'
 '  File "asyncio\\locks.py", line 213, in wait\n'
 'asyncio.exceptions.CancelledError\n')
2023-09-16 00:20:39,761 - ERROR - Couldn't connect to device, try again?
2023-09-16 00:20:39,762 - INFO - Attempting to reconnect...
2023-09-16 00:20:40,767 - INFO - [32mAttempting to connect...[39m
2023-09-16 00:20:40,768 - DEBUG - Created client.
2023-09-16 00:20:40,768 - DEBUG - Connecting.
2023-09-16 00:20:40,810 - DEBUG - Received 7A:AB:03:3A:4D:C8: .
2023-09-16 00:20:40,812 - DEBUG - Received 7A:AB:03:3A:4D:C8: .
2023-09-16 00:20:40,953 - DEBUG - Received 8C:79:F5:7F:8E:14: .
2023-09-16 00:20:40,955 - DEBUG - Received 8C:79:F5:7F:8E:14: [TV] Living Room TV (Q900).
2023-09-16 00:20:41,162 - DEBUG - Received EA:67:72:A1:49:84: LHB-CAAEDB12.
2023-09-16 00:20:41,163 - DEBUG - Received EA:67:72:A1:49:84: .
2023-09-16 00:20:41,165 - DEBUG - Received E2:C3:F9:D3:C3:3A: LHB-B95A8525.
2023-09-16 00:20:41,294 - DEBUG - Received E2:C3:F9:D3:C3:3A: LHB-B95A8525.
2023-09-16 00:20:41,295 - DEBUG - Received E2:C3:F9:D3:C3:3A: .
2023-09-16 00:20:41,420 - DEBUG - Received E2:C3:F9:D3:C3:3A: LHB-B95A8525.
2023-09-16 00:20:41,422 - DEBUG - Received E2:C3:F9:D3:C3:3A: .
2023-09-16 00:20:41,521 - DEBUG - Received C8:45:A8:16:66:CC: Polar H10 C1C93C2D.
2023-09-16 00:20:41,522 - DEBUG - 5 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
2023-09-16 00:20:41,523 - DEBUG - Connecting to BLE device @ C8:45:A8:16:66:CC
2023-09-16 00:20:41,621 - DEBUG - getting services (service_cache_mode=None, cache_mode=None)...
2023-09-16 00:20:41,874 - DEBUG - skipping service 0000180d-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:20:41,881 - DEBUG - skipping service 0000181c-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:20:41,946 - DEBUG - skipping service 6217ff4b-fb31-1140-ad5a-a45545d7ecf3 due to access denied
2023-09-16 00:20:41,955 - DEBUG - skipping service fb005c80-02e7-f387-1cad-8acd2d8df0c8 due to access denied
2023-09-16 00:20:41,961 - DEBUG - skipping service 0000feee-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:21:10,766 - DEBUG - closing requester
2023-09-16 00:21:10,767 - DEBUG - closing session
2023-09-16 00:21:10,769 - DEBUG - Exception: <class 'asyncio.exceptions.CancelledError'> - 
2023-09-16 00:21:10,769 - DEBUG - <traceback object at 0x000001B1F2BC6600>
2023-09-16 00:21:10,770 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 605, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 488, in connect\n'
 '  File "asyncio\\locks.py", line 213, in wait\n'
 'asyncio.exceptions.CancelledError\n')
2023-09-16 00:21:10,770 - ERROR - Couldn't connect to device, try again?
2023-09-16 00:21:10,770 - INFO - Attempting to reconnect...
2023-09-16 00:21:11,787 - INFO - [32mAttempting to connect...[39m
2023-09-16 00:21:11,790 - DEBUG - Created client.
2023-09-16 00:21:11,790 - DEBUG - Connecting.
2023-09-16 00:21:11,837 - DEBUG - Received C8:45:A8:16:66:CC: Polar H10 C1C93C2D.
2023-09-16 00:21:11,839 - DEBUG - 1 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
2023-09-16 00:21:11,839 - DEBUG - Connecting to BLE device @ C8:45:A8:16:66:CC
2023-09-16 00:21:11,969 - DEBUG - getting services (service_cache_mode=None, cache_mode=None)...
2023-09-16 00:21:12,279 - DEBUG - skipping service 0000180d-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:21:12,288 - DEBUG - skipping service 0000181c-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:21:12,357 - DEBUG - skipping service 6217ff4b-fb31-1140-ad5a-a45545d7ecf3 due to access denied
2023-09-16 00:21:12,364 - DEBUG - skipping service fb005c80-02e7-f387-1cad-8acd2d8df0c8 due to access denied
2023-09-16 00:21:12,371 - DEBUG - skipping service 0000feee-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:21:41,792 - DEBUG - closing requester
2023-09-16 00:21:41,793 - DEBUG - closing session
2023-09-16 00:21:41,794 - DEBUG - Exception: <class 'asyncio.exceptions.CancelledError'> - 
2023-09-16 00:21:41,794 - DEBUG - <traceback object at 0x000001B1F2BB3DC0>
2023-09-16 00:21:41,795 - DEBUG - ('Traceback (most recent call last):\n'
 '  File "PyHRPresence.py", line 344, in connect\n'
 '  File "bleak\\__init__.py", line 605, in connect\n'
 '  File "bleak\\backends\\winrt\\client.py", line 488, in connect\n'
 '  File "asyncio\\locks.py", line 213, in wait\n'
 'asyncio.exceptions.CancelledError\n')
2023-09-16 00:21:41,795 - ERROR - Couldn't connect to device, try again?
[... 18 lines cut ...]
2023-09-16 00:21:43,555 - DEBUG - Received 7A:AB:03:3A:4D:C8: .
2023-09-16 00:21:43,557 - DEBUG - Received 7A:AB:03:3A:4D:C8: .
2023-09-16 00:21:43,691 - DEBUG - Received EA:67:72:A1:49:84: LHB-CAAEDB12.
2023-09-16 00:21:43,692 - DEBUG - Received EA:67:72:A1:49:84: .
2023-09-16 00:21:43,695 - DEBUG - Received E2:C3:F9:D3:C3:3A: LHB-B95A8525.
[... cut ...]
2023-09-16 00:21:45,026 - DEBUG - skipping service 0000180d-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:21:45,033 - DEBUG - skipping service 0000181c-0000-1000-8000-00805f9b34fb due to access denied
2023-09-16 00:21:45,088 - DEBUG - skipping service 6217ff4b-fb31-1140-ad5a-a45545d7ecf3 due to access denied
2023-09-16 00:21:45,094 - DEBUG - skipping service fb005c80-02e7-f387-1cad-8acd2d8df0c8 due to access denied
2023-09-16 00:21:45,102 - DEBUG - skipping service 0000feee-0000-1000-8000-00805f9b34fb due to access denied
```

nullstalgia, 2023-09-27:
@dlech Is there more types of logging you would need? Not sure how to proceed from here.