hbldh/bleak #1412: linux TypeError: 'StructMeta' object is not iterable

wangguan1888, 2023-09-04:
* bleak version:0.21.0
* Python version:3.8.11
* Operating System:Linux Yocto
* BlueZ version (`bluetoothctl -v`) in case of Linux:5.49

### Description
Trying to run the example code results in TypeError: 'StructMeta' object is not iterable.

### What I Did
1.install bleak: pip3 install bleak
2.run the example code
```
import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())
```

3.Got the following result.
```
Traceback (most recent call last):
  File "test1.py", line 9, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "test1.py", line 5, in main
    devices = await BleakScanner.discover()
  File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 317, in discover
    async with cls(**kwargs) as scanner:
  File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 145, in __init__
    get_platform_scanner_backend_type() if backend is None else backend
  File "/usr/lib/python3.8/site-packages/bleak/backends/scanner.py", line 284, in get_platform_scanner_backend_type
    from bleak.backends.bluezdbus.scanner import BleakScannerBlueZDBus
  File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/scanner.py", line 5, in <module>
    from dbus_fast import Variant
  File "/usr/lib/python3.8/site-packages/dbus_fast/__init__.py", line 1, in <module>
    from . import aio, glib, introspection, message_bus, proxy_object, service
  File "/usr/lib/python3.8/site-packages/dbus_fast/glib/__init__.py", line 1, in <module>
    from .message_bus import MessageBus
  File "/usr/lib/python3.8/site-packages/dbus_fast/glib/message_bus.py", line 20, in <module>
    from .proxy_object import ProxyObject
  File "/usr/lib/python3.8/site-packages/dbus_fast/glib/proxy_object.py", line 15, in <module>
    from gi.repository import GLib
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "/usr/lib/python3.8/site-packages/gi/importer.py", line 146, in load_module
    dynamic_module = load_overrides(introspection_module)
  File "/usr/lib/python3.8/site-packages/gi/overrides/__init__.py", line 125, in load_overrides
    override_mod = importlib.import_module(override_package_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 86, in <module>
    class _VariantCreator(object):
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 89, in _VariantCreator
    'b': GLib.Variant.new_boolean,
  File "/usr/lib/python3.8/site-packages/gi/module.py", line 221, in __getattr__
    wrapper = metaclass(name, bases, dict_)
TypeError: 'StructMeta' object is not iterable
```

dlech, 2023-09-04:
Thanks for the report. I made a pull request to fix this if you want to test it.

wangguan1888, 2023-09-06:
> 

Thanks for making a pull request. I changed manager.py and scanner.py file which are in the path [/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus]
However, the same error was occured.
Could you please tell me what I should do?

dlech, 2023-09-06:
Can you share the new stack trace after this change?

wangguan1888, 2023-09-07:
> Can you share the new stack trace after this change?

Thanks for your reply. The new stack trace is here. It is same as the error before changing manager.py and scanner.py
`
Traceback (most recent call last):
  File "/home/nabtesco/test1.py", line 9, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/nabtesco/test1.py", line 5, in main
    devices = await BleakScanner.discover()
  File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 317, in discover
    async with cls(**kwargs) as scanner:
  File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 145, in __init__
    get_platform_scanner_backend_type() if backend is None else backend
  File "/usr/lib/python3.8/site-packages/bleak/backends/scanner.py", line 284, in get_platform_scanner_backend_type
    from bleak.backends.bluezdbus.scanner import BleakScannerBlueZDBus
  File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/scanner.py", line 5, in <module>
    from dbus_fast.service import Variant
  File "/usr/lib/python3.8/site-packages/dbus_fast/__init__.py", line 1, in <module>
    from . import aio, glib, introspection, message_bus, proxy_object, service
  File "/usr/lib/python3.8/site-packages/dbus_fast/glib/__init__.py", line 1, in <module>
    from .message_bus import MessageBus
  File "/usr/lib/python3.8/site-packages/dbus_fast/glib/message_bus.py", line 20, in <module>
    from .proxy_object import ProxyObject
  File "/usr/lib/python3.8/site-packages/dbus_fast/glib/proxy_object.py", line 15, in <module>
    from gi.repository import GLib
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "/usr/lib/python3.8/site-packages/gi/importer.py", line 146, in load_module
    dynamic_module = load_overrides(introspection_module)
  File "/usr/lib/python3.8/site-packages/gi/overrides/__init__.py", line 125, in load_overrides
    override_mod = importlib.import_module(override_package_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 86, in <module>
    class _VariantCreator(object):
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 89, in _VariantCreator
    'b': GLib.Variant.new_boolean,
  File "/usr/lib/python3.8/site-packages/gi/module.py", line 221, in __getattr__
    wrapper = metaclass(name, bases, dict_)
TypeError: 'StructMeta' object is not iterable
imx6qsabresd:/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus#
`

wangguan1888, 2023-09-07:
> Can you share the new stack trace after this change?

My linux kernel version is 4.9.88. Is it too old for bleak Library?

dlech, 2023-09-07:
No, I think the `dbus-fast` package needs to be fixed so that it doesn't import `gi` since we don't use it.

dlech, 2023-09-07:
The root of the problem is that the version of `gi` that is installed requires Python 3.9 but you are using 3.8.

dlech, 2023-09-08:
I made a new pull request with an updated version of `dbus-fast` that you can try.

bdraco, 2023-09-09:
Can you try upgrading to bleak 0.21.1 and dbus-fast 2.0.1?

PR #1417 (2023-09-08): v0.21.1
Changed
-------
* Changed ``dbus-fast`` dependency to include v2.x. Fixes #1412.
closed 2025-09-28 by github-actions[bot]