hbldh/bleak #1424: Missing telegrams and scanning stops randomly

micha-1987, 2023-09-22:
* bleak version: 0.20.2
* Python version: 3.11.5
* Operating System: Win10

### Description
Actually I have three issues which maybe related.

1. We are having an environment sensor device which is sending data like temperature, humidity, adv_count, etc every minute.
Unfortunately we are missing several environment telegrams.
Sometimes we receive one telegram after 4 minutes sometimes after 2 minutes.
So it seems to be quite random whether we scan in the right moment when the telegram is sended.
see first log

2. Additionally we are having another sensor which measuring the acceleration and sends every 50ms.
So I would assume to get 20 telegrams per second but we are only getting 1 or 2 per second.
see second log

3. The scanner sometimes stops detecting advertisement telegrams and after a couple of minutes it continues even though the sensor remains untouched.
Can it be the rssi filter when it remains the same value?
see third log

I read about scanning intervals and scanning windows which can be modified. (see [link](https://www.btframework.com/blescanparams.htm#:~:text=In%20BLE%20terminology%2C%20scan%20interval,range%20between%204%20and%2016384.)).
I also saw in other issues that it is OS dependent but can I change the scan interval and/or window in bleak?
I mean even the Windows settings should detect more than one telegram per second.
Or do you have any other suggestions?

Many thanks!

###Environment logger information
```
2023-09-22 11:25:37,239 logger   INFO: ENV frame received - MAC: DB:32:C3:D1:8A:FA - Raw data: ce:6c:02:1e:c5:cc:41:dc:05:c3:47:31:54:4d:42:64:bd:1c:46:00:00:00:12
2023-09-22 11:27:12,469 logger   INFO: ENV frame received - MAC: DB:32:C3:D1:8A:FA - Raw data: ce:6c:02:2e:c5:cc:41:dc:05:c3:47:31:54:4d:42:64:bd:1c:ab:00:00:12:12
2023-09-22 11:30:42,135 logger   INFO: ENV frame received - MAC: DB:32:C3:D1:8A:FA - Raw data: ce:6c:02:5e:c5:ff:41:13:05:43:57:78:84:4d:42:64:bd:1c:ab:00:00:12:12
```

###Acceleration logger information
```
2023-09-22 10:06:52,332 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: d1:6c:05:00:00:01:4f:00:7f:7b:f0:02:fe:2f:f0:17:f8:c0:00:0f:da:ff:00:ff:8b:f0:40
2023-09-22 10:06:53,394 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 89:6c:05:00:00:01:58:00:3f:7c:00:00:fe:6f:d0:07:f8:be:c0:1f:e2:fd:00:7f:9b:f0:40
2023-09-22 10:06:54,456 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: a8:6c:05:00:00:01:5f:00:bf:9b:f8:01:fd:2f:c0:07:f5:c0:40:2f:de:fd:00:7f:9b:f0:40
2023-09-22 10:06:55,517 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: ce:6c:05:00:00:01:61:00:3f:5b:f8:03:fd:6f:f0:0b:f7:bf:00:1f:de:fd:ff:ff:7b:e4:40
2023-09-22 10:06:56,605 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: f1:6c:05:00:00:01:66:00:ff:7b:ec:02:fd:6f:cf:fb:f7:be:40:3f:d6:fa:ff:ff:6b:f4:40
2023-09-22 10:06:59,231 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: a5:6c:05:00:00:01:68:01:bf:9b:fc:03:fd:af:f0:17:f6:bf:00:2f:d6:fc:01:7f:6b:ec:40
2023-09-22 10:07:03,855 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 83:6c:05:00:00:01:6f:00:7f:2b:e0:03:fd:6f:f0:17:f4:bf:00:5f:de:fe:01:7f:7c:08:40
```

###Start - continue logger information (stopping for ~3minutes)
```
2023-09-22 11:25:13,141 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: d8:6c:05:00:00:00:a8:ff:bf:43:f3:fc:fd:cf:cf:f3:f7:40:7f:bf:d4:fb:fe:ff:43:f4:40
2023-09-22 11:25:14,162 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 7a:6c:05:00:00:00:aa:fe:ff:63:e3:fa:fe:0f:ff:ef:f4:3f:3f:bf:e4:f9:ff:3f:53:f0:40
2023-09-22 11:25:15,196 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 03:6c:05:00:00:00:b1:fe:bf:83:e7:fa:fd:d0:1f:ef:f5:3e:7f:bf:d4:fa:ff:3f:73:f8:40
2023-09-22 11:25:16,245 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 59:6c:05:00:00:00:b8:ff:7f:73:ef:fb:fd:0f:bf:f3:f7:3e:7f:bf:e8:fb:ff:ff:84:00:40
2023-09-22 11:28:11,340 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 07:6c:05:00:00:00:05:0a:82:33:e8:2f:15:cc:c0:68:3b:4f:7f:a0:bd:2c:04:3f:63:2c:40
2023-09-22 11:28:12,349 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 7f:6c:05:00:00:00:07:f8:78:f3:ab:d5:e9:d0:ef:9b:c5:45:7d:4e:e0:e5:f7:7c:a3:ac:40
2023-09-22 11:28:13,327 logger   INFO: ACC frame received from - MAC: C2:3E:23:06:5B:AD - Raw data: 22:6c:05:00:00:00:0e:fe:bf:63:fb:fb:fd:4f:df:eb:f7:3e:7f:8f:e0:fa:fd:bf:73:e4:40
```

micha-1987, 2023-09-22:
yes sure, whereas I am mainly sticking to the scanner sample code.
I am just gathering the adv payload and forward it to my internal parser.

```
from service_data_translator import service_data_translator
import asyncio

from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData
from logger import logger
from const import ACCEPTED_DEVICES

def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):

        current_address = device.address
        if current_address in ACCEPTED_DEVICES:
            service_data_dict = advertisement_data[1]
                 
            for key, value in service_data_dict.items():
                        
                service_data_translator(device, value)

async def start_scanner():

    scanner = BleakScanner(
        detection_callback = simple_callback, scanning_mode = 'active'
    )
    
    try:
        logger.info("starting scanner")
        while True:
            
            await scanner.start()
            await asyncio.sleep(1)
            await scanner.stop()
    except KeyboardInterrupt:            
            scanner.stop()
            print("Stop scanning")

if __name__ == "__main__":

    asyncio.run(start_scanner())
```

I am also wondering that the telegrams dont have any service uuids but just manufaturing data (see picture).
[image]

The developer of the sensor firmware told me they cant use standard beacons like eddystone to pack the data and thus created their own telegram types. Does this have any influence?

dlech, 2023-09-26:
> Does this have any influence?

No, this is normal.

> Does bleak switch between those channels to watch each of them?

It is entirely up to the OS to do this. AFAIK all OSes scan all channels.

---

I would suggest logging Bluetooth packets (see troubleshooting page in docs) to better understand what is going on behind the scenes.

bojanpotocnik, 2024-01-08:
@micha-1987 have you solved the issue? Have you tried using `scanning_mode='passive'`? Based on my experience, especially behaviors number 2 and 3 are sometimes present on a BlueZ system when using active scanning.