hbldh/bleak #1425: MacOS + ESP32 "The attribute could not be found" on start_notify

alexanderturner, 2023-09-24:
* bleak version: 0.21.1
* Python version: 3.11.4
* Operating System: macOS 13.5 (Ventura)
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

start_notify calls an error on macOS, the application works fine on Windows devices. The peripheral is an ESP32-S3. The characteristic is valid and subscriptions work on mobile debugging apps without error.

### What I Did

```
    async with BleakClient(esp32, winrt=dict(use_cached_services=False)) as client:

        await client.start_notify(
            OTA_CONTROL_UUID,
            _ota_notification_handler
        )

```

### Logs

```
    asyncio.run(send_ota("bms-main.bin"))
  File "/usr/local/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/aturner/Code/Projects/LiTrek/esp32-micro/ota_pc_update/main.py", line 62, in send_ota
    await client.start_notify(
  File "/Users/aturner/.espressif/python_env/idf5.0_py3.11_env/lib/python3.11/site-packages/bleak/__init__.py", line 822, in start_notify
    raise BleakError(f"Characteristic {char_specifier} not found!")
bleak.exc.BleakError: Characteristic 101 not found!
```
[image]

alexanderturner, 2023-09-26:
Originally I thought this to be a mac issue, however it exists in the same form on a WIndows 10 machine as well. Works fine on Linux (Ubuntu 22.04)

alexanderturner, 2023-10-03:
Bumping this

bojanpotocnik, 2024-01-08:
I know this is a late comment, but if this is still a problem, can you try listing all characteristics before enabling notifications on the specific one
```python
    OTA_CONTROL_UUID = '00000101-0000-1000-8000-00805f9b34fb'

    async with BleakClient(esp32, winrt=dict(use_cached_services=False)) as client:

        # First, list the GATT server structure
        for service in client.services:
            print(f"[Service] {service}")
            for char in service.characteristics:
                print(f"  [Characteristic] {char} ({','.join(char.properties)})")
                for descriptor in char.descriptors:
                    print(f"    [Descriptor] {descriptor}")

        await client.start_notify(
            OTA_CONTROL_UUID,
            _ota_notification_handler
        )
```