hbldh/bleak #1428: Reading and writing is a bit slow and blocks other Bluetooth data transfers

zhou1591, 2023-09-26:
Hello, everyone, highly skilled friends. 

Would you like to ask why this packet will block the data transmission of other Bluetooth devices while awaiting read and write? Like I need a Bluetooth headset.

I used a Bluetooth module compiled by c++ before,which was given up due to compatibility issues between win and mac, but it only takes 70s before transmitting 2M packets at the speed, and then bleak takes about 55min. I see that the time difference is mainly reflected in the writing time. bleak writes 500 bytes about 800ms, before I used about 30ms, is there something wrong with my configuration?

My test code is as follows.  Thank you all.

```
import asyncio
import time

from bleak import BleakClient, BleakScanner

num = 0

address = "70:04:1D:AE:CB:12"
WRITE_IO_UUID = "0000abf1-0000-1000-8000-00805f9b34fb"
READ_IO_UUID = "0000abf2-0000-1000-8000-00805f9b34fb"

async def connect(address):
    async with BleakClient(address) as client:
        global num
        startTime = time.time()
        while (time.time() - startTime)<10:
            print(time.time() - startTime)
            model_number = await client.write_gatt_char(WRITE_IO_UUID,bytes('xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx','utf8'),True)
            model_number = await client.read_gatt_char(READ_IO_UUID)
            print("result: {0}".format("".join(map(chr, model_number))))
            num+=1

asyncio.run(connect(address))

print("10s number:",num ,"，total",num * 447 ,'bit')
```

dlech, 2023-09-26:
The best way to understand what is going on is to log Bluetooth packets (see troubleshooting page in docs).

I think it is common to get significantly degraded performance when more than one Bluetooth device is attached to the same adapter. 

It seems like sometimes adapters will manage to fit in more than one PDU in a single connection interval when multiple devices are connected but most of the time they won't. I haven't ever figured out what the difference is that could make it work sometimes but not others.

zhou1591, 2023-09-27:
> The best way to understand what is going on is to log Bluetooth packets (see troubleshooting page in docs).
>
> I think it is common to get significantly degraded performance when more than one [...]

Thank you for your reply.

I just turned off all other Bluetooth devices, except my esp32 Bluetooth module, but the writing is still about 500ms once.

I would like to ask if there is anything wrong in the sample code I wrote? 

Such as not maintaining io connection and so on. Is there any way to improve the time he writes?

dlech, 2023-09-27:
Like I said already, I think logging Bluetooth packets is the best way to understand what is going on.

zhou1591, 2023-10-08:
> Like I said already, I think logging Bluetooth packets is the best way to understand what is going on.

thanks

closed 2023-10-08 by zhou1591