hbldh/bleak #1432: connect does not succeed if called a second time

aaronfultonnz, 2023-10-04:
* bleak version: 0.21.1
* Python version: 3.10.5
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I am developing a Bluetooth device and often need to reset the MCU. After an MCU reset, I call disconnect() to ensure the PC side of things is shut down, then call connect() again after the MCU reset, however, connect() never completes. I have traced this down to the self._session.add_session_status_changed in the winrt/client.py file not being called on the second connect() call.

If I don't first call disconnect() then I get the following error:
 bleak.backends.winrt.client :: getting services (service_cache_mode=None, cache_mode=None)...
[2023-10-04 16:11:31,563] :: DEBUG :: bleak.backends.winrt.client :: skipping service 6e400001-b5a3-f393-e0a9-e50e24dcca9e due to access denied

If I don't reset the MCU and just call disconnect() and then connect() there are no errors.

### Logs
Here are the Bluetooth capture logs
[Logs.zip](https://github.com/hbldh/bleak/files/12798926/Logs.zip)

dlech, 2023-10-07:
> I have traced this down to the self._session.add_session_status_changed in the winrt/client.py file not being called on the second connect() call.

So in other words, `self._session = await GattSession.from_device_id_async()` is hanging forever and not returning?

> then I get the following error:

This is just a debug message, not an error.

> 
> Here are the Bluetooth capture logs

Thanks for the logs. It looks like the difference is in the good connect, the services are being enumerated whereas in the bad connect, it is using cached services and looking at the database hash characteristic. You might be able to play around with the windows-specific cache options to force it to never use the cache for getting services on Windows.

It would also be helpful to see the program used to reproduce the problem.

miyanyan, 2025-04-27:
> bleak.backends.winrt.client :: getting services (service_cache_mode=None, cache_mode=None)...
[2023-10-04 16:11:31,563] :: DEBUG :: bleak.backends.winrt.client :: skipping service 6e400001-b5a3-f393-e0a9-e50e24dcca9e due to access denied

I got the same error on my pc, but when I changed to use another pc, the error is gone! I think there's may due to driver or hardware issues, but I don't know how to catch that