hbldh/bleak #1433: Passive scanning on Raspberry Pi 4 model B

alphonsekoh, 2023-10-04:
* bleak version: 0.21.1
* Python version: 3.9.2
* Operating System: Linux (Debian Bullseye)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description

I tried passive scanning on my machine and it returned me an error `bleak.exc.BleakError: passive scanning on Linux requires BlueZ >= 5.55 with --experimental enabled and Linux kernel >= 5.10`. 

### What I Did

After getting the error, I checked the kernel version, BlueZ version and the bluetooth service has the `--experimental` tag appended. It all fulfils the requirements. 

I thought it might have been an hardware issue, however when I tried:
```
sudo hcitool lescan --passive
```
It gives me a list of mac addresses of the devices near me via passive scanning, so it led me wondering is there some wrong configurations that I did not know prior to running the example code shown below: 

```python
import asyncio
from bleak import BleakScanner
from bleak.backends.bluezdbus.scanner import BlueZScannerArgs, BlueZDiscoveryFilters
from bleak.assigned_numbers import AdvertisementDataType
from bleak.backends.bluezdbus.advertisement_monitor import OrPattern

def detection_callback(device, advertisement_data):
        print(f"Device (MAC: {device.address}, Name: {device.name}) detected!")

async def run():
        scanner = BleakScanner(scanning_mode="passive", bluez=BlueZScannerArgs(or_patterns=[>
        await scanner.start()
        await asyncio.sleep(10.0)  # Scan for 10 seconds
        await scanner.stop()

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

### Logs

When I run the example code I get this as logs:
```
bleak.backends.bluezdbus.manager MainThread DEBUG: initial properties: {'/org/bluez': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.AgentManager1': {}, 'org.bluez.ProfileManager1': {}, 'org.bluez.HealthManager1': {}}, '/org/bluez/test': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.SimAccessTest1': {}}, '/org/bluez/hci0': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'DC:A6:32:62:66:25', 'AddressType': 'public', 'Name': 'raspberrypi', 'Alias': 'raspberrypi', 'Class': 4194304, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001801-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000112d-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0537', 'Roles': ['central', 'peripheral']}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.GattManager1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 5, 'SupportedIncludes': ['appearance', 'local-name']}, 'org.bluez.Media1': {}, 'org.bluez.NetworkServer1': {}, 'org.bluez.SimAccess1': {'Connected': False}}}
Traceback (most recent call last):
  File "/home/pi/testbleak.py", line 17, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/testbleak.py", line 12, in run
    await scanner.start()
  File "/usr/local/lib/python3.9/dist-packages/bleak/__init__.py", line 198, in start
    await self._backend.start()
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/scanner.py", line 178, in start
    self._stop = await manager.passive_scan(
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/manager.py", line 523, in passive_scan
    raise BleakError(
bleak.exc.BleakError: passive scanning on Linux requires BlueZ >= 5.55 with --experimental enabled and Linux kernel >= 5.10
```

dlech, 2023-10-04:
Does `sudo systemctl status bluetooth.service` show that the `--experimental` flag is actually enabled on the running instance?

alphonsekoh, 2023-10-05:
It shows the `--experimental` flag on the CGroup

```
bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-10-04 16:49:42 +08; 24h ago
       Docs: man:bluetoothd(8)
   Main PID: 2186 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 1599)
        CPU: 56ms
     CGroup: /system.slice/bluetooth.service
             └─2186 /usr/libexec/bluetooth/bluetoothd --experimental
```

dlech, 2023-10-05:
Hmm... maybe we got the version requirements wrong?

If BlueZ supports it, there should be a `interface org.bluez.AdvertisementMonitorManager1` in the initial properties debug log message.

You could also have a look at kernel logs (`dmesg`) and BlueZ logs (also add `--debug` and restart `bluetoothd`) to see if there any hints there.

alphonsekoh, 2023-10-23:
The issue has been resolved, I updated my bluez version to `5.56` from `5.55` by wget from http://www.kernel.org/pub/linux/bluetooth. In 5.56 it fixes the advertisement monitor issue from passive scanning.

commit 2d70d1c0 (2023-10-24): Fixed BlueZ version in passive scanning error message.

Apperently the feature changed between 5.55 and 5.56 so >= 5.56 is requrired.

Fixes #1433.
PR #1562 (2024-05-04): Release/0.22.0

Added
-----
* Added ``BleakCharacteristicNotFoundError`` which is raised if a device does not support a characteristic.
* Added utility function to work around ``pywin32`` setting threading model to STA on Windows.

Changed
-------
* Updated PyObjC dependency on macOS to v10.x.
* Updated missing Bluetooth SIG characteristics and service UUIDs.
* Updated ``BlueZManager`` to remove empty interfaces from `_properties` during InterfacesRemoved message.
* Updated PyWinRT dependency to v2. Fixes #1529.
* Raise exception when trying to scan while in a single-treaded apartment (STA) on [... cut]
closed 2023-10-23 by alphonsekoh