hbldh/bleak #1435: KeyError: 'org.bluez.GattService1' during BleakClient connection

ElieCo, 2023-10-06:
* bleak version: 0.21.1
* Python version: 3.9.2
* Operating System: Armbian 23.08.0 Bullseye on Bananapim2zero
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

I am creating a script that connect and disconnect multiple times to a same BLE device.
The BLE device restart sometimes its services.
When the device restart its services between two connections, a KeyError appears in my script during the second connection.

### Script example

``` py
import asyncio
import time
from bleak import BleakClient, BleakScanner

async def main():
    while True:
        devices = await BleakScanner.discover()
        for device in devices:
            if device.name == "MyDevice":
                client = BleakClient(device.address)
                await client.connect()
                print("Connected to my device.")
                await client.disconnect()

        time.sleep(2)

if __name__ == "__main__":
    asyncio.run(main())
```

### Logs

``` bash
Traceback (most recent call last):
  File "/home/elie/test_connection.py", line 19, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/elie/test_connection.py", line 12, in main
    await client.connect()
  File "/usr/local/lib/python3.9/dist-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/client.py", line 268, in connect
    await self.get_services(
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/client.py", line 656, in get_services
    self.services = await manager.get_services(
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/manager.py", line 651, in get_services
    self._properties[service_path][defs.GATT_SERVICE_INTERFACE],
KeyError: 'org.bluez.GattService1'
```

dlech, 2023-10-06:
* `time.sleep(...)` will break asyncio by blocking program execution. Use `await asyncio.sleep(...)` instead.
* Using `BleakClient(device)` instead of `BleakClient(device.address)` will prevent having to scan a second time.

> ```shell
> KeyError: 'org.bluez.GattService1'
> ```

This sounds like we may have introduced a bug in some changes we made for this release. Will look into later.

shreyasgokhale, 2024-01-02:
Is there a solution / fix for this issue? 

I could reproduce the same issue from bleak v0.16+ and on 5.64 - 5.71 (latest) bluez versions.

savvn001, 2024-01-28:
Same issue here. It happens when I connect to a device with the same mac address but the GATT table has changed. Is this caching related?

dlech, 2024-04-30:
https://github.com/hbldh/bleak/pull/1511 will probably change the error message seen here, but probably doesn't fix the issue completely (and will likely cause the previous workaround to not work anymore).

More logging here would be appreciated to help understand the issue better since I'm not able to reproduce it.

heinemml, 2024-10-10:
Hi,

I ran into this issue today while implementing an OTA Update. The device is restarted via a write to a gatt characteristic. when restarted in OTA mode the service and characterisitics for OTA are still there but all others are not. I suspect that this leads to the problem. As @dlech suggested the error message changes when I go from bleak 0.20.x to 0.22.3. This is what i now see:

```
Traceback (most recent call last):
  File "/home/heine/gecko_ota/./ota.py", line 133, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/heine/gecko_ota/./ota.py", line 130, in main
    await try_update(args.addr, args.gbl)
  File "/home/heine/gecko_ota/./ota.py", line 87, in try_update
    async with BleakClient(device) as client:
  File "/home/heine/gecko_ota/lib/python3.11/site-packages/bleak/__init__.py", line 570, in __aenter__
    await self.connect()
  File "/home/heine/gecko_ota/lib/python3.11/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/heine/gecko_ota/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 273, in connect
    await self.get_services(
  File "/home/heine/gecko_ota/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 661, in get_services
    self.services = await manager.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/heine/gecko_ota/lib/python3.11/site-packages/bleak/backends/bluezdbus/manager.py", line 665, in get_services
    self._properties[service_path][defs.GATT_SERVICE_INTERFACE],
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
KeyError: '/org/bluez/hci0/dev_68_0A_E2_E7_84_6D/service0019'
```

when I keep an eye on dbus monitor I see that after the (re-)connect the discovered characteristics are announces with different "service<number>" paths. In the mentioned case the service assigned the 0019 is a characteristic that is not there when booted in OTA mode.

Please find attached the dbus-monitor log: [dbus.log](https://github.com/user-attachments/files/17333779/dbus.log)

To me this looks like a caching issue in the global manager instance...

If you need any more information to analyse this I'm happy to help.

dlech, 2024-10-27:
What is the purpose of the extra `await BleakScanner().discover()`?

dlech, 2024-10-28:
Here is a patch to try:

```diff
diff --git a/bleak/backends/bluezdbus/manager.py b/bleak/backends/bluezdbus/manager.py
index cedbd64..5c65170 100644
--- a/bleak/backends/bluezdbus/manager.py
+++ b/bleak/backends/bluezdbus/manager.py
@@ -945,6 +945,13 @@ class BlueZManager:
                         if obj_path.startswith(adapter_path):
                             callback(obj_path)
                 elif interface == defs.GATT_SERVICE_INTERFACE:
+                    device_path = obj_path[: obj_path.rfind("/")]
+
+                    try:
+                        self._service_map[device_path].remove(obj_path)
+                    except KeyError:
+                        pass
+
                     try:
                         del self._characteristic_map[obj_path]
                     except KeyError:
```

heinemml, 2024-10-30:
The `await BleakScanner().discover()` was just a try to update/flush the GATT database. Which didn't work out.

But your patch is working. I was now able to trigger a device reset, reconnect and flash the update ten times in a row. 🙃

closed 2025-04-19 by dlech