hbldh/bleak #1436: Apple Sonoma reduced throughput

wiesener, 2023-10-08:
* bleak version: 0.21.1
* Python version: 3.11.4
* Operating System: mac os sonoma 14.0

### Description

When upgrading to apple mac os sonoma the throughput with our bleak client reduced by factor 3. We did not yet figured out why. What we can see is when logging is switched on that there a lot of these messages:

```2023-10-08 15:08:06,765 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_```

It is definitely somehow an issue of apple but maybe something can be configured in bleak to get around it.

### What I Did

This is somehow difficult since we are running a custom program where we configure our device to get into streaming mode and in this mode the device is sending data normally with 56 kbytes / s. Now after the sonoma update we can only get 15 kbytes / s.

### Logs

```
2023-10-08 15:08:01,162 bleak.backends.corebluetooth.CentralManagerDelegate Dummy-1 DEBUG: centralManagerDidUpdateState_
2023-10-08 15:08:01,162 bleak.backends.corebluetooth.CentralManagerDelegate Dummy-1 DEBUG: Bluetooth powered on
2023-10-08 15:08:01,163 bleak.backends.corebluetooth.CentralManagerDelegate MainThread DEBUG: 'isScanning' changed
devices: {}
2023-10-08 15:08:02,437 bleak.backends.corebluetooth.CentralManagerDelegate Dummy-1 DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2023-10-08 15:08:02,440 bleak.backends.corebluetooth.CentralManagerDelegate MainThread DEBUG: Discovered device B6E64BC3-4DB9-B275-E655-91B1F34AF4A1: IMU_8e3f65 @ RSSI: -54 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600003bc85a0>
devices: {'IMU_8e3f65': BleDevice(state=disconnected, rssi=-54)}
connecting
connecting
2023-10-08 15:08:02,450 bleak.backends.corebluetooth.client MainThread DEBUG: CentralManagerDelegate  at <CentralManagerDelegate: 0x1376c8620>
2023-10-08 15:08:02,450 bleak.backends.corebluetooth.client MainThread DEBUG: Connecting to BLE device @ B6E64BC3-4DB9-B275-E655-91B1F34AF4A1
2023-10-08 15:08:02,451 bleak.backends.corebluetooth.CentralManagerDelegate MainThread DEBUG: 'isScanning' changed
2023-10-08 15:08:03,140 bleak.backends.corebluetooth.CentralManagerDelegate Dummy-1 DEBUG: centralManager_didConnectPeripheral_
2023-10-08 15:08:03,141 bleak.backends.corebluetooth.client MainThread DEBUG: Retrieving services...
2023-10-08 15:08:03,434 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didDiscoverServices_
2023-10-08 15:08:03,436 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Services discovered
2023-10-08 15:08:03,437 bleak.backends.corebluetooth.client MainThread DEBUG: Retrieving characteristics for service 80030001-E629-4C98-9324-AA7FC0C66DE7
2023-10-08 15:08:03,553 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didDiscoverCharacteristicsForService_error_
2023-10-08 15:08:03,555 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Characteristics discovered
2023-10-08 15:08:03,556 bleak.backends.corebluetooth.client MainThread DEBUG: Retrieving descriptors for characteristic 80030002-E629-4C98-9324-AA7FC0C66DE7
2023-10-08 15:08:03,557 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2023-10-08 15:08:03,557 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Descriptor discovered 15
2023-10-08 15:08:03,558 bleak.backends.corebluetooth.client MainThread DEBUG: Retrieving descriptors for characteristic 80030003-E629-4C98-9324-AA7FC0C66DE7
2023-10-08 15:08:03,672 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2023-10-08 15:08:03,674 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Descriptor discovered 17
2023-10-08 15:08:03,675 bleak.backends.corebluetooth.client MainThread DEBUG: Services resolved for BleakClientCoreBluetooth (B6E64BC3-4DB9-B275-E655-91B1F34AF4A1)
2023-10-08 15:08:03,732 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateNotificationStateForCharacteristic_error_
2023-10-08 15:08:03,733 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Character Notify Update
2023-10-08 15:08:03,734 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
connected
2023-10-08 15:08:03,792 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didWriteValueForCharacteristic_error_
2023-10-08 15:08:03,793 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:03,794 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Write Characteristic Value
2023-10-08 15:08:03,795 bleak.backends.corebluetooth.client MainThread DEBUG: Write Characteristic 80030002-e629-4c98-9324-aa7fc0c66de7 : b'\x02\x9a\x14\x98\xe9\x000\x01'
2023-10-08 15:08:03,853 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didWriteValueForCharacteristic_error_
2023-10-08 15:08:03,854 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:03,854 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Write Characteristic Value
2023-10-08 15:08:03,855 bleak.backends.corebluetooth.client MainThread DEBUG: Write Characteristic 80030002-e629-4c98-9324-aa7fc0c66de7 : b'\x022\xb7\x80|\x15\x10\x01\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x11\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00'
2023-10-08 15:08:03,912 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didWriteValueForCharacteristic_error_
2023-10-08 15:08:03,912 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:03,913 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Write Characteristic Value
2023-10-08 15:08:03,914 bleak.backends.corebluetooth.client MainThread DEBUG: Write Characteristic 80030002-e629-4c98-9324-aa7fc0c66de7 : b'\x02m\x98\xfc\x90\x03\x16\x01\x01\x00\x00'
2023-10-08 15:08:03,972 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didWriteValueForCharacteristic_error_
2023-10-08 15:08:03,973 bleak.backends.corebluetooth.PeripheralDelegate MainThread DEBUG: Write Characteristic Value
2023-10-08 15:08:03,974 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:03,974 bleak.backends.corebluetooth.client MainThread DEBUG: Write Characteristic 80030002-e629-4c98-9324-aa7fc0c66de7 : b'\x02\xcb\x06Z\xa3\x00 \x01'
2023-10-08 15:08:04,033 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,034 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,035 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,036 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,093 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
[... 30 lines cut ...]
2023-10-08 15:08:04,606 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,664 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,665 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,666 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,666 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,723 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
[... cut ...]
2023-10-08 15:08:04,906 bleak.backends.corebluetooth.PeripheralDelegate Dummy-2 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,963 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,964 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2023-10-08 15:08:04,965 bleak.backends.corebluetooth.PeripheralDelegate Dummy-1 DEBUG: peripheral_didUpdateValueForCharacteristic_error
```

dlech, 2023-10-08:
Did you report the regression to Apple since the only difference is the operating system version?

Sniffing Bluetooth packets is usually the best way to see what is different between a working and not working case.

wiesener, 2023-10-09:
Yes, I've already opened a Feedback Ticket and a Developer Ticket at Apple Dev Platform. And sure it is definitely Apples duty. But I was wondering, whether it might be already a hint that the logging console is already spammed with this error notifications. I will check if these notifications are not present on Apple Ventura systems or Linux / Windows.

wiesener, 2023-10-09:
I've checked the sniffer output and it is definitely a change in the connection interval from 11.25 ms to 30 ms for Ventura to Sonoma. We could force the Sonoma somehow to go to 15 ms. But less was not possible so far.

closed 2023-10-09 by wiesener

dlech, 2023-10-09:
> whether it might be already a hint that the logging console is already spammed with this error notifications

I only see DEBUG, not ERROR. In `peripheral_didUpdateValueForCharacteristic_error_`, the error parameter is optional so doesn't indicate an error unless the argument is not null/none.