hbldh/bleak #1437: find_device_by_filter is called with incomplete data

jpstotz, 2023-10-11:
* bleak version: bleak 0.21.1 / bleak-winrt 1.2.0
* Python version: 3.11.3
* Operating System: Windows 10

### Description

The callback method specified as argument in `find_device_by_filter` is only called once per device. It seems like the callback is called when the first time `AdvertisementData` is received. from the `detect_ble_devices.py` script I can see that from my device I receive two times such data, the first time only containing `manufacturer_data` and `rssi` (this is the data I get in the callback) and then a second time containing `local_name`, `manufacturer_data`, `service_uuids`, `tx_power`and `rssi`. 
This second data will never be seen in the callback, therefore if I want to filter the found devices for `local_name` and `manufacturer_data` or `service_uuids` this is not possible because this data will never be included in the data I see in the callback.

### What I Did

```
    def device_filter(d, ad):
        print(ad.local_name) # always None
        print(ad.manufacturer_data) # contains some data
        print(ad.service_uuids) # always []
        return False

    await BleakScanner.find_device_by_filter(device_filter)
```
Output:
```
None
{76: b'\x12\x19\x00\x9e\\\xdf\x9cf\xfd\xc6\xf1)Lu\xead+\xa3\x14\xc4\xda\x96DJ\xd7\x00\x00'}
[]
```

`device_filter` is only called once for my one device. As it uses a random address and there can be multiple devices of the same name I can neither use `find_device_by_address` nor `find_device_by_name`. I need `local_name` and `manufacturer_data` for selecting the correct device but that is not possible in bleak at the moment.

dlech, 2023-10-11:
Do you see multiple advertisements received when debug logging is enabled?

jpstotz, 2023-10-12:
Yes I see multiple advertisements, first a short one only containing manufacturer_data and then multiple "full" advertisements:

The log of `detect_ble_devices.py` looks like this (all data has been obfuscated):
```
01:02:23:34:14:10: AdvertisementData(manufacturer_data={368: b'somedata'}, rssi=-62)
01:02:23:34:14:10: AdvertisementData(local_name='SomeDevice', manufacturer_data={368: b'somedata'}, service_uuids=['00001234-0000-0000-0000-000000000000'], tx_power=0, rssi=-62)
<last line repeated 5 to 10 times>
(re)starting scanner
<again this complete block>
```

dlech, 2023-10-12:
As far as I know, there isn't anything preventing multiple callbacks in the Bleak code.

I was looking for this line in your debug logs: https://github.com/hbldh/bleak/blob/42797edb99534e5558433deca29fdc08574ceae8/bleak/backends/winrt/scanner.py#L109

But since your lines look different, I'm guessing the information is propagating to code you wrote.

jpstotz, 2023-10-12:
The script I use I think is an example bleak script. It seems that i renamed it as here on Git it is [this script](https://github.com/hbldh/bleak/blob/develop/examples/detection_callback.py). The output I had posted are from this log statement: 
https://github.com/hbldh/bleak/blob/42797edb99534e5558433deca29fdc08574ceae8/examples/detection_callback.py#L23

But I enabled debug logging on the scanner and based on the BT address I get two types of DEBUG messages that match the logging pattern from your last comment:

```
Received 01:02:23:34:14:10: .
Received 01:02:23:34:14:10: SomeDevice.
```
I receive each variant multiple times.

dlech, 2023-10-12:
Can you share the full problematic script?

closed 2023-11-14 by jpstotz