hbldh/bleak #1443: Error Domain=CBATTErrorDomain Code=14 "Unlikely error." when writing to a Write with Response characteristic

mikevoyt, 2023-10-27:
* bleak version: 0.21.1
* Python version: 3.9
* Operating System: macOS Sonoma 14.0
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I have a peripheral with a Write with Response characteristic.  It's defined in Zephyr with `BT_GATT_CHRC_WRITE`.  When I write to it I get this error:

```
bleak.exc.BleakError: Failed to write characteristic 24: Error Domain=CBATTErrorDomain Code=14 "Unlikely error." UserInfo={NSLocalizedDescription=Unlikely error.}
```

### What I Did

Tried calling `await client.write_gatt_char(WRITE_CHARACTERISTIC_UUID, data)`:

```
    await self.client.write_gatt_char(self.tx_uuid, data_fr)
  File "/usr/local/lib/python3.9/site-packages/bleak/__init__.py", line 776, in write_gatt_char
    await self._backend.write_gatt_char(characteristic, data, response)
  File "/usr/local/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 321, in write_gatt_char
    await self._delegate.write_characteristic(
  File "/usr/local/lib/python3.9/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 192, in write_characteristic
    await future
bleak.exc.BleakError: Failed to write characteristic 24: Error Domain=CBATTErrorDomain Code=14 "Unlikely error." UserInfo={NSLocalizedDescription=Unlikely error.}
```

Also tried setting 'response=True`:
```
    await client.write_gatt_char(WRITE_CHARACTERISTIC_UUID, data, response=True)
  File "/usr/local/lib/python3.9/site-packages/bleak/__init__.py", line 776, in write_gatt_char
    await self._backend.write_gatt_char(characteristic, data, response)
  File "/usr/local/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 321, in write_gatt_char
    await self._delegate.write_characteristic(
  File "/usr/local/lib/python3.9/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 192, in write_characteristic
    await future
bleak.exc.BleakError: Failed to write characteristic 24: Error Domain=CBATTErrorDomain Code=14 "Unlikely error." UserInfo={NSLocalizedDescription=Unlikely error.}
```

Note: a Write without Response characteristic works just fine.  This seems to be an issue only with Write with Response.

dlech, 2023-10-28:
If you log Bluetooth packets, you should be able to see if this error is coming from the remote device or not.

mikevoyt, 2023-10-30:
Hmm yes the error appears to come from the remote device:

```
Oct 30 10:38:39.925  ATT Send         0x004B  DF:6F:99:C7:50:AA  Write Request - Handle:0x0019 - 6765A69D-CD79-4DF6-AAD5-043DF9425556 - Value: 010E F106 0000 018B 81AA 8CB3 285A  
	Write Request - Handle:0x0019 - 6765A69D-CD79-4DF6-AAD5-043DF9425556 - Value: 010E F106 0000 018B 81AA 8CB3 285A
	Opcode: 0x12
	Attribute Handle: 0x0019 (25)
	Value: 010E F106 0000 018B 81AA 8CB3 285A
Oct 30 10:38:39.925  L2CAP Send       0x004B  DF:6F:99:C7:50:AA  Channel ID: 0x0004  Length: 0x0011 (17) [ 12 19 00 01 0E F1 06 00 00 01 8B 81 AA 8C B3 28 ... ]  
Oct 30 10:38:39.928  HCI Event        0x004B  DF:6F:99:C7:50:AA  Number Of Completed Packets - Handle: 0x004B - Packets: 0x0002    
Oct 30 10:38:39.952  ATT Receive      0x004B  DF:6F:99:C7:50:AA  Error Response - Attribute Handle: 0x0019 - Error Code: Unlikely Error (0x0E) - Unlikely Error  
	Error Response - Attribute Handle: 0x0019 - Error Code: Unlikely Error (0x0E)
	Opcode: 0x01
	Request Opcode In Error: 0x0012 (Write Request)
	Attribute Handle In Error: 0x0019 (25)
	Error Code: 0x000E (Unlikely Error)

```

closed 2023-10-30 by dlech