hbldh/bleak #1445: Passive scan `or_pattern` is ignored if connected to another device

vboginskey, 2023-11-05:
* bleak version: 0.21.1
* Python version: 3.11.2
* Operating System: Debian Bookworm 64-bit (Raspberry Pi)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

I need to run a passive scan to retrieve advertising packets from a specific device while simultaneously connecting to another device. When I connect to the other device, the passive scan starts returning advertisement data that does not match the provided `or_pattern`.

I expect the passive scan to respect my `or_pattern` at all times and not be affected by a connection to another device.

### What I Did

Here is a minimal reproduction:

```python
import asyncio
import logging
from bleak import BleakClient, BleakScanner
from bleak.assigned_numbers import AdvertisementDataType
from bleak.backends.bluezdbus.advertisement_monitor import OrPattern
from bleak.backends.bluezdbus.scanner import BlueZScannerArgs 

async def scan():
    args = BlueZScannerArgs(
        or_patterns=[OrPattern(0, AdvertisementDataType.MANUFACTURER_SPECIFIC_DATA, b"\xe1\x02")]
    )

    async with BleakScanner(bluez=args, scanning_mode="passive") as scanner:
        async for _, advertisement_data in scanner.advertisement_data():
            mfr_data = advertisement_data.manufacturer_data
            if mfr_data.get(0x02e1):
                logging.info("scan(): found correct device: %s", mfr_data)
            else:
                logging.info("scan(): this should never happen: %s", mfr_data)

async def connect():
    device1 = await BleakScanner.find_device_by_address("01:B6:EC:10:CB:8F")

    async with BleakClient(device1):
        logging.info("connect(): connected to device")
        await asyncio.sleep(60)

async def main():
    logging.info("main(): starting scan")
    asyncio.create_task(scan())
    logging.info("main(): sleeping for 10 seconds")
    await asyncio.sleep(10)
    logging.info("main(): connecting to another device")
    await asyncio.create_task(connect())

logging.basicConfig(
    level=logging.INFO,
    format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
)

asyncio.run(main())
```

Here's the output:
```
2023-11-05 10:41:07,696 root     INFO: main(): starting scan
2023-11-05 10:41:07,697 root     INFO: main(): sleeping for 10 seconds
2023-11-05 10:41:08,149 root     INFO: scan(): found correct device: {737: b'\x10\x02\x83\xa3\x02\x17G\xd0\xb9\x9a\xedh\xc2\xad`\x04\xb2E\xda"\xea=A'}
2023-11-05 10:41:17,708 root     INFO: main(): connecting to another device
2023-11-05 10:41:17,795 root     INFO: scan(): found correct device: {737: b'\x10\x02\x83\xa3\x02 G\xd0\xba\xa5\x13\x15m\x89m\xe8\\\xb1\r\x82\x93\xe8n'}
2023-11-05 10:41:17,854 root     INFO: scan(): this should never happen: {54828: b'\x88\xa0ww\x19\r\xf3<'}
2023-11-05 10:41:17,884 root     INFO: scan(): this should never happen: {49568: b'\x88\xa0ww\x19\r\xe2\xb0'}
2023-11-05 10:41:17,907 root     INFO: scan(): this should never happen: {224: b'2]\xca\x82K\xdc'}
2023-11-05 10:41:17,912 root     INFO: scan(): this should never happen: {61057: b'\x88\xa0\x01\xb6\xec\x10\xcb\x8f'}
2023-11-05 10:41:17,916 root     INFO: scan(): this should never happen: {224: b'2]\xca\x82K\xdc'}
2023-11-05 10:41:17,917 root     INFO: scan(): this should never happen: {61057: b'\x88\xa0\x01\xb6\xec\x10\xcb\x8f'}
2023-11-05 10:41:17,918 root     INFO: scan(): this should never happen: {49568: b'\x88\xa0ww\x19\r\xe2\xb0'}
2023-11-05 10:41:17,919 root     INFO: scan(): this should never happen: {54828: b'\x88\xa0ww\x19\r\xf3<'}
2023-11-05 10:41:17,919 root     INFO: scan(): found correct device: {737: b'\x10\x02\x83\xa3\x02 G\xd0\xba\xa5\x13\x15m\x89m\xe8\\\xb1\r\x82\x93\xe8n'}
2023-11-05 10:41:17,955 root     INFO: scan(): found correct device: {737: b'\x10\x02\x83\xa3\x02!G\xd0,\xb6\x00\x14\xb3>z\xd8p\xe8\xd4\xab\x13\xe1Q'}
2023-11-05 10:41:18,517 root     INFO: scan(): this should never happen: {61057: b'\x88\xa0\x01\xb6\xec\x10\xcb\x8f'}
2023-11-05 10:41:18,763 root     INFO: scan(): this should never happen: {61057: b'\x88\xa0\x01\xb6\xec\x10\xcb\x8f'}
2023-11-05 10:41:18,764 root     INFO: connect(): connected to device
```

dlech, 2023-11-08:
Connecting to a device requires scanning. So this is probably a BlueZ bug/feature. I would suggest asking there and see if this is the expected behavior or a bug (passive scanning is still experimental in BlueZ as far as I know).

dlech, 2023-11-08:
You could also use an asyncio `Lock` to ensure that scanning and connecting don't happen at the same time.

vboginskey, 2023-11-08:
I don't think it's BlueZ, since this doesn't happen with `bluetoothctl`. Here is the code modified to not connect and scan at the same time:

```python
import asyncio
import logging
from bleak import BleakClient, BleakScanner
from bleak.assigned_numbers import AdvertisementDataType
from bleak.backends.bluezdbus.advertisement_monitor import OrPattern
from bleak.backends.bluezdbus.scanner import BlueZScannerArgs 

async def scan(connected):
    logging.info("scan(): waiting for device to connect")
    await connected.wait()
    logging.info("scan(): starting passive scan")

    args = BlueZScannerArgs(
        or_patterns=[OrPattern(0, AdvertisementDataType.MANUFACTURER_SPECIFIC_DATA, b"\xe1\x02")]
    )

    async with BleakScanner(bluez=args, scanning_mode="passive") as scanner:
        async for _, advertisement_data in scanner.advertisement_data():
            mfr_data = advertisement_data.manufacturer_data
            if mfr_data.get(0x02e1):
                logging.info("scan(): found correct device: %s", mfr_data)
            else:
                logging.info("scan(): this should never happen: %s", mfr_data)

async def connect(connected):
    async with BleakClient("01:B6:EC:10:CB:8F"):
        logging.info("connect(): connected to device")
        connected.set()
        await asyncio.sleep(60)

async def main():
    connected = asyncio.Event()

    asyncio.create_task(connect(connected))
    await asyncio.create_task(scan(connected))

logging.basicConfig(
    level=logging.INFO,
    format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
)

asyncio.run(main())
```

And its output:
```
2023-11-08 10:27:15,389 root     INFO: scan(): waiting for device to connect
2023-11-08 10:27:16,286 root     INFO: connect(): connected to device
2023-11-08 10:27:16,287 root     INFO: scan(): starting passive scan
2023-11-08 10:27:16,463 root     INFO: scan(): found correct device: {737: b'\x10\x02\x83\xa3\x02L8\xd0\xa7\x1c\xa1\xa5\x0cv\xea5\xddJ\xf1\x91=\r\x9f'}
2023-11-08 10:28:18,082 root     INFO: scan(): this should never happen: {61057: b'\x88\xa0\x01\xb6\xec\x10\xcb\x8f'}
2023-11-08 10:28:18,084 root     INFO: scan(): this should never happen: {61057: b'\x88\xa0\x01\xb6\xec\x10\xcb\x8f'}
```

The `bluetoothctl` output is rather messy, but let me try to reproduce the relevant lines:
```
[bluetooth]# add-or-pattern 0 255 e102
Advertisement Monitor 0 added
Advertisement monitor 0 activated
[NEW] Device E6:27:C2:30:0E:BC Magnus BMV
Advertisement monitor 0 found device /org/bluez/hci0/dev_E6_27_C2_30_0E_BC
[bluetooth]# scan le
Discovery started

[snip]

Advertisement monitor 0 found device /org/bluez/hci0/dev_E6_27_C2_30_0E_BC
[bluetooth]# scan off
Discovery stopped
[bluetooth]# connect 01:B6:EC:10:CB:8F
Attempting to connect to 01:B6:EC:10:CB:8F
[CHG] Device E6:27:C2:30:0E:BC RSSI: -71
[CHG] Device E6:27:C2:30:0E:BC ManufacturerData Key: 0x02e1
[CHG] Device E6:27:C2:30:0E:BC ManufacturerData Value:
  10 02 83 a3 02 84 43 d0 35 f3 d7 b7 9f 39 83 25  ......C.5....9.%
  72 42 08 89 e5 d6 b8                             rB.....
Advertisement monitor 0 found device /org/bluez/hci0/dev_E6_27_C2_30_0E_BC
[CHG] Device 01:B6:EC:10:CB:8F Connected: yes
Connection successful
[CHG] Device E6:27:C2:30:0E:BC RSSI: -61
[CHG] Device E6:27:C2:30:0E:BC ManufacturerData Key: 0x02e1
[CHG] Device E6:27:C2:30:0E:BC ManufacturerData Value:
  10 02 83 a3 02 85 43 d0 a1 f0 72 0a de ab f9 3b  ......C...r....;
  6d 5f 9e 5a 69 49 a4                             m_.ZiI.
Advertisement monitor 0 found device /org/bluez/hci0/dev_E6_27_C2_30_0E_BC
```

In the above output, note that Advertisement monitor 0 only reports finding `E6:27:C2:30:0E:BC`, which is the correct device. The connection to `01:B6:EC:10:CB:8F`, or even scanning actively do not trigger the advertisement monitor with any other device or data.

dlech, 2023-11-08:
I see lines like `[CHG] Device E6:27:C2:30:0E:BC ManufacturerData Key: 0x02e1` after the connect in the `bluetoothctl` output, which means BlueZ is emitting property change signals for the advertisement data during connect. This is what is triggering the callbacks in Bleak.

Bleak just uses the advertisement monitor as a way to enable passive scanning. There isn't a special event in Bleak that corresponds to the device found event from BlueZ.

vboginskey, 2023-11-08:
> I see lines like [CHG] Device E6:27:C2:30:0E:BC ManufacturerData Key: 0x02e1 after the connect in the bluetoothctl output, which means BlueZ is emitting property change signals for the advertisement data during connect. This is what is triggering the callbacks in Bleak.

Yes, that's what I would expect.

Let me try to clarify the problem: with `bluetoothctl`, my advertisement monitor filters correctly by the provided or_pattern. The property change signals are only seen for the device I'm actually interested in. With Bleak, `scanner.advertisement_data()` returns any discovered advertisement data whether it matches the or_pattern or not. Is this expected behavior in Bleak?

vboginskey, 2023-11-08:
Ok, I understand after rereading your reply. Thanks - feel free to close the issue if this is expected behavior.

dlech, 2023-11-08:
Yeah, I'll go ahead and close since BlueZ will adjust scanning to try to accommodate scanning by multiple requesters by trying to merge the requests and come up with a best fit. So, generally speaking, even if Bleak requested passive scanning, that might not be what we actually get from BlueZ.

closed 2023-11-08 by dlech