hbldh/bleak #1446: Missing connection parameter settings

StefJar, 2023-11-07:
The connection parameters like connection interval and timeouts can be set from peripheral and central side.

Currently I am missing that API.

When trying to run my bleak based python code on Windows the connection is very slow(I guess that is because of a wrongly setup connection interval). 

Please add an API for doing that.

dlech, 2023-11-08:
The OS Bluetooth stacks handle this on the central side so Bleak can't really do anything about it. So if you have control over the peripheral firmware, you will need to fix it there.

StefJar, 2023-11-17:
I am not an expert on bt stack user space APIs. As far as I know you can setup the connection parameters on Android (check the NRFconnect app).
The connection parameters like MTU, interval and timeouts can be set from both sides central and peripheral.

I have no idea, but maybe a config side channel for each backend that can handle different parameters like MTU, connection interval, etc. would be the best solution.

for blueZ something like this:
https://stackoverflow.com/questions/40125867/ble-connect-interval-setting

dlech, 2025-04-20:
FWIW, Windows 11 made some APIs available for this, but still not widely available cross-platform.