hbldh/bleak #1447: Pattern being ignored?

Odud, 2023-11-07:
* Bleak: 0.21.1
* Python version: 3.12.0
* Operating System: Debian GNU/Linux 12 (bookworm) (Raspberry Pi)
* BlueZ version (`bluetoothctl -v`) in case of Linux: bluetoothctl: 5.66

### Description

Trying to only scan for devices from a particular manufacturer, so I used whatI though was the correct argument to the scanner:
`async with BleakScanner(callback, bluez={"Pattern": "A4:C1:38"}) as scanner:`

A4:C1:38 is the start of the MAC addresses thatIam iterested in. However all devices seem to be passed back to the callback

dlech, 2023-11-08:
It sounds like what you want is [BleakScanner.find_device_by_filter()](https://bleak.readthedocs.io/en/latest/api/scanner.html#bleak.BleakScanner.find_device_by_filter).

FYI, filtering by address doesn't work on macOS, so if you were trying to make something that works cross platform, you should consider filters on something from the advertisement data instead.

Odud, 2023-11-08:
That's equivalent to what I do at the moment - I was hoping to offload the
filtering to BlueZ

Pete Barlow

dlech, 2023-11-08:
If the device in question has a service UUID in the advertising data, you ca use the `service_uuids` keyword arg to `BleakScanner()` to offload filtering. Filtering by address (and pretty much anything else) can't be offloaded.

filmkorn, 2024-04-07:
Unsure if this is relevant to this issue - but I found that on bluez the `service_uuids` argument is ignored and the callback is called on non-matching devices.

- Bleak: 0.21.1
- Python version: 3.11.7
- Operating System (`uname -a`): Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- BlueZ version (bluetoothctl -v) in case of Linux: bluetoothctl: 5.66

Example:

```python
import asyncio

from bleak import BleakScanner, AdvertisementData, BLEDevice

SERVICE_UUID = "0e140000-0af1-4582-a242-773e63054c68"

async def detection_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    print(f"Found device: {device.address} (device.name)")
    print(f"Matches service UUID: {SERVICE_UUID in advertisement_data.service_uuids}")

async def discover():
    scanner = BleakScanner(
        detection_callback=detection_callback,
        use_bdaddr=False,
        service_uuids=[SERVICE_UUID],
    )

    await scanner.start()
    await asyncio.sleep(20)

asyncio.run(discover())
```

Results:
```
Found device: 68:27:37:56:DD:C3 - [AV] Samsung Soundbar MS650
Matches service UUID: False
Found device: 0A:9F:E3:B6:74:C9 - 0A-9F-E3-B6-74-C9
Matches service UUID: False
Found device: 00:1B:66:E0:80:F0 - MOMENTUM 3
Matches service UUID: False
```

This doesn't happen on Windows in my tests.

dlech, 2024-04-07:
> Unsure if this is relevant to this issue -

Not relevant, so i moved it to a new issue.