hbldh/bleak #1451: Get different advertisement value after connect to same peripheral multiple time

YunVisal, 2023-11-13:
* bleak version: 0.21.1
* Python version: 3.9.2
* Operating System: Raspbian GNU/Linux 11 (bullseye)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description
I'm trying to get the advertisement data from the peripheral. On the first attempt, I get the correct value. Then I connect to the peripheral using `BleakClient.connect()` function, read its characteristic's value and disconnect. In the next attempt, I always get the different advertisement value from the same peripheral.

- Advertisement Value on the first attempt:

```
AdvertisementData(local_name='Redmi Note 11', service_uuids=['36e4036d-b9b1-ff8a-3a37-e4ef97a5021a'], rssi=-72)
```

- Advertisement Value on the next attempt:
```
AdvertisementData(local_name='Redmi Note 11', service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '5b40bd67-7c36-4eed-d8ad-d87c53f56d68'], rssi=-71)
```

P.S: The `service_uuids` in the first attempt is the actual advertisement UUID, as for the second attempt, the `service_uuids` that I get are the array of GATT service UUIDs.

dlech, 2023-11-13:
I'm pretty sure I fixed a bug in BlueZ related to this, so I would try a newer BlueZ version.

dlech, 2023-11-15:
The latest release is actually 5.70 (https://github.com/bluez/bluez/releases). But the fixes I was referring to were released in 5.62. 

You could try deleting the BlueZ cache (https://bleak.readthedocs.io/en/latest/troubleshooting.html#id4). But beyond that, you would need to debug BlueZ to find out why it is doing that.

closed 2026-01-04 by github-actions[bot]