hbldh/bleak #1455: Could not read characteristic handle 17: Protocol Error 0x02: Read Not Permitted

snowuyl, 2023-11-15:
* bleak version: v0.21.1
* Python version: v3.11.5
* Operating System: Windows 11
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I run main.py and I encountered "Could not read characteristic handle 17: Protocol Error 0x02: Read Not Permitted".

### What I Did
1. Unzipping bleak-example.zip to D:\workspace_Python folder.
2. cd D:\workspace_Python\bleak-example
3. python3 main.py
4. Enter an option: C

### Logs
D:\workspace_Python\bleak-example>python3 main.py
Zephyr Bluetooth device
D7:B7:1D:7D:1E:19
D:\workspace_Python\bleak-example\main.py:103: FutureWarning: is_connected has been changed to a property. Calling it as an async method will be removed in a future version
  x = await client.is_connected()
manual
C. Capture
M. Measure
S. Stop measure
R. Reading result
Q. Quit

Enter an option: c
C
CAPTURE
manual
C. Capture
M. Measure
S. Stop measure
R. Reading result
Q. Quit

Enter an option: b'CAPTURE'
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xde\xddS\x00\x7f\xa3\x1b\x7f\xef\xac\x80\x13?\x7f\xe4,\x80\x13\x7f\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xc5\xae\x80\x17)\x80yx\x7f\xed\x96\x7f\xc9C\x7f\xda\xd8\x7f\xcc\x99')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80@\xbc\x80\n\x95\x7f\xdd\xd7\x80"S\x7f\xf1,\x7f\xf4\x16\x7f\xc6')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b"W\x80(\xff\x7f\xcf,\x80\'?\x7f\xc4\x02\x7f\xf7\x96\x80D\xd3\x80")
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x11?\x7f\xee\xd6\x7f\xb5\xad\x80\x00\x15\x7f\xfe\x95\x7f\xf0\xd6\x7f\xf6\xab')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\xea+\x7f\xf2\xd6\x7f\xdbW\x80\x07\xaa\x7f\xea\x01\x7f\xa3\x83\x7f\xbf')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x18\x80%\x94\x80\x14\xaa\x80\x0c\xeaS\x01\x7f\xc7.\x80\x05\xea\x7f\xea')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b",\x80\'>\x80\x11\xff\x80\x16?\x7f\xcfC\x7f\xb7\xc5\x7f\xe4\x82\x80")
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x16\xff\x7f\xf6\x81\x7f\xd3\xad\x7f\xccn\x7f\xd4m\x80:\xe7\x7f\xb00')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80(=\x7f\xcc.\x7f\xe0\xec\x7f\xed\x01\x7f\xdfl\x7f\xae\xc3\x7f\xcf')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x17\x7f\xe0\xd6\x7f\xf4\xc0\x7f\xb8\x03\x7f\xb4\x98\x7f\xd3\xac\x7f\x98D\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xf5\x80\x80\x16j\x80\x18\x94\x7f\xd4\x97\x80\x03\x00\x80\x0c\xc0\x7f\xc5\x82')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\x85o\x80\x01@\x7f\xce\x17\x7f\xbaB\x80\x08\xd5\x7f\xeeVS\x02')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\xf4\x16\x7f\xfak\x80\x0eT\x80\x05U\x7f\xf3\xeb\x7f\xd5m\x7f\xb4')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xda\x80Hf\x80,=\x80\x11\xea\x80\x17T\x7f\xff\xab\x7fnI\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xa3\xc6\x7f\xe0-\x7f\xdcX\x80Ef\x80\x0e*\x80\x08j\x80.\xd4')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7fi[\x7f\xf7\x80\x7f\xba\x02\x7f\xa0\xc3\x804\xe9\x7f\xeaA\x808')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xfe\x7f\xe1\x16\x7f\xc3B\x7f\xe9A\x7f\xee\xd6\x80\x04\xab\x7f\xd7A\x80')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'Lh\x7f\xfdU\x7f\xec\xd6\x7f\xfb@\x7f\x96.\x7f\xbd\x02\x80@\xd3')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80C>\x7f\xb2mS\x03\x7f\xf3\xd6\x80"\xd3\x80!\xd3\x7f\xc3\x99')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f|\xde\x80\x08\xbf\x7f\xf7k\x7f\xbb/\x80\x0c\xaa\x7f\xedW\x7f\xc6')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xae\x7f\xd6\x03\x80@g\x80\x05\x00\x802\xfd\x7f\xe4\xc2\x80\x03\x80\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xf9\x16\x80\x08\xd5\x7f\x81\xaf\x80 \x14\x80\x02\xd5\x7f\x8e\x84\x7f\xf4\xc0')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\xe5\xc1\x80")\x80\x01U\x80:\xfe\x804\xd3\x7f\xb5\xd8\x7f\xc5')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xad\x7f\x9e\x19\x7f\xc4W\x7f\xfa@\x7f\xe3\xd6\x7f\xd8\x81\x7f\xf4\xd6\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xf3\x96\x80(\xa9\x7f\x9dn\x80\x03\x95\x7f\xb2\xc3S\x04\x7f\xecA\x80')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'l\x8e\x80\x10?\x80\x02\x00\x7f\xfd\xeb\x7f\xb3\x9a\x7f\xcdn\x80Ye')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80\\\x90\x7f\xb5/\x7f\xbb\x9a\x7f\xf7\x01\x80O;\x7f\xef\x96\x7f\xd6')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'C\x7f\xca\xae\x80\x1f\x13\x80\x10T\x7f\xbf\x82\x806S\x7f\xb9-\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xcc\xd7\x80\x02\xeb\x7f\xf3\xc0\x80%i\x7f\xf1\xeb\x7f\xeb\xab\x80\x0f\x95')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\xc4m\x80\x14?\x80\x03\xc0\x80 T\x7f\xfd\x15\x7f\xdb,\x7f\xdd')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xac\x80\x07\x95\x80\r\xaa\x807\xd3\x80\x14\x95\x7f\xe1\xec\x7f\xfc\x80\x80')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'#)S\x05\x7f\xe2\x17\x80\x0c\xea\x7f\xe4\xc2\x7f\xf1\x01\x80\x0f\xaa\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xfe+\x804\xd2\x7f\xbbD\x7f\xf9\xeb\x7f\xbf/\x7f\xd0X\x80U%')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80/\xa8\x7f\xb9\xef\x80\x15\xe9\x80?Q\x7f\xb2\xb0\x80KQ\x7f\xfa')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xc0\x7f\xbf\xed\x80\x1cT\x80+T\x7f\xf5\x96\x80\x1b?\x7f\xed\x16\x80')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x1f\xbf\x7f\xc7\x02\x80\x0c\x15\x7f\xb1\xad\x7f\xf1\xd6\x7f\xf0+\x80-\xe9')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80\x0b\xd5\x80\x15*\x80\x15\xea\x7f\xf2\x96\x7f\xb6X\x7f\xed\xd6\x7f\xee')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xab\x80\x01@\x7f\xc8\xad\x7f\xde\xecS\x06\x80&\xfe\x7f\xf4\xeb\x7f\xf3')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b"+\x80\x1aT\x80\x0b\xff\x80 \xe9\x80=\'\x80\x11j\x7f\xff\xeb\x80")
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x0c\xff\x80@g\x7f\xf1A\x7f\xcc\x99\x80/}\x7f\xb0Z\x80\x05*')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\xff\x80\x80\x1d\xbe\x80\t\xd5\x80\x03\xd5\x7f\xd8\x81\x7f\xcb-\x80@')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x93\x7f\xc9\x17\x7f\xfe+\x7f\xd7,\x7f\xf7\x96\x7f\xea\x81\x7f\xee\x01\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xfcU\x7f\xf4+\x80\x07\xaa\x80\x05U\x7f\xf6V\x7f\xf8\x16\x7f\xcfl')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x7f\xbf\xc2\x80\r\x80\x80\x0e\xbf\x7f\xc5\x82\x7f\xd6,\x80:\x93S\x07')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\x80(=\x80!)\x80\x16\x7f\x808R\x7f\xeb\x97\x7f\xcaY\x80\x00')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'+\x80\x08\xbf\x80\nj\x7f\xd8\x98\x7f\xe1\xed\x7f\xe4\x82\x80mN\x7f')
6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 17): Nordic UART TX: bytearray(b'\xde\x82\x80@<\x80\x1d~\x80\x11\xaa\x7f\xefA\x7f\xd3\xec\x80&\xe9')
Could not read characteristic handle 17: Protocol Error 0x02: Read Not Permitted

snowuyl, 2023-11-15:
[bleak-example.zip](https://github.com/hbldh/bleak/files/13361028/bleak-example.zip)

snowuyl, 2023-11-16:
Which tool do you suggest to capture more log messages?

dlech, 2023-11-18:
https://bleak.readthedocs.io/en/latest/troubleshooting.html#capture-bluetooth-traffic

snowuyl, 2023-11-18:
Thanks for your reply!

closed 2025-09-14 by github-actions[bot]