hbldh/bleak #1456: dbus-fast error when trying to start scanning with BleakScanner

svhoy, 2023-11-15:
* bleak version: 0.21.1
* Python version: 3.11.2
* Operating System: Raspbian 11 
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.65

### Description

Hello, 

After I am upgraded Bleak (0.21.1) and dbus-fast (2.0.0 and 2.14.0) on my Raspberry Pi there are a issue when I like to start scanning for BLE Devices with BleakScanner. It seems something going wrong  when I am initialize the BleakScanner.

### What I Did

In my code I am just calling BleakScanner() and after that the only thing is the Error in Logs. 

```
async def scan(self, timeout: float = 10.0) -> list[BLEDevice]:
        _scanner = BleakScanner()
        return await _scanner.discover(timeout=timeout)
```

### Logs

```2023-11-15 18:39:59,959 - pi_socket - ERROR: Can not create Dataclass: dbus_fast.signature.Variant size changed, may indicate binary incompatibility. Expected 24 from C header, got 20 from PyObject```

Sorry that I can't provide more Logging Informations, when I am runnging my program with dbus-fast 1.84.2 I can see the whole Scanning progress in my log but with dbus-fast >=2.0.0 there is only thats what is comming.

Thanks for your help in advance.

dlech, 2023-11-15:
It sounds like this is a bug in dbus-fast so should be reported at https://github.com/Bluetooth-Devices/dbus-fast

svhoy, 2023-11-24:
I reported it to dbus-fast. Let's see what is comming.

svhoy, 2024-02-29:
I like to close this here. This simple workaround works: [dbus-fast issue](https://github.com/Bluetooth-Devices/dbus-fast/issues/270)

closed 2024-02-29 by svhoy