hbldh/bleak #1470: My program hangs when connecting laptop with device via BLE (Python, Windows, BleakClient, STM32)

mantisbug, 2023-11-30:
* bleak version: 0.21.1
* Python version: 3.11
* Operating System: Windows 10

### Description
The program runs continuously for several hours without a problem but, for no apparent reason, it suddenly hangs when connecting and I cannot catch the event to control it.

I suppose there is no way to prevent the comms problem; I need to catch the error and do something within the program to re-try the connection, instead of stop the program and rerun it.

Can you help, please?

### What I Did

The program is long to put it here. The comms bit is quite simple:

    logger.info('Connecting...')
    async with BleakClient(client_address) as client:
        logger.debug(f'Subscribing to characteristic [{nh}]')
        subscribed = await w.subscribe_to_characteristic(nh)
        ...

### Logs

30Nov2023 02:29:41 [: root :] INFO -> Connecting...
30Nov2023 02:29:41 [: bleak.backends.winrt.scanner :] DEBUG -> Received 80:E1:26:14:11:3D: .
30Nov2023 02:29:41 [: bleak.backends.winrt.scanner :] DEBUG -> Received 09:80:4C:6F:36:12: .
30Nov2023 02:29:41 [: bleak.backends.winrt.scanner :] DEBUG -> Received 80:E1:26:14:11:3D: *NAME_OF_MY_DEVICE_HERE*
30Nov2023 02:29:41 [: bleak.backends.winrt.scanner :] DEBUG -> 2 devices found. Watcher status: 3.
30Nov2023 02:29:41 [: bleak.backends.winrt.client :] DEBUG -> Connecting to BLE device @ 80:E1:26:14:11:3D
30Nov2023 02:29:41 [: bleak.backends.winrt.scanner :] DEBUG -> Received 80:E1:26:14:11:3D: .
30Nov2023 02:29:41 [: bleak.backends.winrt.client :] DEBUG -> getting services (service_cache_mode=None, cache_mode=None)...
30Nov2023 02:29:41 [: bleak.backends.winrt.client :] DEBUG -> calling get_gatt_services_async
30Nov2023 02:29:43 [: bleak.backends.winrt.client :] DEBUG -> session_status_changed_event_handler: id: BluetoothLE#BluetoothLE44:85:00:63:7e:cc-80:e1:26:14:11:3d, error: 0, status: 1
30Nov2023 02:29:43 [: bleak.backends.winrt.client :] DEBUG -> max_pdu_size_changed_handler: 156
30Nov2023 02:29:47 [: bleak.backends.winrt.client :] DEBUG -> returned from get_gatt_services_async
30Nov2023 02:29:47 [: bleak.backends.winrt.client :] DEBUG -> calling get_characteristics_async
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> 80:E1:26:14:11:3D: services changed
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> 80:E1:26:14:11:3D: services changed
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> returned from get_characteristics_async
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> calling get_descriptors_async
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> returned from get_descriptors_async
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> calling get_characteristics_async
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> 80:E1:26:14:11:3D: services changed
30Nov2023 02:29:48 [: bleak.backends.winrt.client :] DEBUG -> 80:E1:26:14:11:3D: services changed
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> returned from get_characteristics_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> calling get_descriptors_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> returned from get_descriptors_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> calling get_descriptors_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> returned from get_descriptors_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> calling get_descriptors_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> returned from get_descriptors_async
30Nov2023 02:29:49 [: bleak.backends.winrt.client :] DEBUG -> calling get_characteristics_async
30Nov2023 02:29:51 [: bleak.backends.winrt.client :] DEBUG -> returned from get_characteristics_async
30Nov2023 02:29:51 [: bleak.backends.winrt.client :] DEBUG -> calling get_descriptors_async
30Nov2023 02:29:51 [: bleak.backends.winrt.client :] DEBUG -> returned from get_descriptors_async
30Nov2023 02:29:51 [: bleak.backends.winrt.client :] DEBUG -> calling get_descriptors_async
30Nov2023 02:29:52 [: bleak.backends.winrt.client :] DEBUG -> returned from get_descriptors_async
30Nov2023 02:29:52 [: bleak.backends.winrt.client :] DEBUG -> calling get_characteristics_async
30Nov2023 02:29:53 [: bleak.backends.winrt.client :] DEBUG -> session_status_changed_event_handler: id: BluetoothLE#BluetoothLE44:85:00:63:7e:cc-80:e1:26:14:11:3d, error: 0, status: 0
30Nov2023 02:29:53 [: bleak.backends.winrt.client :] DEBUG -> max_pdu_size_changed_handler: 23
30Nov2023 02:29:53 [: bleak.backends.winrt.client :] DEBUG -> disposing service objects

mantisbug, 2023-11-30:
> I would suggest logging Bluetooth packets to try to see why it is disconnecting. https://bleak.readthedocs.io/en/latest/troubleshooting.html
> 
> Hard to say if it is behaving as expected or not without seeing a full program.

Thank you for answering, @dlech 

That log I put in my query comes from Bleak. 
The program has to connect to the device and once connected it works just fine. 
**Consistently** achieving the connection is the problem.

I have seen the logged error "**session_status_changed_event_handler: id:  ...**" in another program that tries to connect to the same kind of device, but that program is in C## and my colleague uses another driver.

RaphZufferey, 2024-01-08:
Hi! I am having a similar issue.
Bleak 0.21.1, Python 3.12, Windows 11.

My code hangs on the same line (async with BleakClient), and I don't know how to make sense of this issue. Same code runs on other Windows and MacOS machines.
- `2024-01-08 22:49:29,326 bleak.backends.winrt.scanner Thread-1 DEBUG: 6 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.Device detected. Seeed XIAO nRF58240 BLE`
- `2024-01-08 22:49:32,326 bleak.backends.winrt.client Thread-1 DEBUG: Connecting to BLE device @ C5:38:82:34:CC:D6`
- `2024-01-08 22:49:32,362 bleak.backends.winrt.client Thread-1 DEBUG: getting services (service_cache_mode=None, cache_mode=None)...`
- `2024-01-08 22:49:32,761 bleak.backends.winrt.client Dummy-4 DEBUG: session_status_changed_event_handler: id: BluetoothLE#BluetoothLE2c:6d:c1:8f:e8:d6-c5:38:82:34:cc:d6, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>`
- `2024-01-08 22:49:32,822 bleak.backends.winrt.client Dummy-5 DEBUG: max_pdu_size_changed_handler: 247`
- `2024-01-08 22:49:33,138 bleak.backends.winrt.client Dummy-5 DEBUG: C5:38:82:34:CC:D6: services changed`

Any help would be very much appreciated! Thanks in advance

twobit-five, 2024-01-14:
I had a similar issue where the python program would hang on 
await client.connect()

Sorry no logs or captures, but for me rolling back to 0.20.2 fixed my issue

dlech, 2024-01-14:
What is the difference between the working and not working Windows machines. For example to they have the same Bluetooth adapter and driver version and the same Windows updates?

tonieichelkraut, 2024-04-09:
I am experiencing the same problem. has there been any progress or work around? Thanks a lot!

dlech, 2024-04-09:
Without Bluetooth packet logs, it is impossible to tell what the problem is and if different people are actually having the same problem. 

Someone with the same symptoms was finally able to share Bluetooth packet logs and in https://github.com/hbldh/bleak/issues/1262#issuecomment-2041128015 we determined the problem for that specific case.

dlech, 2024-05-04:
Is this still a problem with Bleak v0.22.0?

lauvinson, 2024-06-25:
> Is this still a problem with Bleak v0.22.0?

yes I had this same phenomenon with 0.22.2

closed 2026-03-01 by github-actions[bot]