hbldh/bleak #1471: Fedora 39: [org.freedesktop.DBus.Error.NoReply] Remote peer disconnected

vovagorodok, 2023-11-30:
* bleak version: 0.21.1
* Python version: 3.12.0
* Operating System: Fedora Linux 39
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.70

### Description

When script trying to read from peripheral characteristics, flowing error occurs:
```
[org.freedesktop.DBus.Error.NoReply] Remote peer disconnected
```

### What I Did

Ble uploader trying to read information about device from characteristics.
Folowing script:
https://github.com/vovagorodok/ArduinoBleOTA/blob/main/tools/uploader.sh
executes `uploader.py`. 

Works correctly when:
1. Used Ubuntu instead Fedora
2. Used `bluezero` instead `bleak` (by chenging to  `bluezero_uploader.py` in uploader.sh)

Looks that issue is with `bleak` in Fedora 39 (trying in Lenovo T490 and HP Zbook 14u G6)

Please let me know what logs I should gather.
Thanks!

### Logs

N/A

dlech, 2023-12-01:
> Please let me know what logs I should gather.

https://bleak.readthedocs.io/en/latest/troubleshooting.html

> 1. Used Ubuntu instead Fedora

Is everything else the same in this case? Same computer? Same Bluetooth adapter? Same BlueZ version?

vovagorodok, 2023-12-03:
> Is everything else the same in this case? Same computer? Same Bluetooth adapter? Same BlueZ version?

Yes the same device (compuer and ble adapter). Before Fedora 39, Ubuntu 23.04 was installed, perhaps with older BlueZ version.
Checked in another computer with Fedora 39 and the issue exist.
Looks that issue is with bleak in Fedora 39.

```
Traceback (most recent call last):
  File "/home/tsiselsk/Projects/private/ArduinoBleOTA/tools/uploader.py", line 241, in try_scan_and_upload
    asyncio.run(scan_and_upload(path))
  File "/usr/lib64/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/asyncio/base_events.py", line 664, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/tsiselsk/Projects/private/ArduinoBleOTA/tools/uploader.py", line 236, in scan_and_upload
    await connect_and_upload(devices[device_num], path)
  File "/home/tsiselsk/Projects/private/ArduinoBleOTA/tools/uploader.py", line 191, in connect_and_upload
    res = await connect(dev)
          ^^^^^^^^^^^^^^^^^^
  File "/home/tsiselsk/Projects/private/ArduinoBleOTA/tools/uploader.py", line 115, in connect
    await acquire_mtu(client)
  File "/home/tsiselsk/Projects/private/ArduinoBleOTA/tools/uploader.py", line 83, in acquire_mtu
    await client._backend._acquire_mtu()
  File "/home/tsiselsk/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 586, in _acquire_mtu
    assert_reply(reply)
  File "/home/tsiselsk/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.NoReply] Remote peer disconnected
```

As I see issue is in `client._backend._acquire_mtu()`. Acquire mtu is needed to achieve maximal upload speed.
Should I attach logs for `export BLEAK_LOGGING=1`?

dlech, 2023-12-03:
> File "/home/tsiselsk/Projects/private/ArduinoBleOTA/tools/uploader.py", line 83, in acquire_mtu
>     await client._backend._acquire_mtu()

I think this could be what is causing the problem. This looks like `_acquire_mtu()` is from a 3rd-party library. Whatever that function is doing could be triggering the disconnect. 

Logging Bluetooth packets is useful too to see what is actually going on over the air.

vovagorodok, 2023-12-05:
Done: 
[bluetooth.zip](https://github.com/hbldh/bleak/files/13559399/bluetooth.zip)

dlech, 2023-12-06:
In the wireshark logs, it looks like it is starting to do some Bluetooth classic stuff after the Bluetooth low energy stuff. Is this a dual-mode device?

I'm not as familiar with those commands, so I don't know what it is supposed to look like, but I'm guessing BlueZ isn't getting an expected response or is getting a wrong response and disconnects.

dlech, 2023-12-06:
If it works with one BlueZ version but not in another, I would report the bug to BlueZ.

vovagorodok, 2023-12-06:
I also think it's probably a problem with BlueZ. Where issues with BlueZ should be reported?

dlech, 2023-12-06:
> Where issues with BlueZ should be reported?

https://github.com/bluez/bluez/ (or the Linux Bluetooth mailing list)

vovagorodok, 2024-02-11:
https://github.com/bluez/bluez/issues/670#issuecomment-1936526131
Looks that issue is somewhere between bleak and BlueZ just in latest Fedora 39. Only dbus is betwwen I gues?

vovagorodok, 2024-03-13:
@dlech have You idea how to workaround issue for this moment? Because I'm without working solution in Fedora.
Trying:
[image]
[image]
and have logs like:
```
Connecting to ArduinoBleOTA
[org.freedesktop.DBus.Error.NoReply] Remote peer disconnected
Client is already connected
```

dlech, 2024-03-13:
Issues with dual mode devices were discussed recently in #1521. I'm not sure what we can do about it besides fix it in BlueZ.

vovagorodok, 2024-03-14:
Created issue on Fedora side: https://bugzilla.redhat.com/show_bug.cgi?id=2269464