hbldh/bleak #1474: Pairing without passkey failed with Nordic nRF5340 device

snowuyl, 2023-12-05:
* bleak version: v0.21.1
* Python version: v3.11.6
* Operating System: Windows 11

### Description
I would like to implement pairing without passkey. But it failed to pair with Nordic nRF5340 device.

### What I Did
python3 pairing.py

### Logs
The following are log messages of RTT Viewer.
00> *** Booting Zephyr OS build v3.3.99-ncs1-1 ***
00> Starting BT test
00> [00:00:00.000,885] <err> bt_gatt: Unable to register handle 0x0016
00> [00:00:00.030,212] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
00> [00:00:00.030,242] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
00> [00:00:00.030,242] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 224.11902 Build 2231721665
00> [00:00:00.032,409] <inf> bt_hci_core: Identity: D7:D2:19:5F:4D:5D (random)
00> [00:00:00.032,440] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x2077, manufacturer 0x0059
00> [00:00:00.032,440] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0x2077
00> [00:00:00.032,470] <inf> bt: Bluetooth initialized
00> [00:00:09.599,456] <inf> bt: Connected D8:F8:83:9F:72:1A (public)
00> [00:00:09.601,531] <inf> bt: MTU=20
00> [00:00:09.601,531] <inf> bt: connected
00> [00:00:10.583,404] <wrn> bt_smp: Not connected!
00> [00:00:10.583,587] <err> bt: Security failed: D8:F8:83:9F:72:1A (public) level 1 err 9
00> [00:00:10.583,709] <err> bt: Pairing failed conn: D8:F8:83:9F:72:1A (public), reason 9
00> [00:00:10.583,984] <inf> bt: disconnected: D8:F8:83:9F:72:1A (public) (reason 19)

snowuyl, 2023-12-05:
[pairing.zip](https://github.com/hbldh/bleak/files/13558199/pairing.zip)

snowuyl, 2023-12-06:
The related source code of nRF5340 DK are listed below. And from log messages (Security failed: D8:F8:83:9F:72:1A (public) level 1 err 9). nRF5340 DK failed to change to BT_SECURITY_L2 (Encryption and no authentication (no MITM)). 
ret = bt_conn_set_security(conn, BT_SECURITY_L2);
if (ret < 0) {
        LOG_ERR("Failed to set security (err %d)\n", ret);
}
/** Security level. */
typedef enum __packed {
	/** Level 0: Only for BR/EDR special cases, like SDP */
	BT_SECURITY_L0,
	/** Level 1: No encryption and no authentication. */
	BT_SECURITY_L1,
	/** Level 2: Encryption and no authentication (no MITM). */
	BT_SECURITY_L2,
	/** Level 3: Encryption and authentication (MITM). */
	BT_SECURITY_L3,
	/** Level 4: Authenticated Secure Connections and 128-bit key. */
	BT_SECURITY_L4,
	/** Bit to force new pairing procedure, bit-wise OR with requested
	 *  security level.
	 */
	BT_SECURITY_FORCE_PAIR = BIT(7),
} bt_security_t;

snowuyl, 2023-12-06:
And nRF5340 DK can pair successfully with Device enumeration and pairing sample (https://learn.microsoft.com/en-us/samples/microsoft/windows-universal-samples/deviceenumerationandpairing/). The following are log messages of RTT Viewer.
00> *** Booting Zephyr OS build v3.3.99-ncs1-1 ***
00> Starting BT test
00> [00:00:00.000,885] <err> bt_gatt: Unable to register handle 0x0016
00> [00:00:00.030,212] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
00> [00:00:00.030,242] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
00> [00:00:00.030,273] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 224.11902 Build 2231721665
00> [00:00:00.032,409] <inf> bt_hci_core: Identity: D7:D2:19:5F:4D:5D (random)
00> [00:00:00.032,440] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x2077, manufacturer 0x0059
00> [00:00:00.032,470] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0x2077
00> [00:00:00.032,470] <inf> bt: Bluetooth initialized
00> [00:00:39.298,339] <inf> bt: Connected D8:F8:83:9F:72:1A (public)
00> [00:00:39.299,957] <inf> bt: MTU=20
00> [00:00:39.299,957] <inf> bt: connected
00> [00:00:40.115,325] <inf> bt: Pairing completed: D8:F8:83:9F:72:1A (public), bonded: 0
00> [00:00:40.115,509] <inf> bt: Security changed: D8:F8:83:9F:72:1A (public) level 2

bojanpotocnik, 2024-01-04:
In the `pairing.py` file you provided, you try to write to the CCCD before initiating pairing: 
```python
    async with BleakClient(device, disconnected_callback=handle_disconnect) as client:
        await client.start_notify(UART_TX_CHAR_UUID, handle_rx)
        paired = await client.pair(protection_level=2)
        print(f"Paired: {paired}")
```
This probably causes the pairing procedure to start implicitly (Just Works) using default protection level 1, as indicated in RTT logs
```plain
00> [00:00:10.583,404] bt_smp: Not connected!
00> [00:00:10.583,587] bt: Security failed: D8:F8:83:9F:72:1A (public) level 1 err 9
00> [00:00:10.583,709] bt: Pairing failed conn: D8:F8:83:9F:72:1A (public), reason 9
00> [00:00:10.583,984] bt: disconnected: D8:F8:83:9F:72:1A (public) (reason 19)
```