hbldh/bleak #1475: DBus.Error.AccessDenied Client tried to send a message other than Hello without being registered

amitbhanja, 2023-12-05:
* bleak version: 0.21.1
* Python version: 3.11.6
* Operating System: Debian GNU/Linux 11 (bullseye) , Raspberry Pi 4 
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description

I am running a python script which has to scan the ble devices in the surrounding.

We are running the script inside a docker container where following volumes are mounted

/run/dbus/system_bus_socket
/etc/dbus-1/system.d/bluetooth.conf

### What I Did

I am using the function BleakScanner.discover(timeout=4, return_adv=True)

We are able to get the ble devices for some time. May be aroundd 2-3 hours. 

But after that we are getting an error as described below.

Root user is running the python script inside the container.

Furthermore the pod is started by root user.

### Logs

[org.freedesktop.DBus.Error.AccessDenied] Client tried to send a message other than Hello without being registered

We have enabled the logs by setting BLEAK_LOGGING to 1.

tchern0, 2024-01-31:
Same issue here, only differences that i run the script without containers

OS: Linux raspberrypi 6.1.0-rpi7-rpi-v8 
BlueZ: 5.66
Python: Python 3.11.2
bleak version: 0.21.1

l-schilling, 2024-02-20:
Try searching for dbus with journalctl. I had the same issue, but the real error was from the dbus-daemon "The maximunm number of active connections for UID 1000 has been reaed (max_connections_per_user=256)"

This was caused by the BleakScanner class being instanced too often and not being shutdown properly by bleak or at least the dbus part. We solved this using an endless while loop scanning all the time.

dlech, 2024-02-21:
> This was caused by the BleakScanner class being instanced too often and not being shutdown properly by bleak or at least the dbus part. We solved this using an endless while loop scanning all the time.

When you had the problem, were you using `BleakScanner` with a context manager to ensure it closed properly?

gkecskes78, 2024-05-10:
I am facing this issue, when repeatedly connecting/disconnecting using context manager (see the code snippet below)

- bleak version: 0.22.1
- Python version: 3.10.12
- Operating System: Ubuntu 22.04.4 LTS, Raspberry Pi 4
- BlueZ version (bluetoothctl --version): 5.64

```
import asyncio, datetime
from bleak import BleakClient
address = "c6:11:17:2d:11:c3"

import logging
logger = logging.getLogger()
logger.setLevel(logging.INFO)
fh = logging.FileHandler("log_{:%Y-%m-%d %H:%M:%S}.txt".format(datetime.datetime.now()))
fh.setLevel(logging.INFO)
LogFormat = '%(asctime)s.%(msecs)03d %(levelname)s: %(message)s'
logging.basicConfig(format = LogFormat, datefmt='%Y-%m-%d %H:%M:%S', level = logging.INFO)
formatter = logging.Formatter(LogFormat, datefmt='%Y-%m-%d %H:%M:%S')
fh.setFormatter(formatter)
logger.addHandler(fh)

async def main(address):
    try:
        async with BleakClient(address) as client:
            await asyncio.sleep(5)
    except Exception as e:
        logger.info("Error:")
        logger.info(e)

i=1
while i<=1000:
    logger.info("Start loop " + str(i))
    asyncio.run(main(address))
    i += 1
```

How can I make this running ?

dlech, 2024-05-10:
We need a reproducible test case or logs showing what is going on under the hood to be able to understand the problem. Currently there is not enough information.

It is possible that `bluetoothd` or the DBus daemon is restarting while Bleak is running?

gkecskes78, 2024-05-14:
Use the code snippet as TC.
According syslog no restart occurs.

[log_2024-05-14 18:32:13.txt](https://github.com/hbldh/bleak/files/15311950/log_2024-05-14.18.32.13.txt)
[syslog.txt](https://github.com/hbldh/bleak/files/15311951/syslog.txt)

dlech, 2024-05-15:
I think the problem is that you are starting multiple run loops that don't run forever. Bleak isn't designed for this.

Change

```python
i=1
while i<=1000:
    logger.info("Start loop " + str(i))
    asyncio.run(main(address))
    i += 1
```

to 

```python
async def real_main():
    i=1
    while i<=1000:
        logger.info("Start loop " + str(i))
        await main(address)
        i += 1

asyncio.run(real_main())
```

dlech, 2024-05-15:
I've documented this now: https://bleak.readthedocs.io/en/latest/troubleshooting.html#calling-asyncio-run-more-than-once

Closing the issue since it is caused by improper usage.

Thanks @gkecskes78 for providing the reproducible test case.

closed 2024-05-15 by dlech