hbldh/bleak #1481: client.write_gatt_char() does not work on Win11

ghisbo, 2023-12-16:
* bleak==0.21.1
bleak-winrt==1.2.0
* Python version: 3.11.7
* Operating System: Win11
 

### Description

I want to write a bytearray to a gatt characteristic. I can read from all the characterisitcs but when writing, the error message  "Access Denied" appears.

### What I Did
when i use nRF connect, i can write the characteristics. I suppose it is the Bleak version (win Api?) .
I notice that the error indicates the **handle** and not the characteristic ( is that a problem ? )
# bleak.exc.BleakError: Could not write value b'\x02\x08' to characteristic 000F: Access Denied**

Test program below:

import asyncio
import logging
import struct

from bleak import BleakClient, BleakScanner
from bleak.backends.characteristic import BleakGATTCharacteristic

logger = logging.getLogger(__name__)

async def notification_handler(characteristic: BleakGATTCharacteristic, data: bytearray):
    """Simple notification handler which prints the data received."""
    print(data)

    

async def main():
    logger.info("starting scan...")

    device = await BleakScanner.find_device_by_name(
            'XXXXXXXXXXXXr', cb=dict(use_bdaddr=False)
        )
    if device is None:
        logger.error("could not find device with name")
        return
    async with BleakClient(device) as client:
        # value = await client.read_gatt_char("XXXXXXXXXXXX")
        # print(value)
        # value = await client.read_gatt_char("XXXXXXXXXXXX")
        # print(value)
        await client.write_gatt_char("XXXXXXXXXXXX')
        
        
# Exception has occurred: BleakError
# Could not write value b'\x02\x08' to characteristic 000F: Access Denied
#   File "E:\axinGolf\axinBle\testBleWrite.py", line 32, in main
#     await client.write_gatt_char("842d2697-eeaa-4af2-a111-b001db824e30",b'\x02\x08')
#   File "E:\axinGolf\axinBle\testBleWrite.py", line 39, in <module>
#     asyncio.run(main())
# bleak.exc.BleakError: Could not write value b'\x02\x08' to characteristic 000F: Access Denied

if __name__ == "__main__":
   
    asyncio.run(main())`

```
# Exception has occurred: BleakError
# Could not write value b'\x02\x08' to characteristic 000F: Access Denied
#   File "E:\axinGolf\axinBle\testBleWrite.py", line 32, in main
#     await client.write_gatt_char("842d2697-eeaa-4af2-a111-b001db824e30",b'\x02\x08')
#   File "E:\axinGolf\axinBle\testBleWrite.py", line 39, in <module>
#     asyncio.run(main())
# bleak.exc.BleakError: Could not write value b'\x02\x08' to characteristic 000F: Access Denied

```

### Logs

dlech, 2023-12-18:
It depends on what `XXXXXXXXXXXX` is. Windows blocks access to certain characteristics, like HID characteristics.

ghisbo, 2023-12-18:
a guuid generated online
await client.write_gatt_char("842d2697-eeaa-4af2-a111-b001db824e30",b'\x02\x08') it is esp32 controlling a  imu sensor

dlech, 2023-12-18:
In that case, I would log Bluetooth packets to see if the error is coming from the ESP32. Maybe pairing is required?

ghisbo, 2023-12-24:
Pairing did not change this. I insalled wireshark but i see no ble packets. I have no experience with this so any advice on tracing packets is welcome as well.

dlech, 2023-12-24:
Instructions are at https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

ghisbo, 2023-12-25:
I do not have access to the Traduci board.
However, i tested from Mac and there i can read and write. It looks a Windows 11 problem.

tlustosp, 2024-03-24:
Hi, did you found a way to fix that? I am having the same problem.

ghisbo, 2024-03-26:
Sorry to hear that. 
I stopped looking for a solution in Windows and continue on Mac. In the end it has to run on mobile so i did not pursue the windows path.

closed 2025-12-28 by github-actions[bot]