hbldh/bleak #1482: The second request cannot accept the message

tsiens, 2023-12-17:
* bleak version:0.21.1
* Python version:3.10.0
* Operating System:python

### Description
``` python
async def connect(addr,char):
    def notification_handler(_,r):
        print(r)
    async with BleakClient(addr,timeout=10) as client:
        await client.start_notify(char, notification_handler)
        await asyncio.sleep(3)
        await client.stop_notify(char)

while True:
      asyncio.run(connect(addr,char))
      if input(f'conitue')!='y':
          break
```

When I first connect, I can receive notifications，But when I press ‘y‘ for the second run, I can't see any output.
I switched to a visualization software and it's the same.
After restarting the Bluetooth device, you can receive messages.
I think it's my first time not really ending the connection, and I've been occupying it all along.
What should I do？
tks

dlech, 2023-12-17:
`asyncio.run()` should only be used once per program. There is an `aioconsole` library that you can use to get input in a non-blocking way if you need to.

tsiens, 2023-12-18:
> `asyncio.run()` should only be used once per program. There is an `aioconsole` library that you can use to get input in a non-blocking way if you need to.
Thank you for your prompt response，That's the problem with my code.
For the aioconsole , I couldn't find sufficient documentation on Google. How should I rewrite it based on my functional requirements? Thank you

tsiens, 2023-12-18:
> `asyncio.run()` should only be used once per program. There is an `aioconsole` library that you can use to get input in a non-blocking way if you need to.

```
from bleak import BleakClient,BleakScanner
import asyncio
async def scan():
    print(await BleakScanner.discover())

while True:
    asyncio.run(scan())
    if input(f'conitue')!='':
        break
```

"asyncio.run() should only be used once per program."，Why can I run the above code repeatedly