hbldh/bleak #1484: Cannot identify how to fetch an updated device name after connection

coderjoe, 2023-12-25:
* bleak version: 0.21.1
* Python version: 3.10.12
* Operating System: Ubuntu 22.04.3 Linux 5.15.0-91-generic
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description

I am working with a device that advertises under one device name, but upon successful connection and protocol negotiation it changes its device name to another name.

The mobile app I'm trying to emulate appears to make a second request to the Generic Access Profile service (0x1800) for Device Name (0x2a00). Unfortunately bleak does not appear to expose that service so I cannot make my own requests.

Per closed issue  #120 I believe it was said that this data was wrapped up in the BLEDevice, but I am not sure how to get the updated device name once I have connected.

My debug logs show that DBus picks up the property change, but I do not know how to fetch the updated property.

### What I Did

I am emulating the android application by performing a bluetooth device scan and looking for advertised devices by either a known address or product name.

Once connected, I listen for device updates on its RX characteristic, and perform a connection negotiation by sending a vendor specific command over its TX characteristic.

However the device name never gets updated even though bluez reports the correct change.

At timestamp `2023-12-25 15:18:28` the `device.name` output is `HPA250B` which is the advertised device name
At timestamp `2023-12-25 15:18:28,936` output in the logs bluez reports the property change to the new name of `HEPA 250 JB`

But I cannot figure out how to reference this updated name in bleak.

```python3
import argparse
import asyncio
import logging

from uuid import UUID
from bleak import BleakClient, BleakScanner
from bleak.backends.characteristic import BleakGATTCharacteristic

logger = logging.getLogger(__name__)

address = "f4:5e:ab:bb:a8:94"
RX_UUID = UUID('{0000ffe4-0000-1000-8000-00805f9b34fb}')
TX_UUID = UUID('{0000ffe9-0000-1000-8000-00805f9b34fb}')

def notification_handler(characteristic: BleakGATTCharacteristic, data: bytearray):
    display_data("\n\n\nNotify from device: %s", data)

async def main(args: argparse.Namespace):
    logger.info("\n\n\nstarting find...")

    device = await BleakScanner.find_device_by_address(address)
    logger.info("\n\n\nDevice: %s", device)
    logger.info("\n\n\nDevice Details: %s", device.details)

    if device == None:
        logger.error("Could not find a matching device. Aborting...")
        return

    logger.info("\n\n\nconnecting to device...")

    async with BleakClient(device) as client:
        logger.info("\n\n\nConnected to address %s!", client.address)
        await client.start_notify(RX_UUID, notification_handler)

        mac_cmd = b'MAC+\xf4\x5e\xab\xbb\xa8\x94'
        logger.info("\n\n\nSending MAC Command: %s", mac_cmd)
        await client.write_gatt_char(TX_UUID, mac_cmd)

        await asyncio.sleep(10)
        logger.info("\n\n\nDevice: %s", device)
        logger.info("\n\n\nDevice Details: %s", device.details)
[... 57 lines cut ...]

Sending MAC Command: b'MAC+\xf4^\xab\xbb\xa8\x94'
2023-12-25 15:18:29,649 bleak.backends.bluezdbus.client DEBUG: Write Characteristic 0000ffe9-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_F4_5E_AB_BB_A8_94/service0023/char0024: b'MAC+\xf4^\xab\xbb\xa8\x94'
2023-12-25 15:18:31,180 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\xa5\x00\x00\x00 \x00\x00\x91@\x86@\x00\x00\t\x00\x17;\x00\x00'))>}, []]
2023-12-25 15:18:31,181 __main__ INFO: 

Notify from device: a5 00 00 00 20 00 00 91 40 86 40 00 00 09 00 17 3b 00 00
2023-12-25 15:18:39,658 __main__ INFO: 

Device: F4:5E:AB:BB:A8:94: HPA250B
2023-12-25 15:18:39,658 __main__ INFO: 

Device Details: {'path': '/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94', 'props': {'Address': 'F4:5E:AB:BB:A8:94', 'AddressType': 'public', 'Name': 'HPA250B', 'Alias': 'HPA250B', 'Paired': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], 'Modalias': 'bluetooth:v000Dp0000d0110', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'RSSI': -48, 'TxPower': 0}}
2023-12-25 15:18:39,659 bleak.backends.bluezdbus.client DEBUG: Disconnecting ({/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94})
2023-12-25 15:18:42,392 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94): ['org.bluez.Device1', {'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, []]
2023-12-25 15:18:42,393 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]
2023-12-25 15:18:42,393 bleak.backends.bluezdbus.client DEBUG: Device disconnected (/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94)
2023-12-25 15:18:42,393 bleak.backends.bluezdbus.client DEBUG: _cleanup_all(/org/bluez/hci0/dev_F4_5E_AB_BB_A8_94)
```

### Logs

See above program output for debug logs.

In particular timestamp `2023-12-25 15:18:28` for the advertised device name and timestamp `2023-12-25 15:18:28,936` for the bluez property update debug log.

coderjoe, 2023-12-26:
Updated to reference the correct other issue, meant #250 not 150

dlech, 2023-12-26:
You can probably read the device name characteristic. Also see https://github.com/hbldh/bleak/issues/849#issuecomment-1255370264 about a proposed enhancement for a `name` property on the `BleakClient` object.

coderjoe, 2023-12-29:
Thank you @dlech I appreciate the suggestion, but I could not figure out how to query the Device Name characteristic from within bleak. Attempts to reference it by UUID give me problems as the Generic Access Profile does not seem listed as an available service.

Is there some other way to query it without being able to find the service via `bleak`?

coderjoe, 2023-12-29:
Ok no - the problem was on my end, teh format of the UUID I was passing was valid for the UUID library, but a typo in my code had it being interpreted as a string. As a result the error 

`bleak.exc.BleakError: Characteristic with UUID {00002a00-0000-1000-8000-00805f9b34fb} could not be found!`

Had me believing that I could not read that characteristic. That was wrongminded - it works fine when I create my UUID properly.

My mistake - thank you for your suggestion as it eventually lead me to the correct solution!

closed 2023-12-29 by coderjoe