hbldh/bleak #1489: bleak.exc.BleakError: failed to discover services, device disconnected

bisudw, 2024-01-04:
* bleak version: 0.21.1
* Python version: 3.10
* Operating System: Ubuntu
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description

I have a Vernier Caliper from TESA which has a BLE module and it gives data from that BLE Module. I am trying to read the data from the BLE Module. I am ble to read it in My Phone using LightBlue App (a ble scanner app).

### What I Did

When I am rnning the Service_Explorer I am able to discover the device and the advertised service UUID is also visible. Its the same service UUID that i get in my phone app.

It looks like this:
```
E0:4B:16:4A:55:EA: TLC-BLE
--------------------------
AdvertisementData(local_name='TLC-BLE', manufacturer_data={1341: b'\x0800530322\x069E0445'}, service_uuids=['6e400001-b5a3-f393-e0a9-e50e24dcca9e'], tx_power=4, rssi=-49)
```

But when I am trying to connect to the same BLE device I am getting error which says:

```
bleak.exc.BleakError: failed to discover services, device disconnected
```

here is the code which I am using to trying to connect:

```
import asyncio
from bleak import BleakClient

async def main():
    async with BleakClient("E0:4B:16:4A:55:EA") as client:
        # Read a characteristic, etc.
        print("Connected")

    # Device will disconnect when block exits.
    print("disconnect")

# Using asyncio.run() is important to ensure that device disconnects on
# KeyboardInterrupt or other unhandled exception.
asyncio.run(main())

```

### Logs

```
Traceback (most recent call last):
  File "/home/biswa/PycharmProjects/BLEScanner/main.py", line 15, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/biswa/PycharmProjects/BLEScanner/main.py", line 6, in main
    async with BleakClient("E0:4B:16:4A:55:EA") as client:
  File "/home/biswa/PycharmProjects/BLEScanner/.venv/lib/python3.10/site-packages/bleak/__init__.py", line 565, in __aenter__
    await self.connect()
  File "/home/biswa/PycharmProjects/BLEScanner/.venv/lib/python3.10/site-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
  File "/home/biswa/PycharmProjects/BLEScanner/.venv/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 268, in connect
    await self.get_services(
  File "/home/biswa/PycharmProjects/BLEScanner/.venv/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 656, in get_services
    self.services = await manager.get_services(
  File "/home/biswa/PycharmProjects/BLEScanner/.venv/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 644, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/home/biswa/PycharmProjects/BLEScanner/.venv/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 779, in _wait_for_services_discovery
    raise BleakError("failed to discover services, device disconnected")
bleak.exc.BleakError: failed to discover services, device disconnected

```

dlech, 2024-01-04:
Can you run the same program with debug logging enabled? If that doesn't provide enough info you may need to log Bluetooth packets as well.

https://bleak.readthedocs.io/en/latest/troubleshooting.html

closed 2024-04-28 by dlech

hussainaloum, 2025-01-23:
I'm having the same issue above. I used the same code above, just changing the MAC address. This issue doesn't always happen, sometimes the devices connects okay.

My setup:

bleak version: 0.21.1
Python version: 3.12.3
Operating System: Ubuntu 24.04.1 LTS
BlueZ version (bluetoothctl -v) in case of Linux: 5.72

Output I get when running the script:

> Traceback (most recent call last):
  File "/home/aliang/Downloads/ATEST/main.py", line 143, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/aliang/Downloads/ATEST/main.py", line 134, in main
    async with BleakClient("F0:F4:AF:DE:B7:44") as client:
  File "/usr/lib/python3/dist-packages/bleak/__init__.py", line 565, in __aenter__
    await self.connect()
  File "/usr/lib/python3/dist-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/client.py", line 268, in connect
    await self.get_services(
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/client.py", line 656, in get_services
    self.services = await manager.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 644, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 779, in _wait_for_services_discovery
    raise BleakError("failed to discover services, device disconnected")
bleak.exc.BleakError: failed to discover services, device disconnected

With debug logs enabled

> 025-01-23 11:04:00,137 bleak.backends.bluezdbus.client MainThread DEBUG: Connecting to device @ F0:F4:AF:DE:B7:44
2025-01-23 11:04:00,143 bleak.backends.bluezdbus.manager MainThread DEBUG: initial properties: {'/org/bluez': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.AgentManager1': {}, 'org.bluez.ProfileManager1': {}, 'org.bluez.HealthManager1': {}}, '/org/bluez/hci0': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': '04:CF:4B:79:D2:D5', 'AddressType': 'public', 'Name': 'aliang-alice', 'Alias': 'aliang-alice', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0548', 'Roles': ['central', 'peripheral'], 'Manufacturer': 2, 'Version': 12}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.BatteryProviderManager1': {}, 'org.bluez.GattManager1': {}, 'org.bluez.Media1': {'SupportedUUIDs': ['0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb']}, 'org.bluez.NetworkServer1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 12, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name'], 'SupportedSecondaryChannels': ['1M', '2M', 'Coded']}}, '/org/bluez/hci0/dev_F0_64_10_6F_90_7E': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': 'F0:64:10:6F:90:7E', 'AddressType': 'random', 'Name': 'CJ5014', 'Alias': 'CJ5014', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {65535: bytearray(b'\x07d(\x00\x03\x03\x0c\x00\x01\x00\x02\x03\x04\x00\x01')}, 'ServicesResolved': False}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_EC_54_3C_56_A1_66': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': 'EC:54:3C:56:A1:66', 'AddressType': 'random', 'Name': 'C100EE', 'Alias': 'C100EE', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {65535: bytearray(b'\x08d7\x00\x04\x00\x0c\x00\x00\x00\x02\x03\x05\x00\xf1')}, 'ServicesResolved': False}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_EF_32_0D_D8_F6_82': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': 'EF:32:0D:D8:F6:82', 'AddressType': 'random', 'Alias': 'EF-32-0D-D8-F6-82', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: bytearray(b'\x12\ [... cut ...]
    self.services = await manager.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 644, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 779, in _wait_for_services_discovery
    raise BleakError("failed to discover services, device disconnected")
bleak.exc.BleakError: failed to discover services, device disconnected

dlech, 2025-01-23:
> bleak version: 0.21.1

The fix was released in Bleak v0.22.0

dakhnod, 2025-01-26:
I also had issues with timeouts. What helped me is setting the BLE timeout to 20 vial `BleakClient(address, timeout=20)`.

For me, bluez was reporting a connected device, but was taking its time with service discovery, as it seems. Not sure why it only started happening some weeks ago...

PR #1544 (2024-04-28): Fix _wait_removed completion on invalid object path
The _wait_removed method registers the "InterfacesRemoved" callback on the adapter path of the device without checking which object was removed when called.
This means that any removed interface while a connection is being established can cause the _wait_removed to complete and cancel the connection.

This commit simply checks that the callback is for the proper device object path.

Fixes #1489
commit 5b445874 (2024-04-28): Fix _wait_removed completion on invalid object path (#1544)

* Fix _wait_removed completion on invalid object path

The _wait_removed method registers the "InterfacesRemoved" callback on
the adapter path of the device without checking which object was
removed when called.
This means that any removed interface while a connection is
being established can cause the _wait_removed to complete and cancel
the connection.

This commit simply checks that the callback is for the proper device
object path.

Fixes #1489
PR #1562 (2024-05-04): Release/0.22.0

Added
-----
* Added ``BleakCharacteristicNotFoundError`` which is raised if a device does not support a characteristic.
* Added utility function to work around ``pywin32`` setting threading model to STA on Windows.

Changed
-------
* Updated PyObjC dependency on macOS to v10.x.
* Updated missing Bluetooth SIG characteristics and service UUIDs.
* Updated ``BlueZManager`` to remove empty interfaces from `_properties` during InterfacesRemoved message.
* Updated PyWinRT dependency to v2. Fixes #1529.
* Raise exception when trying to scan while in a single-treaded apartment (STA) on [... cut]