hbldh/bleak #1491: Discovery of Services works with CoreBluetooth on MacOS, fails on Debian

wleese, 2024-01-05:
* bleak version:  0.21.1
* Python version: 3.11.2
* Operating System: Linux rpi400 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
* BlueZ version: bluetoothctl: 5.66

### Description

Running the script https://github.com/hbldh/bleak/blob/master/examples/service_explorer.py modified just to list services, shows all service UUIDs when run on MacOS. However when run on Raspbian (rpi400), zero services are found.

### What I Did

Ran this command on MacOS and then on Raspbian:

MacOS
```
python3 service_explorer.py --name "Klipsch The Three Plus" -d

2024-01-05 21:21:16,065 asyncio  DEBUG: Using selector: KqueueSelector
2024-01-05 21:21:16,065 __main__ INFO: starting scan...
2024-01-05 21:21:16,292 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManagerDidUpdateState_
2024-01-05 21:21:16,292 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
2024-01-05 21:21:16,292 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
2024-01-05 21:21:16,663 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,664 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 0AE652C9-4651-6288-0410-6B0855E01D78: [TV] Samsung 8 Series (55) @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTimestamp', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,682 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,682 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 0AE652C9-4651-6288-0410-6B0855E01D78: [TV] Samsung 8 Series (55) @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTimestamp', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,683 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,684 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 0AE652C9-4651-6288-0410-6B0855E01D78: [TV] Samsung 8 Series (55) @ RSSI: -78 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataLocalName', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTimestamp', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,693 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,694 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 1E2FA667-2C91-45FB-82D1-064D6E7D9798: Klipsch The Three Plus @ RSSI: -71 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataManufacturerData', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTimestamp', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,702 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,702 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,703 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 7E18D186-6F78-0241-6628-9984E092CA3A: None @ RSSI: -74 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,704 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 7E18D186-6F78-0241-6628-9984E092CA3A: None @ RSSI: -74 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,705 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,705 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 9ADDCA39-21A7-C2A5-8744-95ED6DA0599A: iphone (100) @ RSSI: -56 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,705 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,706 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 9ADDCA39-21A7-C2A5-8744-95ED6DA0599A: iphone (100) @ RSSI: -56 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,714 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,714 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,715 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 8D53E5AE-CC07-B5D2-DEC3-2D4561066BC8: Living Room @ RSSI: -74 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,715 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 8D53E5AE-CC07-B5D2-DEC3-2D4561066BC8: Living Room @ RSSI: -74 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,801 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,802 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 1E2FA667-2C91-45FB-82D1-064D6E7D9798: Klipsch The Three Plus @ RSSI: -76 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataManufacturerData', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTimestamp', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,803 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2024-01-05 21:21:16,803 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 1E2FA667-2C91-45FB-82D1-064D6E7D9798: Klipsch The Three Plus @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600002b80000>
2024-01-05 21:21:16,804 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
2024-01-05 21:21:16,804 __main__ INFO: connecting to device...
2024-01-05 21:21:16,816 bleak.backends.corebluetooth.client DEBUG: CentralManagerDelegate  at <CentralManagerDelegate: 0x145faff80>
2024-01-05 21:21:16,816 bleak.backends.corebluetooth.client DEBUG: Connecting to BLE device @ 1E2FA667-2C91-45FB-82D1-064D6E7D9798
2024-01-05 21:21:17,032 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didConnectPeripheral_
2024-01-05 21:21:17,033 bleak.backends.corebluetooth.client DEBUG: Retrieving services...
2024-01-05 21:21:17,325 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverServices_
2024-01-05 21:21:17,325 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Services discovered
[... cut ...]
[working_service_discovery_bleak_corebluetooth.zip](https://github.com/hbldh/bleak/files/13846244/working_service_discovery_bleak_corebluetooth.zip)

I've also been unable to list service uuids with bluetoothctl on raspbian.

Next to MacOS, I also installed LineageOS/Android on the Raspberry 400. The official application to connect to this device worked. Here's the btsnoop file I captured using Androids builtin bug report feature, also readable by Wireshark.
[btsnoop_hci.log.zip](https://github.com/hbldh/bleak/files/13846299/btsnoop_hci.log.zip)

dlech, 2024-01-06:
Thanks for the logs. This looks like a BlueZ issue since the wireshark logs show that the device is being enumerated (I didn't see anything suspicious there) but BlueZ isn't passing on the info to Bleak (as seen in the debug logs - there should be InterfacesAdded signals for all of the services and characteristics).

So I would suggest enabling debug logging in BlueZ and see what you can find there.

wleese, 2024-01-06:
Thanks for the swift response.

Here's the debug logs from starting bluetooth with debug logging and then running the script which simply scans and tries to discover services:

```
Jan 06 08:34:04 rpi400 dbus-daemon[517]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.39' (uid=1000 pid=1331 comm="/usr/bin/wireplumber")
Jan 06 08:34:04 rpi400 systemd[1]: bthelper@hci0.service: Deactivated successfully.
Jan 06 08:34:04 rpi400 systemd[1]: Stopped bthelper@hci0.service - Raspberry Pi bluetooth helper.
Jan 06 08:34:04 rpi400 systemd[1]: Stopping bthelper@hci0.service - Raspberry Pi bluetooth helper...
Jan 06 08:34:04 rpi400 systemd[1]: Starting hciuart.service - Configure Bluetooth Modems connected by UART...
Jan 06 08:34:04 rpi400 systemd[1]: hciuart.service: Deactivated successfully.
Jan 06 08:34:04 rpi400 systemd[1]: Started hciuart.service - Configure Bluetooth Modems connected by UART.
Jan 06 08:34:04 rpi400 systemd[1]: Starting bthelper@hci0.service - Raspberry Pi bluetooth helper...
Jan 06 08:34:05 rpi400 bthelper[4962]: Raspberry Pi BDADDR already set
Jan 06 08:34:05 rpi400 systemd[1]: Finished bthelper@hci0.service - Raspberry Pi bluetooth helper.
Jan 06 08:34:05 rpi400 (uetoothd)[5074]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Jan 06 08:34:05 rpi400 systemd[1]: Starting bluetooth.service - Bluetooth service...
Jan 06 08:34:05 rpi400 bluetoothd[5074]: Bluetooth daemon 5.66
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “AlwaysPairable” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “JustWorksRepairing” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “TemporaryTimeout” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “RemoteNameRequestRetryDelay” in group “General”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “Cache” in group “GATT”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “KeySize” in group “GATT”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “ExchangeMTU” in group “GATT”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “Channels” in group “GATT”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “SessionMode” in group “AVDTP”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “StreamMode” in group “AVDTP”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_config() Key file does not have key “RSSISamplingPeriod” in group “AdvMon”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “PageScanType” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “PageScanInterval” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “PageScanWindow” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “InquiryScanType” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “InquiryScanInterval” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “InquiryScanWindow” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “LinkSupervisionTimeout” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “PageTimeout” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “MinSniffInterval” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “MaxSniffInterval” in group “BR”
Jan 06 08:34:05 rpi400 bluetoothd[5074]: src/main.c:parse_mode_config() Key file does not have key “MinAdvertisementInterval” in group “LE”
[... cut ...]
Jan 06 08:34:14 rpi400 bluetoothd[5074]: src/device.c:att_disconnected_cb() Software caused connection abort (103)
Jan 06 08:34:14 rpi400 bluetoothd[5074]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
Jan 06 08:34:14 rpi400 bluetoothd[5074]: src/device.c:att_disconnected_cb() Automatic connection disabled
Jan 06 08:34:14 rpi400 bluetoothd[5074]: src/gatt-database.c:btd_gatt_database_att_disconnected()
Jan 06 08:34:14 rpi400 bluetoothd[5074]: attrib/gattrib.c:g_attrib_unref() 0x55878924f0: g_attrib_unref=0
```

dlech, 2024-01-06:
It looks like BlueZ doesn't give the exact reason for the error, but I'm guessing there is something about some information it receives from the device that it doesn't like. It tends to be more strict than other Bluetooth stacks. I would suggest posting the issue on BlueZ's github to try to get more insight.

wleese, 2024-01-06:
@dlech  I'll create an issue for BlueZ. If you have a Kofi or any other way for me to buy you a coffee/beer, let me know.
 Thanks so far!

closed 2024-01-06 by wleese

dlech, 2024-01-07:
https://github.com/sponsors/dlech  :cookie: :milk_glass: