hbldh/bleak #1493: Characteristic with UUID could not be found

Honusnap, 2024-01-09:
* bleak version: 0.21
* Python version: 3.8.14
* Operating System: VenusOS 3.13 (linux)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description
I'm using an other project called dbus-serialbattery, it use Bleak API to connect to bluetooth devices. I've narrowed the error at this command 
client = BleakClient(3C:A5:50:81:D7:C6)

Error => **Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!**
###

I then created 2 test scripts ( BleakTest1.py and  BleakTest2.py) with your 2 examples here : https://bleak.readthedocs.io/en/latest/usage.html and have included my own address for the "address" constant.

root@einstein:~# python3 BleakTest1.py
Traceback (most recent call last):
  File "BleakTest2.py", line 12, in <module>
    asyncio.run(main(address))
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "BleakTest2.py", line 8, in main
    async with BleakClient(address) as client:
  File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 565, in __aenter__
    await self.connect()
  File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 249, in connect
    assert_reply(reply)
  File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort
root@einstein:~#

running BleakTest2.py return  => Characteristic with UUID 2A24 could not be found!

Using "bluetoothctl connect 3C:A5:50:81:D7:C6" work and return a "connected" message
There is a PIN code for this device and it has been paired succesfully in bluetoothctl 
### Logs

dlech, 2024-01-09:
Bleak is meant to be run as a normal user, not root.

If you run the service explorer example, is the 2A24 characteristic listed?

Honusnap, 2024-01-09:
> Bleak is meant to be run as a normal user, not root.
> 
> If you run the service explorer example, is the 2A24 characteristic listed?

I can't see it, here is the output : 

root@einstein:~# python3 service_explorer.py --address 3C:A5:50:81:D7:C6
2024-01-09 17:19:12,083 __main__ INFO: starting scan...
2024-01-09 17:19:13,569 __main__ INFO: connecting to device...
2024-01-09 17:19:14,768 __main__ INFO: connected
2024-01-09 17:19:14,770 __main__ INFO: [Service] 0000ffe0-0000-1000-8000-00805f9b34fb (Handle: 8): Vendor specific
2024-01-09 17:19:14,772 __main__ INFO:   [Characteristic] 0000ffe3-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific (write-without-response,write)
2024-01-09 17:19:14,806 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2024-01-09 17:19:14,807 __main__ INFO:   [Characteristic] 0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 9): Vendor specific (write-without-response,write,notify)
2024-01-09 17:19:14,836 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 11): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2024-01-09 17:19:14,837 __main__ INFO:   [Characteristic] 0000ffe2-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific (write-without-response,write,notify)
2024-01-09 17:19:14,866 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 14): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2024-01-09 17:19:14,868 __main__ INFO: disconnecting...
2024-01-09 17:19:17,257 __main__ INFO: disconnected

Honusnap, 2024-01-09:
> 2A24

This is the VendorString right ? Does it mean that if this device do not have this characteristic prevent from using it in Bleak ?
I can connect to this bluetooth device from 2 different PC (windows / home assistant), from my Phone (Android) with no problem.

I used btmon, is this log usefull ?

Bluetooth monitor ver 5.55
= Note: Linux version 5.10.109-venus-12 (armv7l)                       0.678686
= Note: Bluetooth subsystem version 2.22                               0.678700
= New Index: DC:1A:B7:C6:96:78 (Primary,UART,hci1)              [hci1] 0.678705
= Open Index: DC:1A:B7:C6:96:78                                 [hci1] 0.678707
= Index Info: DC:1A:B7:C6:96:78 (Broadcom Corporation)          [hci1] 0.678708
= New Index: 68:4E:05:4A:51:A3 (Primary,USB,hci0)               [hci0] 0.678711
= Open Index: 68:4E:05:4A:51:A3                                 [hci0] 0.678713
= Index Info: 68:4E:05:4.. (Realtek Semiconductor Corporation)  [hci0] 0.678714
@ MGMT Open: bluetoothd (privileged) version 1.18             {0x0001} 0.678718
@ MGMT Command: Disconnect (0x0014) plen 7             {0x0001} [hci0] 2.231098
        LE Address: 3C:A5:50:81:D7:C6 (OUI 3C-A5-50)
< HCI Command: Disconnect (0x01|0x0006) plen 3               #1 [hci0] 2.231207
        Handle: 16
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                    #2 [hci0] 2.232164
      Disconnect (0x01|0x0006) ncmd 2
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4               #3 [hci0] 2.240171
        Status: Success (0x00)
        Handle: 16
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Command Complete (0x0001) plen 10        {0x0001} [hci0] 2.240255
      Disconnect (0x0014) plen 7
        Status: Success (0x00)
        LE Address: 3C:A5:50:81:D7:C6 (OUI 3C-A5-50)
> HCI Event: Command Complete (0x0e) plen 4                  #4 [hci0] 2.241149
      Vendor (0x3f|0x0019) ncmd 3
        Status: Success (0x00)
@ MGMT Command: Start Service Disco.. (0x003a) plen 4  {0x0001} [hci0] 3.398554
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        RSSI: invalid (0x7f)
        UUIDs: 0
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7   #5 [hci0] 3.398709
        Type: Active (0x01)
        Interval: 22.500 msec (0x0024)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                  #6 [hci0] 3.400140
      LE Set Scan Parameters (0x08|0x000b) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2       #7 [hci0] 3.400232
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                  #8 [hci0] 3.401123
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
< HCI Command: Inquiry (0x01|0x0001) plen 5                  #9 [hci0] 3.401183
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                   #10 [hci0] 3.402116
      Inquiry (0x01|0x0001) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4         {0x0001} [hci0] 3.402184
      Start Service Discovery (0x003a) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
[... 528 lines cut ...]
          BR/EDR Not Supported
        Company: Victron Energy BV (737)
          Data: 100010a1017f7c94840d938dd7fbd1a1df798615
        RSSI: -40 dBm (0xd8)
> HCI Event: LE Meta Event (0x3e) plen 26                   #82 [hci0] 8.844106
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Random (0x01)
        Address: C9:DF:8D:06:A1:D5 (Static)
        Data length: 14
        Name (complete): MPPT 450/100
        RSSI: -40 dBm (0xd8)
@ MGMT Event: Device Found (0x0012) plen 55            {0x0001} [hci0] 8.844152
        LE Address: C9:DF:8D:06:A1:D5 (Static)
        RSSI: -40 dBm (0xd8)
        Flags: 0x00000000
        Data length: 41
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Company: Victron Energy BV (737)
          Data: 100010a1017f7c94840d938dd7fbd1a1df798615
        Name (complete): MPPT 450/100
root@einstein:~#

Honusnap, 2024-01-10:
> Then either the device doesn't have that characteristic or BlueZ is hiding it. You could log Bluetooth packets to see if the device actually provides this characteristic.

Hello,
I've replaced the device with a newer version of this same device and .. it work. Would you like me to get some more info on this non working bluetooth device to improve connection to those devices ?

mr-manuel, 2024-02-23:
@Honusnap if you can proceed with this others would be very thankful!

https://github.com/Louisvdw/dbus-serialbattery/issues/819

mr-manuel, 2024-02-23:
Here some data from the BlueZ (version 5.55) dbus from a JKBMS where "00002a24-0000-1000-8000-00805f9b34fb" is not found. This indicates that the data is already missing in BlueZ right?

```bash
manuel@raspberrypi2:~# dbus-send --system --dest=org.bluez --print-reply /org/bluez/hci0/dev_20_22_08_25_17_41 org.freedesktop.DBus.Properties.GetAll string:"org.bluez.Device1"
method return time=1708694825.738213 sender=:1.225 -> destination=:1.284 serial=102 reply_serial=2
   array [
      dict entry(
         string "Address"
         variant             string "20:22:08:25:17:41"
      )
      dict entry(
         string "AddressType"
         variant             string "public"
      )
      dict entry(
         string "Name"
         variant             string "JK_BD4A24S04"
      )
      dict entry(
         string "Alias"
         variant             string "JK_BD4A24S04"
      )
      dict entry(
         string "Paired"
         variant             boolean false
      )
      dict entry(
         string "Trusted"
         variant             boolean true
      )
      dict entry(
         string "Blocked"
         variant             boolean false
      )
      dict entry(
         string "LegacyPairing"
         variant             boolean false
      )
      dict entry(
         string "Connected"
         variant             boolean true
      )
      dict entry(
         string "UUIDs"
         variant             array [
               string "0000ffe0-0000-1000-8000-00805f9b34fb"
            ]
      )
      dict entry(
         string "Adapter"
         variant             object path "/org/bluez/hci0"
      )
      dict entry(
         string "ManufacturerData"
         variant             array [
               dict entry(
                  uint16 19274
                  variant                      array of bytes [
                        00 01
                     ]
               )
            ]
      )
      dict entry(
         string "ServicesResolved"
         variant             boolean true
      )
      dict entry(
         string "AdvertisingFlags"
         variant             array of bytes [
               06
            ]
      )
   ]

manuel@raspberrypi2:~# bluetoothctl info 20:22:08:25:17:41
Device 20:22:08:25:17:41 (public)
        Name: JK_BD4A24S04
        Alias: JK_BD4A24S04
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        ManufacturerData Key: 0x4b4a
        ManufacturerData Value:
  00 01                                            ..
        AdvertisingFlags:
  06                                               .
```