hbldh/bleak #1495: asyncio.exceptions.TimeoutError connecting to BLE (WinRT)

kierenj, 2024-01-19:
* bleak version: 0.21.1
* Python version: 3.9.4
* Operating System: Window 11 64-bit
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I'm trying to connect to a BLE device, SP107E. My bluetooth-scanning app on my phone works correctly, but Bleak times out.

### What I Did

```
import asyncio
from bleak import BleakClient, discover

device_name = "SP107E"
service_uuid = "FFB0"
characteristic_uuid = "0000ffe1-0000-1000-8000-00805f9b34fb"
command = b'\x00\x00\x00\x01'

async def main():
    devices = await discover()
    device = next((d for d in devices if d.name == device_name), None)

    async with BleakClient(device, services=[service_uuid], timeout=60.0) as client:
        print(f"listening...")
        await client.start_notify(characteristic_uuid, notification_handler)
        print(f"sending command")
        await client.write_gatt_char(characteristic_uuid, command)
        print("Command sent")
        await asyncio.sleep(30) 
        await client.stop_notify(characteristic_uuid)

asyncio.run(main())
```

I've tried:
- Not filtering by service
- Larger timeouts
- Checking with another app (LightBlue on my iPhone)

### Logs

Wireshark capture: [sp107e-pcapng.zip](https://github.com/hbldh/bleak/files/13991964/sp107e-pcapng.zip)

```
2024-01-19 16:48:25,199 bleak.backends.winrt.client MainThread DEBUG: Connecting to BLE device @ 62:20:04:01:AF:29
2024-01-19 16:48:25,259 bleak.backends.winrt.client MainThread DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2024-01-19 16:48:25,360 bleak.backends.winrt.client Dummy-1 DEBUG: session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe8:48:b8:c8:20:00-62:20:04:01:af:29, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
2024-01-19 16:48:25,422 bleak.backends.winrt.client Dummy-2 DEBUG: max_pdu_size_changed_handler: 131
2024-01-19 16:48:25,662 bleak.backends.winrt.client Dummy-1 DEBUG: 62:20:04:01:AF:29: services changed
2024-01-19 16:48:35,269 bleak.backends.winrt.client MainThread DEBUG: closing requester
2024-01-19 16:48:35,270 bleak.backends.winrt.client MainThread DEBUG: closing session
Traceback (most recent call last):
  File "C:\git\mpg\venv\lib\site-packages\bleak\backends\winrt\client.py", line 480, in connect
    self.services = await self.get_services(
  File "C:\git\mpg\venv\lib\site-packages\bleak\backends\winrt\client.py", line 724, in get_services
    await FutureLike(
  File "C:\git\mpg\venv\lib\site-packages\bleak\backends\winrt\client.py", line 1122, in __await__
    yield self  # This tells Task to wait for completion.
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\git\mpg\main.py", line 35, in <module>
    asyncio.run(main())
  File "C:\Python39\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\git\mpg\main.py", line 26, in main
    async with BleakClient(device, services=[service_uuid]) as client:
  File "C:\git\mpg\venv\lib\site-packages\bleak\__init__.py", line 565, in __aenter__
    await self.connect()
  File "C:\git\mpg\venv\lib\site-packages\bleak\__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
  File "C:\git\mpg\venv\lib\site-packages\bleak\backends\winrt\client.py", line 480, in connect
    self.services = await self.get_services(
  File "C:\git\mpg\venv\lib\site-packages\async_timeout\__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "C:\git\mpg\venv\lib\site-packages\async_timeout\__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

dlech, 2024-01-20:
From the logs, it looks like the device never responded to the read request for Peripheral Preferred Connection Parameters. So this is likely why Windows is timing out.

[image]

kierenj, 2024-01-21:
Ok, thank you - I will try different adapters and tools on this machine to see if I can find any differences. It does just seem like the WinRT call is timing out. Thanks for taking a look at the trace 🙂

dlech, 2024-05-04:
Is this still a problem with Bleak v0.22.0?