hbldh/bleak #1500: Multiple clients crashes bluetooth stack: Bluetooth: hci0: Opcode 0x200e failed: -16

zevv, 2024-02-02:
- raspberry pi 4B
- debian bookworm
- bleak 0.20.2-1 from `python3-bleak` package
- bluetoothctl: 5.66

### Description

I have a simple python script that has the task to poll a large number of devices (50 to 100). I want to optimize the process by not serializing all connects + polls + disconnects, see below.

The result is that the bluetooth stack on my rpi crashes. Clients are no longer disconnecting and scanning never terminates.

kernel logs show:

```
[  193.752537] Bluetooth: hci0: Opcode 0x200e failed: -16
[  205.529410] Bluetooth: hci0: Opcode 0x200e failed: -16
[  210.648622] Bluetooth: hci0: Opcode 0x200e failed: -16
```
Not able to power off/on as well:

Also tested with an external USB bluetooth dongle, same results.

```
[Biro 07]# power off
[Biro 07]# power on
Failed to set power on: org.bluez.Error.Busy
```

### What I Did

, the procedure I have now is:

```
while true:
  start scanning 
  scan for 30 seconds, collect devices in callback
  stop scanning
  for 1 minute:
    pick the oldest polled device
    await connect
    await read data
    await disconnect
```

This works pretty much ok but is rather slow as there is only one active client connection at a time.

I understand that doing multiple connects in parallel is no recommended (yes I have looked at the two-devices example), but having multiple connections established should be fine. This is what I changed my code to now, effectively this serializes the `connect()s`, but allows the polling and disconnect to run async:

```
while true:
  start scanning 
  scan for 30 seconds, collect devices in callback
  stop scanning
  for 1 minute:
    pick the oldest polled device
    await connect
    create async process that does:
       await read_data
       await disconnect
```

As far as I know, this should abide to all the rules:

- do not connect while scanning
- do not perform connects in parallel

I also tried to postpone the scanning until all the active clients have disconnecetd, but to no avail.

Minimal workable example is not easy since there is a large number of specific devices involved.

### Logs

`BLEAK_LOGGING=1` log attached (sorry for the newlines, this was copy/pasted from a scrollback buffer)

[debug-log.txt](https://github.com/hbldh/bleak/files/14140445/debug-log.txt)

### Code

Below is the code as I run it now, albeit stripped a bit from the protocol details that are not relevant:

```
#!/usr/bin/python

from bleak import BleakScanner, BleakClient
import logging
import contextlib
import traceback
import os
import asyncio
import signal
import sys
import json
import logging
import time
import struct
from time import sleep

SERVICE_UUID = "0000ffe0-0000-1000-8000-00805f9b34fb"
CHAR_HANDLE = "0000ffe1-0000-1000-8000-00805f9b34fb"

class JK:

    def __init__(self, scanner, dev, address, name):
        self.scanner = scanner
        self.dev = dev
        self.client = BleakClient(self.dev)
        self.address = address
        self.name = name
        self.t_last_poll = time.time() - 60 * 10

    async def reg_wr(self, add, vals: bytearray, length: int):
        # write something to device
        pass

    def ncallback(self, sender: int, data: bytearray):
        # handle data
        pass

    async def read(self):
        await self.client.start_notify(CHAR_HANDLE, self.ncallback)
        await self.reg_wr(COMMAND_DEVICE_INFO, b"\0\0\0\0", 0x00)
        self.poll_ready = False
        while not self.poll_ready:
            await asyncio.sleep(0.1)
        await self.client.disconnect()
        self.scanner.decref()

    async def poll(self):
        self.scanner.incref()
        client = self.client
        await client.connect()
    
        # Awaiting works fine:
        # await self.read()
    
        # But running in an async tast not:
        asyncio.create_task(self.read())

class JKviewer:

    def __init__(self):
        self.devices = {}
        self.npolling = 0

    def incref(self):
        self.npolling += 1

    def decref(self):
        self.npolling -= 1

    async def poll(self):
            
        # Find the jk with the oldest t_last_poll
        jk = None
        t_oldest = t_now
        for address in self.devices.keys():
            jk2 = self.devices[address]
            if jk2.t_last_poll < t_oldest:
                t_oldest = jk2.t_last_poll
                jk = jk2

        if jk == None:
            print("nothing to do")
            return

        # Poll oldest JK
        async def aux():
            jk.t_last_poll = time.time()
            await jk.poll()
            self.busy = None
        asyncio.create_task(aux())

    async def start(self):

        def on_scan(dev, data):
            if SERVICE_UUID in data.service_uuids:
                if dev.address not in self.devices:
                    jk = JK(self, dev, dev.address, dev.name)
                    self.devices[dev.address] = jk
                if dev.address in self.devices:
                    self.devices[dev.address].t_last_seen = time.time()

        while True:
            # scan for a few seconds
            print("scanning")
            async with BleakScanner(on_scan) as scanner:
                await asyncio.sleep(3)
            # work the devices
            print("working")
            for i in range(20):
                for i in range(10):
                    await self.poll()
                    await asyncio.sleep(0.1)
            # wait for all the connections to be gone
            print("waiting")
            while self.npolling > 0:
                await asyncio.sleep(0.5)

jkviewer = JKviewer()
asyncio.run(jkviewer.start())

# vi: ft=python ts=4 sw=4 et
```

dlech, 2024-02-02:
If things are going wrong in the kernel, there isn't much we can do in Bleak. I would suggest logging Bluetooth traffic with `btmon` and reporting the issue to the folks that wrote the kernel driver for whichever bluetooth adapter you are using.

Siecje, 2024-02-13:
> As far as I know, this should abide to all the rules:

>    do not connect while scanning
>    do not perform connects in parallel

Where do these rules come from?

> do not perform connects in parallel

Is this covered by always using `await client.connect()`?

In your code it is possible that the event loop runs `await client1.connect()` followed by `await client2.connect()`. With each operation in a different task.

zevv, 2024-02-13:
> Where do these rules come from?

I think I gathered these from various comments and examples throughout the project. Please correct me if any of those are wrong or not true.

> Is this covered by always using await client.connect()?

Yes. As far as I understand my own code, `await client1.connect()` needs to finish before `await client2.connect()` can ever happen; the main poller keeps track of any in-progress connects with the ` self.busy` flag. Only *after* `await client1.connect()` finishes `JK.,poll` will return, and another client connect can be tried.

Anyway, I have abandoned any attempts to parallelize my polling, as I'm already having a hard time to get this to work reliably in a strictly serial way, see issue #1507...

Siecje, 2024-02-13:
What does `self.busy` do? It is only set to `None` in your snippet.

Your snippet does the following.

```
start = [await for poll] -> add aux to event loop -> repeat start
aux = [await for poll] -> await connect -> add read to event loop
read = [await notify, await write, await disconnect]
```

It's possible for aux to be in the event loop more than once, that means that multiple tasks can be on `await connect` and then after multiple connections can be open at the same time.

zevv, 2024-02-13:
I apologize, I minimized my code to share the example in this issue and accidentally took out too much code; the `busy` flag was set to a value when creating the client poll task. I updated the above code.

Like I said, I should have simplified this example even more, the problem also occurs without running the asyncio task from the `JKviewer.poll()` function.

Having said all this: I'm still sure any abuse or wrong API usage from userspace should result in the kernel/driver/firmware getting confused beyond repair, but that's another store.

Thanks,,

Siecje, 2024-02-13:
`self.busy` will show the name of the current device but it can also be set multiple times without being `None`. It can be set to a name when it is currently a name.

> the problem also occurs without running the asyncio task from the JKviewer.poll() function.

Can you modify your code snippet to be the minimal example that shows the problem?

dlech, 2024-02-14:
FYI, calling `asyncio.create_task()` in the `poll()` methods without keeping a handle to the returned task will allow the task to be garbage collected before the task completes. See https://docs.python.org/3.12/library/asyncio-task.html#asyncio.create_task

zevv, 2024-02-14:
> FYI, calling `asyncio.create_task()` in the `poll()` methods without keeping a handle to the returned task will allow the task to be garbage collected before the task completes. 

Right, I'll have to take care of that.

> Polling with `await asyncio.sleep(0.1)` also kind of defeats the purpose of asyncio. There are better alternatives like asyncio events, futures and queues depending on what you are trying to do.

Sure, I'm aware of that - my example is just lousy.

Shall I close this issue and reopen if/when I reproduce with a simpler and more compact program?

dlech, 2024-02-14:
> Shall I close this issue and reopen if/when I reproduce with a simpler and more compact program?

Sounds good. So far, this sounds more like a hardware issue than a Bleak issue.

Siecje, 2024-02-14:
Is it a problem if multiple tasks `await client.connect()`?

For example, event loop runs task1 which runs `await client1.connect()` which yields to the event loop and then the event loop runs tasks2 which runs `await client2.connect()`?

zevv, 2024-02-14:
> Is it a problem if multiple tasks await client.connect()?

Yes, it seems that is a reliable way to trigger this problem. My example code was messy though, so if I find the time to get this running again I'll come up with a more simple snippet.

Closed for now.

Siecje, 2024-02-14:
Should that work though?

Why doesn't it work? Because the backend can't handle multiple connection requests at the same time?

If that's the case can `BleakClient` ensure only one connect is requested to the backend at a time?

How do backends handle multiple processes call connect at the same time?

zevv, 2024-02-14:
I'm not sure what causes the limitation, but the "two clients" examples jumps through hoops to make sure only one `connect()` is running at a time:

https://github.com/hbldh/bleak/blob/develop/examples/two_devices.py#L40-L41

dlech, 2024-02-14:
> Why doesn't it work? Because the backend can't handle multiple connection requests at the same time???

It can be an OS limitation like BlueZ returning "operation already in progress" errors. Or it can be a hardware limitation like the issue here.

> If that's the case can `BleakClient` ensure only one connect is requested to the backend at a time?

I would rather leave it up to the user in case they want a different behavior, like getting a busy error instead of just waiting a indefinite amount of time. If we changed it now, I think it would break a lot of people's code that depend on the current behavior.

> How do backends handle multiple processes call connect at the same time?

They probably don't handle it well since it was never really designed to do that (since it often doesn't work at the OS level anyway).

Siecje, 2024-02-14:
Thanks for explaining and sorry if I missed the documentation on this.

> like getting a busy error

Is that what happens? Why not handle a busy error instead of using the lock in the two_devices.py example?

dlech, 2024-02-14:
The busy error is specific to the BlueZ backend. The lock works cross-platform - and avoids the error in the first place.

Siecje, 2024-02-15:
> The busy error is specific to the BlueZ backend. The lock works cross-platform - and avoids the error in the first place.

It only avoids the error if another process doesn't use Bluetooth.

Siecje, 2024-02-15:
>  If we changed it now, I think it would break a lot of people's code that depend on the current behavior.

There can be a new class with the new behaviour and when you use the new class you opt in to the new behaviour.

LyWilson, 2025-03-20:
I dont know if this is related, but i have a problème with my iot project, so i use a rasp 4 to read data sent by bluetooth from other things. but after 30-40 min i stop getting data, and if i reboot it work for another 30-40 min and need a reboot. I found out that i got a similare error too. *last time i received data was 18:14:04

(venv) lrima@raspberrypi:~/iot_obj-sicro-sensor/code/central $ dmesg | grep -i bluetooth
journalctl -u bluetooth --no-pager | tail -n 20
[    1.602252] usb 1-1.1: Product: Bluetooth Radio
[    4.204865] Bluetooth: Core ver 2.22
[    4.204936] NET: Registered PF_BLUETOOTH protocol family
[    4.204940] Bluetooth: HCI device and connection manager initialized
[    4.204953] Bluetooth: HCI socket layer initialized
[    4.204958] Bluetooth: L2CAP socket layer initialized
[    4.204966] Bluetooth: SCO socket layer initialized
[    4.247071] Bluetooth: HCI UART driver ver 2.3
[    4.247086] Bluetooth: HCI UART protocol H4 registered
[    4.247153] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    4.252634] Bluetooth: HCI UART protocol Broadcom registered
[    4.514829] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    4.517847] Bluetooth: hci1: RTL: rom_version status=0 version=1
[    4.517874] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[    4.530147] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_config.bin
[    4.534882] Bluetooth: hci1: RTL: cfg_sz 6, total sz 30210
[    4.618407] Bluetooth: hci0: BCM: chip id 107
[    4.618987] Bluetooth: hci0: BCM: features 0x2f
[    4.620088] Bluetooth: hci0: BCM4345C0
[    4.620116] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    4.644389] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,4-model-b.hcd' Patch
[    5.002862] Bluetooth: hci1: RTL: fw version 0xdfc6d922
[    5.425348] Bluetooth: hci0: BCM: features 0x2f
[    5.426632] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    5.426646] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    5.592116] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.592133] Bluetooth: BNEP filters: protocol multicast
[    5.592145] Bluetooth: BNEP socket layer initialized
[    5.597250] Bluetooth: MGMT ver 1.22
[    5.930050] Bluetooth: MGMT ver 1.22
[   11.443929] Bluetooth: RFCOMM TTY layer initialized
[   11.443957] Bluetooth: RFCOMM socket layer initialized
[   11.443975] Bluetooth: RFCOMM ver 1.11
[   31.558652] Bluetooth: hci1: Opcode 0x200e failed: -16
[   90.138954] Bluetooth: hci1: Opcode 0x200e failed: -16
[   99.816249] Bluetooth: hci1: Opcode 0x200e failed: -16
[  182.978044] Bluetooth: hci1: Opcode 0x200e failed: -16
[  195.259850] Bluetooth: hci1: Opcode 0x200e failed: -16
[  228.329471] Bluetooth: hci1: Opcode 0x200e failed: -16
[  282.770323] Bluetooth: hci1: Opcode 0x200e failed: -16
[  301.716905] Bluetooth: hci1: Opcode 0x200e failed: -16
[  307.312739] Bluetooth: hci1: Opcode 0x200e failed: -16
[  321.626763] Bluetooth: hci1: Opcode 0x200e failed: -16
[  326.696915] Bluetooth: hci1: ACL packet for unknown connection handle 16
[  327.415529] Bluetooth: hci1: Opcode 0x200e failed: -16
[  340.416482] Bluetooth: hci1: Opcode 0x200e failed: -16
[  379.573030] Bluetooth: hci1: Opcode 0x200e failed: -16
[  455.522741] Bluetooth: hci1: Opcode 0x200e failed: -16
[  534.179975] Bluetooth: hci1: Opcode 0x200e failed: -16
[  554.792890] Bluetooth: hci1: Opcode 0x200e failed: -16
[  565.755384] Bluetooth: hci1: Opcode 0x200e failed: -16
[  570.770453] Bluetooth: hci1: ACL packet for unknown connection handle 16
[  584.599370] Bluetooth: hci1: Opcode 0x200e failed: -16
[  612.415019] Bluetooth: hci1: Opcode 0x200e failed: -16
[  628.339264] Bluetooth: hci1: Opcode 0x200e failed: -16
[  632.120766] Bluetooth: hci1: Opcode 0x200e failed: -16
[  633.358880] Bluetooth: hci1: ACL packet for unknown connection handle 16
[  658.693873] Bluetooth: hci1: Opcode 0x200e failed: -16
[  712.207176] Bluetooth: hci1: Opcode 0x200e failed: -16
[  740.416255] Bluetooth: hci1: Opcode 0x200e failed: -16
[  804.995415] Bluetooth: hci1: Opcode 0x200e failed: -16
[  933.987966] Bluetooth: hci1: Opcode 0x200e failed: -16
[ 1060.988717] Bluetooth: hci1: Opcode 0x200e failed: -16
[ 1090.602208] Bluetooth: hci1: Opcode 0x200e failed: -16
[ 1127.834161] Bluetooth: hci1: Opcode 0x200e failed: -16
[ 1132.895510] Bluetooth: hci1: ACL packet for unknown connection handle 16
[ 1137.533587] Bluetooth: hci1: Opcode 0x200e failed: -16
[ 1162.816250] Bluetooth: hci1: Opcode 0x200e failed: -16
Mar 20 18:12:13 raspberrypi bluetoothd[471]: No matching connection for device
Mar 20 18:12:42 raspberrypi bluetoothd[471]: No matching connection for device
Mar 20 18:13:32 raspberrypi bluetoothd[471]: No matching connection for device
Mar 20 18:13:55 raspberrypi bluetoothd[471]: No matching connection for device
Mar 20 18:14:15 raspberrypi bluetoothd[471]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
Mar 20 18:14:15 raspberrypi bluetoothd[471]: Unable to create object for found device 00:00:00:00:00:00
Mar 20 18:14:15 raspberrypi bluetoothd[471]: src/device.c:device_new() Unable to register device interface for C6:72:16:C2:01:2A
Mar 20 18:14:15 raspberrypi bluetoothd[471]: Unable to create object for found device C6:72:16:C2:01:2A
Mar 20 18:14:15 raspberrypi bluetoothd[471]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
[... cut ...]
Mar 20 18:14:16 raspberrypi bluetoothd[471]: Unable to create object for found device 00:00:00:00:00:00
Mar 20 18:14:17 raspberrypi bluetoothd[471]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
Mar 20 18:14:17 raspberrypi bluetoothd[471]: Unable to create object for found device 00:00:00:00:00:00
Mar 20 18:14:17 raspberrypi bluetoothd[471]: src/device.c:device_new() Unable to register device interface for 00:00:00:00:00:00
Mar 20 18:14:17 raspberrypi bluetoothd[471]: Unable to create object for found device 00:00:00:00:00:00

kariudo, 2026-08-17:
It is definitely still a problem and should not be closed as stale without a resolution.

[image]