hbldh/bleak #1501: Wait notification functionality feature

vovagorodok, 2024-02-04:
Will be nice to have functionality like `client.wait_for_notify()` in order to have possibility to write simple scripts of serial communication that works like:
```
await client.wait_for_notify()
await client.read_gatt_char(tx_char)
```
I have this issue in my ArduinoBleOTA library for firmware uploading over ble: 
https://github.com/vovagorodok/ArduinoBleOTA/blob/7fb9e079cf54e02b0701d117754e14ebe5db34cf/tools/uploader.py#L144

dlech, 2024-02-04:
As a workaround, you could use an `asyncio.Event()`. The notification callback would need to call `event.set()`, then you would do:

```python
await event.wait()
await client.read_gatt_char(tx_char)
event.clear()
```

vovagorodok, 2024-02-04:
As I see callback has `bytearray` as second argument. It means that value from characteristics is being read during callback and `await client.read_gatt_char(tx_char)` will read from characteristics second time?

dlech, 2024-02-04:
Based on your example, I thought that maybe you had a case where the value was too large to fit in a notification and you had to do a read to get the full value. But if you just want to use the value from the notification, then you can use `asyncio.Queue() instead of `asyncio.Event()` and leave out the read.

vovagorodok, 2024-03-13:
Thanks! I will use `asyncio.Queue()` as solution. There are any plans to add method like `client.wait_for_notify()` or solution with `asyncio.Queue()` looks ok from Your PoV?

vovagorodok, 2024-03-13:
Should we close this issue or hold in order to track progress?

dlech, 2024-03-14:
I don't see any duplicate issues, so let's leave this one open.