hbldh/bleak #1505: Timeout exception connecting to BLE plug - bluetoothctl works.

baudneo, 2024-02-10:
* bleak version:
0.21.1
* Python version:
3.10.12 and 3.11.6
* Operating System:
Ubuntu Jammy and Arch
* BlueZ version (`bluetoothctl -v`) in case of Linux:
5.66 and 5.72

### Description
I have tested this on 2 different machines. Each machine has a different bt adapter and different versions of bluez/kernels.

I can connect to other BT devices using bleak, it is just this specififc device. The device is a bluetooth/wifi telink device; Cync by GE smart plug. I can get a connection using `bluetoothctl` but I can not get a connection using bleak. 
There are a few different errors, I detail 3 of them below in logging.

### What I Did
The issue is with `BleakClient(mac_addr).connect()`, using a context manager and without. I am assuming this is some sort of bad device implementation? I am trying to get good `bluetoothctl`, `btmon` and pcaps for each error.

### Logs

Here is my script output with `BLEAK_LOGGING=1`. I will detail a few different errors with this device.

#### Timeout
The plug connects, throws some errors and disconnects
```log
02/10/24 11:30:40.0603 DEBUG - acync.mesh mesh:408 -> telink mesh:connect: attempt: 1/3 to MAC (4/4): 34:13:43:70:D9:2E [Cync Plug] - Scan Timeout: 10
2024-02-10 11:30:40,603 bleak.backends.bluezdbus.client MainThread DEBUG: Connecting to device @ 34:13:43:70:D9:2E
2024-02-10 11:30:40,646 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
2024-02-10 11:30:40,747 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_C0_23_8D_D6_3E_AA', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', C0:23:8D:D6:3E:AA)>, 'AddressType': <dbus_fast.signature.Variant ('s', public)>, 'Name': <dbus_fast.signature.Variant ('s', [TV] Samsung TU7000 50 TV)>, 'Alias': <dbus_fast.signature.Variant ('s', [TV] Samsung TU7000 50 TV)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -96)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {117: <dbus_fast.signature.Variant ('ay', bytearray(b'B\x04\x01\x80f\xc0#\x8d\xd6>\xaa\xc2#\x8d\xd6>\xa9\x01\x00\x00\x00\x00\x00\x00'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2024-02-10 11:30:40,747 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_78_6D_EB_28_EA_A3', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 78:6D:EB:28:EA:A3)>, 'AddressType': <dbus_fast.signature.Variant ('s', public)>, 'Name': <dbus_fast.signature.Variant ('s', 7B19400144A4D0F4)>, 'Alias': <dbus_fast.signature.Variant ('s', 7B19400144A4D0F4)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -58)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {529: <dbus_fast.signature.Variant ('ay', bytearray(b'\x11\x02\xa3\xea(\xeb\x1f\x00\x01\x01\x00\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2024-02-10 11:30:40,901 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_34_13_43_70_D9_2E', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 34:13:43:70:D9:2E)>, 'AddressType': <dbus_fast.signature.Variant ('s', public)>, 'Name': <dbus_fast.signature.Variant ('s', 7B19400144A4D0F4)>, 'Alias': <dbus_fast.signature.Variant ('s', 7B19400144A4D0F4)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -56)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {529: <dbus_fast.signature.Variant ('ay', bytearray(b'\x11\x02.\xd9pCD\x00\x01\x04\x004\x13Cx\xd9.\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2024-02-10 11:30:40,910 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C0_23_8D_D6_3E_AA): ['org.bluez.Device1', {}, ['RSSI']]
2024-02-10 11:30:40,911 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved (/): ['/org/bluez/hci0/dev_C0_23_8D_D6_3E_AA', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
2024-02-10 11:30:40,911 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_34_13_43_70_D9_2E): ['org.bluez.Device1', {}, ['RSSI']]
2024-02-10 11:30:40,911 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_6D_EB_28_EA_A3): ['org.bluez.Device1', {}, ['RSSI']]
2024-02-10 11:30:40,911 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
2024-02-10 11:30:40,914 bleak.backends.bluezdbus.client MainThread DEBUG: Connecting to BlueZ path /org/bluez/hci0/dev_34_13_43_70_D9_2E
2024-02-10 11:30:50,915 bleak.backends.bluezdbus.client MainThread DEBUG: _cleanup_all(/org/bluez/hci0/dev_34_13_43_70_D9_2E)
02/10/24 11:30:50.0915 WARNING - acync.mesh mesh:428 -> telink mesh:connect: EXCEPTION! Unable to CONNECT to device: 34:13:43:70:D9:2E [Cync Plug] -->
Traceback (most recent call last):
  File "/home/baudneo/cync2mqtt/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 209, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/home/baudneo/cync2mqtt/venv/lib/python3.11/site-packages/dbus_fast/aio/message_bus.py", line 384, in call
    await future
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/baudneo/PycharmProjects/cync2mqtt/src/acync/mesh.py", line 422, in connect
    await self.client.connect()
  File "/home/baudneo/PycharmProjects/cync2mqtt/src/acync/mesh.py", line 267, in connect
    status = await self.client.connect(timeout=timeout)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/baudneo/cync2mqtt/venv/lib/python3.11/site-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/baudneo/cync2mqtt/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 146, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
[... cut ...]
2024-02-10 11:50:33,046 bleak.backends.bluezdbus.client MainThread DEBUG: Connecting to BlueZ path /org/bluez/hci0/dev_34_13_43_70_D9_2E
2024-02-10 11:50:33,049 bleak.backends.bluezdbus.client MainThread DEBUG: _cleanup_all(/org/bluez/hci0/dev_34_13_43_70_D9_2E)
02/10/24 11:50:33.0049 WARNING - acync.mesh mesh:428 -> telink mesh:connect: EXCEPTION! Unable to CONNECT to device: 34:13:43:70:D9:2E [Cync Plug] --> Device with address 34:13:43:70:D9:2E was not found. It may have been removed from BlueZ when scanning stopped.
```

baudneo, 2024-02-10:
timeout error pcap, `btmon` and `bluetoothctl` logs ->

[plug-connect_timeout.pcapng.txt](https://github.com/hbldh/bleak/files/14230252/plug-connect_timeout.pcapng.txt)
[btctl.timeout.log](https://github.com/hbldh/bleak/files/14230254/btctl.timeout.log)
[btmon.timeout.log](https://github.com/hbldh/bleak/files/14230255/btmon.timeout.log)

baudneo, 2024-02-10:
write not permitted logs ->

[btmon.writenotpermitted.log](https://github.com/hbldh/bleak/files/14230264/btmon.writenotpermitted.log)
[btctl.writenotpermitted.log](https://github.com/hbldh/bleak/files/14230265/btctl.writenotpermitted.log)
[write-not-permitted.pcapng.txt](https://github.com/hbldh/bleak/files/14230267/write-not-permitted.pcapng.txt)

dlech, 2024-02-11:
The packet logs for the timeout error also have the write not permitted error, so I'm guessing it is the same problem causing both exceptions.

The write not permitted error is coming from the device you are connecting to when trying to write the Client Characteristic Configuration descriptor for one of the characteristics. So either this needs some sort of authentication or the device is defective. In either case, similar to #972.

closed 2024-06-15 by baudneo