hbldh/bleak #1506: Crashes with Excit Code -1073741819 (0xC0000005)

savejeff, 2024-02-11:
* bleak version: 0.21.1
* Python version: 3.12
* Operating System: Windows 10

### Description

I use bleak in my project for an optional feature for a while but about half a year ago. Then i started getting random crashes with this error message:` Process finished with exit code -1073741819 (0xC0000005)`

My Project is a modular program where each module runs in its own thread using Threading (not Multithreading, so no multi core) 
The BLE Module basically only scans for devices, after a few seconds it crashes. 

I used bleak with threading in the project for over two years without having any issues, then it started out of the blue. I don't remember updating bleak to a newer version at the time these crashes started, so my first thought was maybe a windows update that changed the Bluetooth API ?

As it is an optional feature, I've put of investigating what causes this. Any suggestions what might cause this?

dlech, 2024-02-11:
Can you share a minimal reproducible test case?

savejeff, 2024-02-11:
Funny thing, im currently unable to reproduce it, Ill report back when i made progress

closed 2025-12-28 by github-actions[bot]