hbldh/bleak #1529: Windows 10: DLL load failed while importing _winrt_windows_devices_bluetooth

chrisgeli, 2024-03-21:
* bleak version: 0.21.1 
* Python version: 3.12.2 
* Operating System: Windows 10 Pro, build 19045.4170

### Description

I am trying to use bleak on Windows 10, but when instantiating BleakScanner, it crashes.

### What I Did

```
Python 3.12.2 (tags/v3.12.2:6abddd9, Feb  6 2024, 21:26:36) [MSC v.1937 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.                
>>> from bleak import BleakScanner, BleakClient, BleakGATTCharacteristic, BleakError
>>> detection_callback = lambda bt_dev, adv_data : print(f'detection_callback({bt_dev!r}, {adv_data!r})')
>>> bs = BleakScanner(detection_callback)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python3.12\Lib\site-packages\bleak\__init__.py", line 145, in __init__
    get_platform_scanner_backend_type() if backend is None else backend
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python3.12\Lib\site-packages\bleak\backends\scanner.py", line 294, in get_platform_scanner_backend_type
    from bleak.backends.winrt.scanner import BleakScannerWinRT
  File "C:\Python3.12\Lib\site-packages\bleak\backends\winrt\scanner.py", line 8, in <module>
    from winrt.windows.devices.bluetooth.advertisement import (
  File "C:\Python3.12\Lib\site-packages\winrt\windows\devices\bluetooth\__init__.py", line 6, in <module>
    from winrt import _winrt_windows_devices_bluetooth
ImportError: DLL load failed while importing _winrt_windows_devices_bluetooth: The specified module could not be found.
```

### Logs

I repeated the same as above after `set BLEAK_LOGGING=1` in my cmd.exe, but no additional log output was generated.

chrisgeli, 2024-03-21:
Some additional version information that I thought might be helpful:
```
python -m pip install bleak==0.21.1
Requirement already satisfied: bleak==0.21.1 in c:\python3.12\lib\site-packages (0.21.1)
Requirement already satisfied: winrt-Windows.Devices.Bluetooth==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-Windows.Devices.Bluetooth.Advertisement==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-Windows.Devices.Bluetooth.GenericAttributeProfile==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-Windows.Devices.Enumeration==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-Windows.Foundation==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-Windows.Foundation.Collections==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-Windows.Storage.Streams==2.0.0b1 in c:\python3.12\lib\site-packages (from bleak==0.21.1) (2.0.0b1)
Requirement already satisfied: winrt-runtime==2.0.0-beta.1 in c:\python3.12\lib\site-packages (from winrt-Windows.Devices.Bluetooth==2.0.0b1->bleak==0.21.1) (2.0.0b1)
```

The [bleak documentation page about the Windows backend](https://bleak.readthedocs.io/en/latest/backends/windows.html) states that bleak uses the [dlech/bleak-winrt](https://github.com/dlech/bleak-winrt) package, but `pip`'s version output seems to suggest the package used is [pywinrt/pywinrt](https://github.com/pywinrt/pywinrt/)?

A potentially related issue: https://github.com/hbldh/bleak/issues/741

dlech, 2024-03-22:
> The [bleak documentation page about the Windows backend](https://bleak.readthedocs.io/en/latest/backends/windows.html) states that bleak uses the [...]

I fixed the docs now.

> Maybe this is a bug in `winrt`?

Maybe? Does `_winrt_windows_devices_bluetooth.pyd` exist alongside other similarly named files in `C:\Python3.12\Lib\site-packages\winrt\`?

chrisgeli, 2024-03-22:
## Testing with Python 3.11:

The dependencies are completely different:
```
> python -V
Python 3.11.5
> python -m pip install bleak==0.21.1
Requirement already satisfied: bleak==0.21.1 in c:\users\chris.g\appdata\local\programs\python\python311\lib\site-packages (0.21.1)
Requirement already satisfied: bleak-winrt<2.0.0,>=1.2.0 in c:\users\chris.g\appdata\local\programs\python\python311\lib\site-packages (from bleak==0.21.1) (1.2.0)
Requirement already satisfied: typing-extensions>=4.7.0 in c:\users\chris.g\appdata\local\programs\python\python311\lib\site-packages (from bleak==0.21.1) (4.10.0)
```

And, importantly, it works:
```
Python 3.11.5 (tags/v3.11.5:cce6ba9, Aug 24 2023, 14:38:34) [MSC v.1936 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import winrt
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'winrt'
>>> import bleak_winrt
>>> dir(bleak_winrt)
['ExtensionFileLoader', 'ModuleType', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '_import_ns_module', '_winrt', 'module_from_spec', 'spec_from_loader']
>>> from bleak import BleakScanner, BleakClient, BleakGATTCharacteristic, BleakError
>>> detection_callback = lambda bt_dev, adv_data : print(f'detection_callback({bt_dev!r}, {adv_data!r})')
>>> bs = BleakScanner(detection_callback)
>>> # no exception here ^^
```

chrisgeli, 2024-03-22:
> > Maybe this is a bug in `winrt`?
> 
> Maybe? Does `_winrt_windows_devices_bluetooth.pyd` exist alongside other similarly named files in `C:\Python3.12\Lib\site-packages\winrt\`?

There is a `_winrt_windows_devices_bluetooth.pyi` file, but none with `.pyd` extension:

```
C:\Python3.12\Lib\site-packages\winrt>dir                                                                                          
<...>
29/02/2024  01:45 PM    <DIR>          .                                                                                           
29/02/2024  01:45 PM    <DIR>          ..                                                                                          
29/02/2024  01:45 PM    <DIR>          system                                                                                      
29/02/2024  01:45 PM    <DIR>          windows                                                                                     
29/02/2024  01:45 PM            57,856 _winrt.cp312-win_amd64.pyd                                                                  
29/02/2024  01:45 PM           425,472 _winrt_windows_devices_bluetooth.cp312-win_amd64.pyd                                        
29/02/2024  01:45 PM            22,767 _winrt_windows_devices_bluetooth.pyi                                                        
29/02/2024  01:45 PM           339,968 _winrt_windows_devices_bluetooth_advertisement.cp312-win_amd64.pyd                          
29/02/2024  01:45 PM            13,777 _winrt_windows_devices_bluetooth_advertisement.pyi                                          
29/02/2024  01:45 PM           908,288 _winrt_windows_devices_bluetooth_genericattributeprofile.cp312-win_amd64.pyd                
29/02/2024  01:45 PM            42,632 _winrt_windows_devices_bluetooth_genericattributeprofile.pyi                                
29/02/2024  01:45 PM           500,224 _winrt_windows_devices_enumeration.cp312-win_amd64.pyd                                      
29/02/2024  01:45 PM            19,605 _winrt_windows_devices_enumeration.pyi                                                      
29/02/2024  01:45 PM           325,632 _winrt_windows_foundation.cp312-win_amd64.pyd                                               
29/02/2024  01:45 PM            23,635 _winrt_windows_foundation.pyi                                                               
29/02/2024  01:45 PM           190,464 _winrt_windows_foundation_collections.cp312-win_amd64.pyd                                   
29/02/2024  01:45 PM            14,516 _winrt_windows_foundation_collections.pyi                                                   
29/02/2024  01:45 PM           438,272 _winrt_windows_storage_streams.cp312-win_amd64.pyd                                          
29/02/2024  01:45 PM            27,300 _winrt_windows_storage_streams.pyi                                                          
              15 File(s)      3,350,408 bytes                                                                                      
```

PR #1562 (2024-05-04): Release/0.22.0

Added
-----
* Added ``BleakCharacteristicNotFoundError`` which is raised if a device does not support a characteristic.
* Added utility function to work around ``pywin32`` setting threading model to STA on Windows.

Changed
-------
* Updated PyObjC dependency on macOS to v10.x.
* Updated missing Bluetooth SIG characteristics and service UUIDs.
* Updated ``BlueZManager`` to remove empty interfaces from `_properties` during InterfacesRemoved message.
* Updated PyWinRT dependency to v2. Fixes #1529.
* Raise exception when trying to scan while in a single-treaded apartment (STA) on [... cut]
closed 2024-04-29 by dlech