hbldh/bleak #1530: No module named 'dbus_fast' in Windows when import BleakClientBlueZDBus

vovagorodok, 2024-03-25:
* bleak version: 0.21.1
* Python version: latest
* Operating System: Windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
In workaround function:
```
async def acquire_mtu(client: BleakClient):
    from bleak.backends.bluezdbus.client import BleakClientBlueZDBus
    if type(client._backend) is not BleakClientBlueZDBus:
        return

    # in Linux acquire mtu should be called in order to have more than 23
    await client._backend._acquire_mtu()
```
When `import BleakClientBlueZDBus` called in Windows, following exception raised:
`No module named 'dbus_fast'`

Code: https://github.com/vovagorodok/ArduinoBleOTA/blob/98934bd7e069ddaf090b5ae60cf466e3f4a072c0/tools/uploader.py#L87

Should it be fixed in bleak side or script should be changed somehow?

### What I Did

N/A

### Logs

N/A

dlech, 2024-03-25:
The problem isn't with Bleak but rather with the code you linked. It could be fixed by wrapping the import statement in a `try` block or by putting [this line](https://github.com/vovagorodok/ArduinoBleOTA/blob/98934bd7e069ddaf090b5ae60cf466e3f4a072c0/tools/uploader.py#L134) in an `if` block so that it only runs on Linux.

vovagorodok, 2024-03-25:
Ok, thanks @dlech !

closed 2024-03-25 by vovagorodok

vovagorodok, 2024-03-25:
Fix: https://github.com/vovagorodok/ArduinoBleOTA/commit/ae00dcc2795379015a9dae8321e0edef692e5231