hbldh/bleak #1534: BleakScanner service_uuids argument not working in BlueZ backend

dlech, 2024-04-07:
Unsure if this is relevant to this issue - but I found that on bluez the `service_uuids` argument is ignored and the callback is called on non-matching devices.

- Bleak: 0.21.1
- Python version: 3.11.7
- Operating System (`uname -a`): Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- BlueZ version (bluetoothctl -v) in case of Linux: bluetoothctl: 5.66

Example:

```python
import asyncio

from bleak import BleakScanner, AdvertisementData, BLEDevice

SERVICE_UUID = "0e140000-0af1-4582-a242-773e63054c68"

async def detection_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    print(f"Found device: {device.address} (device.name)")
    print(f"Matches service UUID: {SERVICE_UUID in advertisement_data.service_uuids}")

async def discover():
    scanner = BleakScanner(
        detection_callback=detection_callback,
        use_bdaddr=False,
        service_uuids=[SERVICE_UUID],
    )

    await scanner.start()
    await asyncio.sleep(20)

asyncio.run(discover())
```

Results:
```
Found device: 68:27:37:56:DD:C3 - [AV] Samsung Soundbar MS650
Matches service UUID: False
Found device: 0A:9F:E3:B6:74:C9 - 0A-9F-E3-B6-74-C9
Matches service UUID: False
Found device: 00:1B:66:E0:80:F0 - MOMENTUM 3
Matches service UUID: False
```

This doesn't happen on Windows in my tests.

_Originally posted by @filmkorn in https://github.com/hbldh/bleak/issues/1447#issuecomment-2041572075_

dheerajluffy, 2024-04-22:
is this anyway related to the issue, i.e not able to list UUIDS of some LE devices? I have been facing this issue from a week.

dlech, 2024-04-28:
I am not able to reproduce this issue. I tried:

```
python examples/detection_callback.py --services 0e140000-0af1-4582-a242-773e63054c68
```

And I get no callbacks.

BlueZ version: 5.64

Dheeraj-Tachyon, 2024-04-29:
> I am not able to reproduce this issue. I tried:
> 
> ```
> python examples/detection_callback.py --services 0e140000-0af1-4582-a242-773e63054c68
> ```
> 
> And I get no callbacks.
> 
> BlueZ version: 5.64

Could it be a bug in bluez user space?

filmkorn, 2024-04-29:
> python examples/detection_callback.py --services 0e140000-0af1-4582-a242-773e63054c68

I just gave that command a try. Found that the simple callback is only triggered for other service ids if home assistant is running (has a bleak based bluetooth integration). My guess is that home assistant somehow removes the service id filter.

dlech, 2024-04-29:
If another application is scanning, then yes the filter could be removed. I suppose that we could add an additional filter in Bleak to get the expected results even when this happens.

closed 2024-04-30 by dlech

MarkusPiotrowski, 2024-05-21:
So, service_uuid filtering does work in Windows? 
I try to filter on:
``` service_uuids = ['0000fef3-0000-1000-8000-00805f9b34fb']```
like this:
```
...
service_uuids = ['0000fef3-0000-1000-8000-00805f9b34fb']
# Without filter
async with Scanner(service_uuids=None) as scanner:
    await asyncio.sleep(10)
    self.show_scan_result(
        scanner.discovered_devices_and_advertisement_data
    )
# With filter
async with Scanner(service_uuids=service_uuids) as scanner:
    await asyncio.sleep(10)
    self.show_scan_result(
        scanner.discovered_devices_and_advertisement_data
    )
...
```
(where `show_scan_results` is a method to display the results).
However, I can't see a difference in the devices beeing found, and some of them don't have the service uuid in their AdvertisementData. 
Tested with 0.21.1 and 0.22.1

Am I doing something wrong or do I misunderstand the concept of service UUID filtering?