hbldh/bleak #1535: Scannner does not return any BLE devices [backends: winrt]

Jackiii1989, 2024-04-09:
* **bleak version:** 
bleak                                                   0.21.1
numpy                                                   1.26.4
pip                                                     23.2.1
winrt-runtime                                           2.0.0b1
winrt-Windows.Devices.Bluetooth                         2.0.0b1
winrt-Windows.Devices.Bluetooth.Advertisement           2.0.0b1
winrt-Windows.Devices.Bluetooth.GenericAttributeProfile 2.0.0b1
winrt-Windows.Devices.Enumeration                       2.0.0b1
winrt-Windows.Foundation                                2.0.0b1
winrt-Windows.Foundation.Collections                    2.0.0b1
winrt-Windows.Storage.Streams                           2.0.0b1

* **Python version:**
Python 3.12.1

* ** Operating System:**
OS Name:                   Microsoft Windows 11 Pro
OS Version:                10.0.22631 N/A Build 22631

### Description

I used the bleak library yesterday and everything worked fine. I tried it again today and bleak does not find any BLE devices in the area, even though there are tons of BLE devices. I did a Windows update. Maybe that explains the problem.  Am I doing something wrong? Thanks for the help.

### What I Did

```
def test_bleak():

    import logging
    from bleak import BleakScanner
    from bleak.backends.device import BLEDevice
    from bleak.backends.scanner import AdvertisementData

    logger = logging.getLogger(__name__)

    logging.basicConfig(
        level=logging.DEBUG,
        format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
    )

    async def main1():
        logger.info("starting scanner example 1")
        devices = await BleakScanner.discover()
        for d in devices:
            logger.info(d)

    async def main2():

        def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
            logger.info("%s: %r", device.address, advertisement_data)

        scanner = BleakScanner(
            simple_callback
        )

        logger.info("starting scanner example 2")
        await scanner.start()
        await asyncio.sleep(19.0)
        await scanner.stop()

    asyncio.run(main1())
    asyncio.run(main2())

    logger.info("Example finished")

if __name__ == '__main__':
    test_bleak()
```

### Logs

```
(.venv) PS C:\python_workspace\test> $env:BLEAK_LOGGING=1                                           
(.venv) PS C:\python_workspace\test> python.exe .\main.py                                           
2024-04-09 11:57:11,660 asyncio  DEBUG: Using proactor: IocpProactor
2024-04-09 11:57:11,661 __main__ INFO: starting scanner example 1
2024-04-09 11:57:11,784 bleak.backends.winrt.scanner MainThread DEBUG: 0 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>.
2024-04-09 11:57:11,784 bleak.backends.winrt.scanner DEBUG: 0 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>.
2024-04-09 11:57:16,774 bleak.backends.winrt.scanner MainThread DEBUG: skipping waiting for stop because status is <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>
2024-04-09 11:57:16,774 bleak.backends.winrt.scanner DEBUG: skipping waiting for stop because status is <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>
2024-04-09 11:57:16,775 asyncio  DEBUG: Using proactor: IocpProactor
2024-04-09 11:57:16,775 __main__ INFO: starting scanner example 2
2024-04-09 11:57:16,776 bleak.backends.winrt.scanner MainThread DEBUG: 0 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>.
2024-04-09 11:57:16,776 bleak.backends.winrt.scanner DEBUG: 0 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>.
```

This is what I see on nRF Connect App from Nordic:

[image]

dlech, 2024-04-09:
>     asyncio.run(main1())
>     asyncio.run(main2())

There should only be one `asyncio.run()` per application. Otherwise things can break easily. Doesn't cause the problem you are seeing though.

> Watcher status: <BluetoothLEAdvertisementWatcherStatus.ABORTED: 4>.

This indicates that there is something wrong with your Bluetooth adapter. E.g. it is disabled or something like that.

We should probably catch this and raise and exception instead of failing silently with no advertisements received.

So, yes, likely something related to Windows update or just have Bluetooth switch turned off. But there is some improvement we could make to Bleak to better help diagnose such issues.

Jackiii1989, 2024-04-10:
Hey, 

Thanks for the ideas on how to write it better. The problem seemed to be on the Windows side. Now it works again. 

> We should probably catch this and raise and exception instead of failing silently with no advertisements received.

For my side that would be helpful, or an error message then I can narrow down my problem next.  

Cheers,

dlech, 2024-04-10:
> The problem seemed to be on the Windows side.

If the problem is reproducible, knowing how to reproduce it for testing would be helpful.

Jackiii1989, 2024-04-14:
> If the problem is reproducible, knowing how to reproduce it for testing would be helpful.

I do not know how to reproduce the problem. After the second Windows Update, the bug was gone.

PR #1562 (2024-05-04): Release/0.22.0

Added
-----
* Added ``BleakCharacteristicNotFoundError`` which is raised if a device does not support a characteristic.
* Added utility function to work around ``pywin32`` setting threading model to STA on Windows.

Changed
-------
* Updated PyObjC dependency on macOS to v10.x.
* Updated missing Bluetooth SIG characteristics and service UUIDs.
* Updated ``BlueZManager`` to remove empty interfaces from `_properties` during InterfacesRemoved message.
* Updated PyWinRT dependency to v2. Fixes #1529.
* Raise exception when trying to scan while in a single-treaded apartment (STA) on [... cut]
closed 2024-04-28 by dlech