hbldh/bleak #1565: Add option to supress thread is not MTA exception

dlech, 2024-05-07:
Although, using the workaround from the troubleshooting docs might break Kivy if Kivy needs the main thread to be STA for the UI.

So we might need to add an option to avoid the exception in cases like this.

_Originally posted by @dlech in https://github.com/hbldh/bleak/discussions/1564#discussioncomment-9346278_

If Bleak is integrated with a UI framework that has proper asyncio integration, we need to be able to avoid this exception.

closed 2024-05-07 by dlech

MarkusPiotrowski, 2024-05-21:
@dlech 
Let me start by saying that I do not understand anything about STA or MTA and what the exact reason was to introduce this "not MTA" BleakError.

However, .... if I understand correctly what you write above, this breaks Kivy (?), it breaks BeeWare (!, just tested by me) and it breaks other GUIs like tkinter although the asyncio integration might be done properly.
So, there is some code around breaking and needing a somewhat ugly (sorry) import switch that wasn't needed before.

I cannot judge what was worse, the situation before 0.22 or after 0.21. From my own perspective I would say: Please, revert this change. Or find a solution that does not break major Python frameworks by default.

dlech, 2024-05-21:
> what the exact reason was to introduce this "not MTA" BleakError.

There have been at least half a dozen long standing issues of Bleak just locking up on Windows. We finally figured this out was due to importing packages with unwanted side effects. This was the best way I could figure out how to detect those unwanted side effects. But I'm not really happy with the result either. I did not expect very many people to be using Bleak with GUI's on Windows though, but maybe I was wrong!

> Or find a solution that does not break major Python frameworks by default.

Open to suggestions here. For example, is there a way to detect if an app is running via BeeWare at runtime?

MarkusPiotrowski, 2024-05-22:
I asked the BeeWare authors, waiting for an answer. Actually, BeeWare is a set of different tools and libraries, the GUI part is called "toga".

I read through #1132 and realized that this is a Win 11 issue. So you could possibly lift the burden from Win 10 users? Also, how about downgrading the exception to a warning?

MarkusPiotrowski, 2024-05-22:
I got an answer from Russell Keith-Magee (@freakboy3742), the 'boss' of the BeeWare project:

> Any "am I running in Toga" runtime check that I can think of would involve trying to import toga and then checking if toga.App.app exists. If it does, a Toga app has been created; it won't necessarily be running yet, though (and it might have terminated)
> AIUI, the STA requirement is needed for Winforms to operate correctly. If I've misunderstood that requirement, I'm happy to entertain fixes.

PR #1568 (2024-05-07): Release/v0.22.1

Added
-----
* Added ``bleak.backends.winrt.util.allow_sta()`` method to allow integration
  with graphical user interfaces on Windows. Fixes #1565.