hbldh/bleak #1570: macOS 10.14 returns empty list after discovery

Krupskis, 2024-05-09:
* bleak version: 0.22.1
* Python version: 3.12.3
* Operating System: macOS 10.14

### Description

Discover bluetooth devices

### What I Did
Running this simple script works well on m2 macbook Sonoma 14.1.2, but fails on macbook intel 10.14, as it returns an empty list.
```
async def main():
    scanner = BleakScanner(service_uuids=[SERVICE_UUID])
    await scanner.start()
    # Wait for 5 seconds to find the device
    await asyncio.sleep(5)
    await scanner.stop()
    devices = await scanner.get_discovered_devices()
    print(devices)
    compass = None

    for device in devices:
        if device.name and DEVICE_NAME in device.name:
            compass = device

    if compass is None:
        print("Could not find the compass device. Try restarting the wearable.")
        return
    

    try:
        async with BleakClient(compass.address) as client:
            # Read the value of the characteristic
            await client.start_notify(CHARACTERISTIC_UUID, handle_audio_data)

            while True:
                await asyncio.sleep(1)
            
    except Exception as e:
        print(e)
        print(f"Error: {e}")

    

asyncio.run(main())
```

Initially had a script without specifying `service_uuids`, but that failed as well.

```
Could not find the compass device. Try restarting the wearable.
```

### Logs
no errors

dlech, 2024-05-09:
> scanner = BleakScanner(service_uuids=[SERVICE_UUID])

Hmm... I suppose the behavior could have changed in the CoreBluetooth implementations between macOS versions. I doubt I have a version of macOS that old for testing though.

As a workaround, you can move the service UUID check elsewhere.

> # Wait for 5 seconds to find the device

Tip: Using `BleakScanner.find_device_by_name()` would save you from always having to wait 5 seconds and would save you a lot of lines of code.

> async with BleakClient(compass.address) as client:

Tip: just pass `compass`, otherwise it has to scan again.

dlech, 2024-05-09:
> Initially had a script without specifying `service_uuids`, but that failed as well.

Oh, didn't see that part. Not sure what could be going on in that case then. You could try logging Bluetooth packets as described in the troubleshooting page of the docs.

dlech, 2025-05-10:
Closing since this seems to be an issue with old macOS.

closed 2025-05-10 by dlech