hbldh/bleak #1585: List adapters and specify which adapter to use?

EdwardDeaver, 2024-06-01:
* bleak version: Latest
* Python version: 3.10
* Operating System: Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I can not tell how to connect to one adapter when you have multiple and list the devices that the adapter sees.

### What I Did

Read the docs but I don't see how you can set the deviceID to use. And even how to get that devicesID.

dlech, 2024-06-01:
There currently isn't a way to list adapters.

On Linux, you can add an `adapter="hci1"` keyword argument to `BleakScanner` and `BleakClient`.

koenvervloesem, 2024-06-02:
To list the available Bluetooth adapters on your system, you can use the [bluetooth-adapters](https://github.com/Bluetooth-Devices/bluetooth-adapters) package.

roethigj, 2025-02-25:
How does bleak handle the `adapter=hci1` argument in different OS's? Is it just ignored, or will it cause trouble?

dlech, 2025-02-25:
I'm pretty sure it is just ignored.